zephyrproject-rtos/zephyr #70860: scripts: twister: Can we ever find unrecognized sections in tests?

LukaszMrugala, 2024-03-28:
**Describe the bug**
Currently, in Twister, there is a flag `--disable-unrecognized-section-test`. It supposed to work like this: if the flag is enabled, the test is disabled and tests with unrecognized sections shall pass. If the flag is disabled or not used at all, the test is enabled and any test with any unrecognized sections shall fail.

However, no matter what, the tests will not fail. Most end users will never see an error message and will never see a test failure stemming from unrecognized sections.

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using?
   - Ubuntu Linux

**To Reproduce**
Steps to reproduce the behavior:
1. Create such a test, using a template for blackbox tests:
```python
    @pytest.mark.parametrize(
        'flags, expected_result',
        # '1' here means test failure. 0 means test pass.
        [([], '1'), (['--disable-unrecognized-section-test'], '0')],
        ids=['no disable flag', 'disabled']
    )
    @mock.patch.object(TestPlan, 'TESTSUITE_FILENAME', testsuite_filename_mock)
    # This mock makes all read only sections unrecognized
    @mock.patch.object(SizeCalculator, 'ro_sections', [])
    def test_disable_warnings_as_errors(self, out_path, flags, expected_result):
        test_platforms = ['frdm_k64f']
        path = os.path.join(TEST_DATA, 'tests', 'dummy', 'device', 'group')
        args = ['-i', '--outdir', out_path, '-T', path] + \
               flags + \
               ['--enable-size-report'] + \
               [val for pair in zip(
                   ['-p'] * len(test_platforms), test_platforms
               ) for val in pair]

        with mock.patch.object(sys, 'argv', [sys.argv[0]] + args), \
                pytest.raises(SystemExit) as exc:
            self.loader.exec_module(self.twister_module)

        assert str(exc.value) == expected_result
```
2. Run the test
3. Notice that no error was raised.

**Expected behavior**
End user should be notified of unrecognized sections via Twister test failure (return code 1) and some log information about them.

**Impact**
Whole feature of section recognition seems useless in Twister.

**Logs and console output**
Lack of relevant output is the bug.

**Environment (please complete the following information):**
 - OS: Ubuntu Linux
 - Toolchain Zephyr SDK
 - Commit SHA: e0133cfa170e196b60e7d55c3b4b712d9ea2e56e

**Additional context**
We can notice which code fragments cause such a reaction from Twister.

Log information should be provided by the `Reporting.summary(...)` method. It is the only place where the previously mentioned flag is checked: https://github.com/zephyrproject-rtos/zephyr/blob/cb331c2371144f3c391856eec4417ce65cbf0957/scripts/pylib/twister/twisterlib/reports.py#L459-L513

Value for `instance.metrics` is set in the `runner.py`, more specifically inside `gather_metrics(...)` and `calc_size(...)`: https://github.com/zephyrproject-rtos/zephyr/blob/cb331c2371144f3c391856eec4417ce65cbf0957/scripts/pylib/twister/twisterlib/runner.py#L1135-L1164

However, every time we `run(...)` tests, the relevant `metrics` is cleared, thus making our effort vain: https://github.com/zephyrproject-rtos/zephyr/blob/cb331c2371144f3c391856eec4417ce65cbf0957/scripts/pylib/twister/twisterlib/runner.py#L1230-L1239

If we comment out the `inst.metrics["unrecognized"] = []` line, we can see the test detecting unrecognized sections properly. However, the previously mentioned `summary(...)` method **does not** mark tests as failed per se, rather increments the `failed` variable, which controls only the console colour for the summary. Thus, we get the FAILED, ERROR log, red summary of failed tests, but Twister return code will be `0` (all clear) and we will be able to read that `1 of 1 tests passed`.

Relevant logs for such a case:
[image]

nashif, 2025-02-05:
Lets remove this functionality completely from twister

LukaszMrugala, 2025-04-08:
Functionality removed.