zephyrproject-rtos/zephyr #70885: My Device does not assign IP address to client devices.

munir-zin, 2024-03-29:
I am using PicoW and ESP32 devices to setup WiFIManager. I am starting device in AP and STA both modes simultaneously. When I try to connect a client device, I am getting these issue:
1. AP mode is enabled, but no event for `NET_EVENT_WIFI_AP_ENABLE_RESULT` is triggered.
2. When I connect a client device (Mobile) to AP mode, it is not getting any IP addresses and no event is triggered for `NET_EVENT_WIFI_AP_STA_CONNECTED`. 
3. During DHCP setup I get error as `<err> net_dhcpv4_server: Address pool does not belong to the interface subnet.`.

I have setup modes as below. I also setup the DHCP and tried many configurations. 
```
bool IS_AP_ENABLED = false;

void enableApMode(void)
{
    LOG_DBG("Enabling AP Mode.");
    struct net_if * netifInterface = net_if_get_default();
    struct in_addr gateway_addr = {192, 168, 4, 1};
    struct in_addr netmask_addr = {255, 255, 255, 0};
    struct in_addr baseAddr = gateway_addr;
    LOG_DBG("Gateway: %" PRIu32 ", Netmask: %" PRIu32 "", gateway_addr.s_addr, netmask_addr.s_addr);
    net_if_ipv4_set_netmask(netifInterface, &netmask_addr);
    net_if_ipv4_set_gw(netifInterface, &gateway_addr);
    // netifInterface->config.ip.ipv4->gw.s_addr = htonl(*(uint32_t *)gateway_addr);
    // netifInterface->config.ip.ipv4->netmask.s_addr = htonl(*(uint32_t *)netmask_addr);

    struct wifi_connect_req_params wifiConfig = {
        .ssid = (const uint8_t *)AP_SSID,
        .ssid_length = strlen(AP_SSID),
        .psk = (const uint8_t *)AP_PSK,
        .psk_length = strlen(AP_PSK),
        .band = WIFI_FREQ_BAND_2_4_GHZ,
        .channel = WIFI_CHANNEL_ANY,
    };

    if (strlen(AP_PSK) > 0)
    {
        wifiConfig.security = WIFI_SECURITY_TYPE_PSK;
    }
    else
    {
        wifiConfig.security = WIFI_SECURITY_TYPE_NONE;
    }

    if (net_mgmt(NET_REQUEST_WIFI_AP_ENABLE, netifInterface, &wifiConfig, sizeof(struct wifi_connect_req_params)))
    {
        LOG_ERR("AP Mode Request Failed\n");
    }

    netifInterface = net_if_get_default();

    if (!netifInterface)
    {
        printk("Error: No network interface found!\n");
        return;
    }

    printk("DHCPv4 server started...\n");
    net_dhcpv4_server_start(netifInterface, &baseAddr);
}

void connectWiFi(void)
{
   struct net_if * netifInterface = net_if_get_default();

    struct wifi_connect_req_params wifiConfig =
        {
            .ssid = (const uint8_t *)SSID,
            .ssid_length = strlen(SSID),
            .psk = (const uint8_t *)PSK,
            .psk_length = strlen(PSK),
            .band = WIFI_FREQ_BAND_2_4_GHZ,
            .channel = WIFI_CHANNEL_ANY,
        };

    if (strlen(PSK) > 0)
    {
        wifiConfig.security = WIFI_SECURITY_TYPE_PSK;
    }
    else
    {
        wifiConfig.security = WIFI_SECURITY_TYPE_NONE;
    }

    wifiConfig.band = WIFI_FREQ_BAND_2_4_GHZ;
    wifiConfig.mfp = WIFI_MFP_OPTIONAL;

    LOG_DBG("Connecting to SSID: %s\n", wifiConfig.ssid);
    if (net_mgmt(NET_REQUEST_WIFI_CONNECT, netifInterface, &wifiConfig, sizeof(struct wifi_connect_req_params)))
    {
        LOG_ERR("WiFi Connection Request Failed\n");
    }
}

void handleWiFiEvents(struct net_mgmt_event_callback *cb, uint32_t mgmt_event, struct net_if *iface)
{
    switch (mgmt_event)
    {
    case NET_EVENT_WIFI_CONNECT_RESULT:
       const struct wifi_status *status = (const struct wifi_status *)cb->info;
       LOG_DBG("NET_EVENT_WIFI_CONNECT_RESULT");
       
       if (status->status)
       {
            LOG_ERR("WiFi connection failed (%d).", status->status);
            if (!IS_AP_ENABLED)
                enableApMode();
            else
                connectWiFi();
        }
        break;

    case NET_EVENT_WIFI_DISCONNECT_RESULT:
        LOG_DBG("NET_EVENT_WIFI_DISCONNECT_RESULT");
       if (!IS_AP_ENABLED)
                enableApMode();
      else
                connectWiFi();
        break;

    case NET_EVENT_WIFI_AP_ENABLE_RESULT:
        LOG_DBG("NET_EVENT_WIFI_AP_ENABLE_RESULT");
         IS_AP_ENABLED = true;
        connectWiFi();
        break;

    case NET_EVENT_WIFI_AP_DISABLE_RESULT:
        LOG_DBG("NET_EVENT_WIFI_AP_DISABLE_RESULT");
        IS_AP_ENABLED = false;
        break;

    case NET_EVENT_WIFI_AP_STA_CONNECTED:
        LOG_DBG("NET_EVENT_WIFI_AP_STA_CONNECTED");
        break;

    case NET_EVENT_WIFI_AP_STA_DISCONNECTED:
        LOG_DBG("NET_EVENT_WIFI_AP_STA_DISCONNECTED");
        break;

    case NET_EVENT_IPV4_ADDR_ADD:
        LOG_DBG("NET_EVENT_IPV4_ADDR_ADD");
       
        break;

    default:
        break;
    }
}

void main()
{
    LOG_DBG("Setting up WiFi on %s\n", CONFIG_BOARD);

    net_mgmt_init_event_callback(&cbWiFi, handleWiFiEvents, WIFI_EVENT_MASK);
    net_mgmt_init_event_callback(&cbIPV4, handleWiFiEvents, IPV4_EVENT_MASK);

    net_mgmt_add_event_callback(&cbWiFi);
    net_mgmt_add_event_callback(&cbIPV4);

    // enableApMode();
    connectWiFi();

    LOG_DBG("WiFi Ready...\n\n");
}

jukkar, 2024-03-29:
> 3\. During DHCP setup I get error as `<err> net_dhcpv4_server: Address pool does not belong to the interface subnet.`.

What kind of configuration you have, can you add information about it here?

munir-zin, 2024-03-29:
I have followinc configurations in `prj.conf`
```
CONFIG_NET_DHCPV4=y
CONFIG_NET_DHCPV4_SERVER=y
CONFIG_NET_DEFAULT_IF_WIFI=y
CONFIG_NET_L2_ETHERNET=y
CONFIG_NET_DHCPV4_SERVER_ADDR_COUNT=255
CONFIG_NET_UDP=y
CONFIG_NET_TCP=y
CONFIG_NET_SOCKETS=y
CONFIG_NET_SOCKETS_POSIX_NAMES=y
CONFIG_NET_IPV4=y

CONFIG_WIFI=y
CONFIG_NETWORKING=y
CONFIG_NET_L2_WIFI_MGMT=y
CONFIG_NET_CONFIG_INIT_TIMEOUT=60
CONFIG_NET_CONFIG_AUTO_INIT=n
CONFIG_NET_MGMT_EVENT_STACK_SIZE=2048
CONFIG_NET_MGMT=y
CONFIG_NET_MGMT_EVENT=y
CONFIG_WIFI_MGMT_RAW_SCAN_RESULTS=y
CONFIG_NET_MGMT_EVENT_MONITOR=y
CONFIG_CPP=y
CONFIG_REQUIRES_FULL_LIBCPP=y
CONFIG_INIT_STACKS=y

CONFIG_NET_STATISTICS=y
CONFIG_NET_STATISTICS_WIFI=y
CONFIG_NET_STATISTICS_USER_API=y
```

jukkar, 2024-03-29:
You have not set IPv4 address to the interface, that is why you get DHCPv4 server address poo error.
You can use `net_if_ipv4_addr_add()` to set that example.

munir-zin, 2024-04-01:
I updated code to add address for DHCPv4 as below, but now getting `<err> net_dhcpv4_server: Address pool overlaps with server address.` I tried with different addresses without success.
```
struct net_if *netifInterface = net_if_get_default();
    struct in_addr gateway_addr = {192, 168, 4, 1};
    struct in_addr netmask_addr = {255, 255, 255, 0};
    struct in_addr baseAddr = {192, 168, 4, 1}; // tried with 192.168.44.1
    LOG_DBG("Gateway: %" PRIu32 ", Netmask: %" PRIu32 "", gateway_addr.s_addr, netmask_addr.s_addr);
    net_if_ipv4_set_netmask(netifInterface, &netmask_addr);
    net_if_ipv4_set_gw(netifInterface, &gateway_addr);
    net_if_ipv4_addr_add(netifInterface, &baseAddr, NET_ADDR_DHCP, 4294967295);
    net_dhcpv4_server_start(netifInterface, &baseAddr);
    ```

munir-zin, 2024-04-01:
> I updated code to add address for DHCPv4 as below, but now getting `<err> net_dhcpv4_server: Address pool overlaps with server address.` I tried with different addresses without success.
>
> ```
> struct [...]

I have commented out the code checking for IP overlap as below.
```
// if ((htonl(server_addr->s_addr) >= htonl(base_addr->s_addr)) &&
	//     (htonl(server_addr->s_addr) <
	//      htonl(base_addr->s_addr) + CONFIG_NET_DHCPV4_SERVER_ADDR_COUNT)) {
	// 	LOG_ERR("Address pool overlaps with server address.");
	// 	LOG_ERR("server_addr->s_addr:%d, base_addr->s_addr:%d", server_addr->s_addr, base_addr->s_addr);
	// 	return -EINVAL;
	// }
```

Now, When I try to connect the client device, it still does not get IP address and I am getting following error on console. 

`<err> net_dhcpv4_server: Failed to send Offer message, -115
<err> net_dhcpv4_server: Failed to send Offer message, -115`]

jukkar, 2024-04-01:
> struct in_addr gateway_addr = {192, 168, 4, 1};
>     struct in_addr netmask_addr = {255, 255, 255, 0};
>     struct in_addr baseAddr = {192, 168, 4, 1}; // tried with 192.168.44.1
>     LOG_DBG("Gateway: [...]

You cannot set DHCP client address manually to the interface. The type for manually added addresses is `NET_ADDR_MANUAL`

You cannot set the local address / base address to be the same as gateway address. Gateway tells the host/device in your network where the packets that do not belong to local network are sent. Typically this is for example your wireless access point, or IP router etc.

Your local IP address could be for example 192.168.4.2, gateway address could stay, but base address which specifies the DHCP address pool could be for example 192.168.4.16. All of these addresses depend on your network setup, it is difficult to give exact numbers.

munir-zin, 2024-04-01:
> You cannot set DHCP client address manually to the interface. The type for manually added addresses is `NET_ADDR_MANUAL`
>
> You cannot set the local address / base address to be the same as gateway [...]

I need to configure it for DHCP server not for DHCP client. I added `192.168.4.1` when calling `net_if_ipv4_addr_add()` and `192.168.4.10` when calling `net_dhcpv4_server_start()`, it resolved the errors. 
Now I am facing issues with network interface not configured when I try to connect client device to AP mode. I am wondering, when I setup device in STA mode, it can connect to router and get's IP address while in AP mode, it is giving error of Network interface is not configured. If that is the case, why it is working for STA mode?

munir-zin, 2024-04-01:
I am getting "Network Interface is not configured". 
I have configured the network interface by setting up following configurations in `prj.conf`.

```
CONFIG_NETWORKING=y
CONFIG_NET_CONFIG_INIT_TIMEOUT=60
CONFIG_NET_CONFIG_AUTO_INIT=y
CONFIG_NET_CONFIG_NEED_IPV4=y
```

The error logs are:
```

*** Booting Zephyr OS build v3.6.0 ***
[00:00:02.584,000] <inf> net_config: Initializing network
[00:00:02.584,000] <inf> net_config: Waiting interface 1 (0x3ffb2508) to be up...
[00:00:02.585,000] <err> esp32_wifi: Failed to send packet
[00:00:02.586,000] <err> esp32_wifi: Failed to send packet
[00:00:02.586,000] <err> esp32_wifi: Failed to send packet
[00:00:02.586,000] <inf> net_config: Interface 1 (0x3ffb2508) coming up
[00:00:02.587,000] <err> esp32_wifi: Failed to send packet
[00:00:02.587,000] <inf> net_config: Running dhcpv4 client...
[00:00:03.587,000] <err> esp32_wifi: Failed to send packet
[00:00:04.588,000] <err> esp32_wifi: Failed to send packet
[00:00:11.588,000] <err> esp32_wifi: Failed to send packet
[00:00:16.588,000] <err> esp32_wifi: Failed to send packet
[00:00:24.589,000] <err> esp32_wifi: Failed to send packet
[00:00:41.589,000] <err> esp32_wifi: Failed to send packet
[00:00:50.588,000] <err> net_config: Timeout while waiting network setup
[00:00:50.588,000] <err> net_config: Network initialization failed (-116)
```

**When I try to connect client device, then error logs are:**
```
[00:02:49.293,000] <wrn> net_if: iface 0x3ffb2508 is down
[00:02:49.293,000] <err> net_dhcpv4_server: Failed to send ICMP probe
[00:02:51.209,000] <wrn> net_if: iface 0x3ffb2508 is down
[00:02:51.209,000] <err> net_dhcpv4_server: Failed to send ICMP probe
[00:02:53.224,000] <wrn> net_if: iface 0x3ffb2508 is down
[00:02:53.224,000] <err> net_dhcpv4_server: Failed to send ICMP probe
[00:02:58.027,000] <wrn> net_if: iface 0x3ffb2508 is down
[00:02:58.027,000] <err> net_dhcpv4_server: Failed to send ICMP probe
```

jukkar, 2024-04-01:
The network interface is down so sending is not possible. Are you sure that STA and AP mode can work at the same time, does the device only have one radio?

munir-zin, 2024-04-02:
Yes, I have tested the workings of AP and STA simultaneously on ESP32 and RPi Picow through their direct development platforms. [esp-idf](https://github.com/espressif/esp-idf) for esp32 and [pico-sdk](https://github.com/raspberrypi/pico-sdk).

Even in this case, device is getting IP address in STA mode and exposing it's SSID in AP mode. The only issue is that AP mode is not assigning IP addresses to client devices. 
Network initalization is failing at boot time by setting `CONFIG_NET_CONFIG_AUTO_INIT=y` in `prj.conf` and wifi STA and AP mode are enabled later in application code.

jukkar, 2024-04-02:
I do not have ESP32 so cannot try this myself. Assigning also @sylvioalves as this might be related to ESP32.