zephyrproject-rtos/zephyr #70886: Grove 16*2 LCD problem

Ashu12Aj, 2024-03-29:
I use nrf52832 board , 
the issue I face in sample file,
i try to make a overlay file for i2c communication, the code is build succesfully and and flash also , but when i see in ythe terminal it show  Device is not ready .
is any one have the working code for  grove LCD share ......

my overlay code is this. 

/ {
    chosen {
        zephyr,display = &glcd;
    };

};

&pinctrl {
    i2c0_default: i2c0_default {
        group1 {

            psels = <NRF_PSEL(TWIM_SDA, 0, 26)>,
                    <NRF_PSEL(TWIM_SCL, 0, 27)>;
        };
    };

    i2c0_sleep: i2c0_sleep {
        group1 {

            psels = <NRF_PSEL(TWIM_SDA, 0, 26)>,
                    <NRF_PSEL(TWIM_SCL, 0, 27)>;
            low-power-enable;
        };
    };

};

&i2c0 {
    compatible = "nordic,nrf-twim";
    status = "okay";
    clock-frequency = <I2C_BITRATE_FAST>;
    zephyr,concat-buf-size = <4096>;

    glcd: glcd@3e {
        compatible = "seeed,grove-lcd-rgb";
        reg = <0x3e>;
        label = "GLCD";
        status = "okay";

    };
};

henrikbrixandersen, 2024-03-30:
Please use our bug template when reporting bugs. You need to edit this issue to include the information requested in https://github.com/zephyrproject-rtos/zephyr/blob/main/.github/ISSUE_TEMPLATE/001_bug_report.md

thedjnK, 2024-04-02:
That error likely means the device was not detected, ensure you have connected to the correct pins and have pull up resistors in place

thedjnK, 2024-05-01:
@Ashu12Aj were you able to fix this issue?

closed 2024-07-16 by github-actions[bot]