zephyrproject-rtos/zephyr #70887: ieee802154_header_ie relies on undefined behavior of bit-fields

Damian-Nordic, 2024-03-29:
**Is your enhancement proposal related to a problem? Please describe.**
The `ieee802154_header_ie` [structure](https://github.com/zephyrproject-rtos/zephyr/blob/ff13cb5adc0b4abeb3aeb8a94de59b509d4988a3/include/zephyr/net/ieee802154_ie.h#L133) seems to have been designed in a way that enables direct serialization of the structure into the binary representation of a 15.4 Header IE. The problem is that it assumes that bit-fields in a single byte are encoded in a specific order, which is undefined behavior in C.

After https://en.cppreference.com/w/c/language/bit_field:

> The following usages of bit-fields causes undefined behavior:
> - ...
> - The order of bit-fields within an allocation unit (on some platforms, bit-fields are packed left-to-right, on others right-to-left) 

**Describe the solution you'd like**
Replace the bit-fields with:
```
struct ieee802154_header_ie {
        uint8_t length_and_id_low;
        uint8_t id_high_and_type;
```

Then adjust or provide helpers and macros to encode and decode these fields properly.

**Describe alternatives you've considered**
Make it clear that the structure must not be serialized directly and put the burden of the actual serialization on concrete 15.4 drivers.
I think this would lead to unnecessary code duplication across drivers.

**Additional context**
The current code works correctly with Zephyr SDK toolchain on the only in-tree platform that uses it, but still we shouldn't have the code that relies on UBs.

Damian-Nordic, 2024-03-29:
@fgrandel @jciupis fyi

fgrandel, 2024-03-29:
Hi @Damian-Nordic, in principle you're right. However as we're limited to a specific toolchain, this was a pre-established practice in the stack. Compare `subsys/net/l2/ieee802154/ieee802154_frame.h`.

Maybe I'm missing some difference here but at least that was the intention. :-)

Damian-Nordic, 2024-03-29:
Thanks, I wasn't aware that this was a preexisting pattern. Zephyr supports many toolchains and may support new ones in the future, so given these circumstances I assumed writing portable code would be a good idea that would pay off sooner or later, but maybe the bit-field layout is something so well established across toolchains that it makes sense to relax the rules in this case.

I'll leave the issue open for a day as this fact puzzled me a bit but if the community accepts it, I'll close the issue.

fgrandel, 2024-03-29:
Thanks for your quick response. I totally agree with you. As soon as it becomes practically relevant, we need to do something here.

Damian-Nordic, 2024-04-02:
closing it for now as agreed above

closed 2024-04-02 by Damian-Nordic