zephyrproject-rtos/zephyr #70889: LoRaWAN US915 NRF52840DK+sx1262 unable to TTN join

hallard, 2024-03-29:
**Describe the bug**

When using US915 LoRaWAN class_a sample we're unable to join on TTN

First to mention, there is no subband (FSB) selection possible under kconfig (see FSB usage [explanation](https://github.com/ARMmbed/mbed-os/blob/master/connectivity/lorawan/FSB_Usage.txt)) so as US915 has 64 channels and our gateway has only 8 channels, we miss about lot of joins request that are not in our subband (just as a reminder, not related to our issue)

Using NRF52840 + semtech_sx1262mb2das shield, tried also with RAK4631 same issue

**To Reproduce**

We changed `main.c` to loop for join to better track of this issue

```c
	ret = -1;
	while (ret < 0) {
    	        // tried this one also just in case without any better success
        	// join_cfg.otaa.dev_nonce = (uint16_t)sys_rand32_get();

		LOG_INF("Joining network over OTAA with devnonce %d", join_cfg.otaa.dev_nonce);
		ret = lorawan_join(&join_cfg);
		if (ret < 0) {
			LOG_ERR("lorawan_join_network failed: %d", ret);
			k_sleep(K_MSEC(1000));
		}
	}

```

```
west build -p -b nrf52840dk_nrf52840 ./zephyr/samples/subsys/lorawan/class_a/ -- -DSHIELD=semtech_sx1262mb2da
```

**Expected behavior**

Device to join LoRaWAN .

**Impact**

LoRaWAN not working US915 on TTN in our config  (not sure on others LNS)

**Logs and console output**

```
*** Booting nRF Connect SDK v3.5.99-ncs1 ***
[00:00:00.339,813] <dbg> lorawan: lorawan_start: LoRaMAC Initialized
[00:00:00.339,965] <inf> lorawan_class_a: Joining network over OTAA with devnonce 0
[00:00:00.357,421] <dbg> lorawan: lorawan_join: Network join request sent!
[00:00:06.906,372] <dbg> lorawan: mlme_confirm_handler: Received MlmeConfirm (for MlmeRequest 1)
[00:00:06.906,402] <err> lorawan: MlmeConfirm failed : Rx 2 timeout
[00:00:06.908,508] <err> lorawan_class_a: lorawan_join_network failed: -116
[00:00:07.908,599] <inf> lorawan_class_a: Joining network over OTAA with devnonce 0
[00:00:07.923,858] <dbg> lorawan: lorawan_join: Network join request sent!
[00:00:14.471,710] <dbg> lorawan: mlme_confirm_handler: Received MlmeConfirm (for MlmeRequest 1)
[00:00:14.471,740] <err> lorawan: MlmeConfirm failed : Rx 2 timeout
[00:00:14.473,846] <err> lorawan_class_a: lorawan_join_network failed: -116
[00:00:15.473,937] <inf> lorawan_class_a: Joining network over OTAA with devnonce 0
[00:00:15.489,196] <dbg> lorawan: lorawan_join: Network join request sent!
[00:00:22.038,055] <dbg> lorawan: mlme_confirm_handler: Received MlmeConfirm (for MlmeRequest 1)
[00:00:22.038,085] <err> lorawan: MlmeConfirm failed : Rx 2 timeout
[00:00:22.040,191] <err> lorawan_class_a: lorawan_join_network failed: -116
[00:00:23.040,283] <inf> lorawan_class_a: Joining network over OTAA with devnonce 0
[00:00:23.055,572] <dbg> lorawan: lorawan_join: Network join request sent!
[00:00:29.604,522] <dbg> lorawan: mlme_confirm_handler: Received MlmeConfirm (for MlmeRequest 1)
[00:00:29.604,553] <err> lorawan: MlmeConfirm failed : Rx 2 timeout
[00:00:29.606,658] <err> lorawan_class_a: lorawan_join_network failed: -116
[00:00:30.606,750] <inf> lorawan_class_a: Joining network over OTAA with devnonce 0
[00:00:30.622,009] <dbg> lorawan: lorawan_join: Network join request sent!
[00:00:37.152,587] <dbg> lorawan: mlme_confirm_handler: Received MlmeConfirm (for MlmeRequest 1)
[00:00:37.152,618] <err> lorawan: MlmeConfirm failed : Rx 1 timeout
[00:00:37.152,648] <dbg> lorawan: mcps_indication_handler: Received McpsIndication 0
[00:00:37.152,679] <err> lorawan: McpsIndication failed : Error
[00:00:37.154,754] <err> lorawan_class_a: lorawan_join_network failed: -116
[00:00:38.154,876] <inf> lorawan_class_a: Joining network over OTAA with devnonce 0
[00:00:38.170,135] <dbg> lorawan: lorawan_join: Network join request sent!
[00:00:44.676,788] <dbg> lorawan: mlme_confirm_handler: Received MlmeConfirm (for MlmeRequest 1)
[00:00:44.676,818] <err> lorawan: MlmeConfirm failed : Rx 2 timeout
[00:00:44.678,924] <err> lorawan_class_a: lorawan_join_network failed: -116
[00:00:45.679,016] <inf> lorawan_class_a: Joining network over OTAA with devnonce 0
[00:00:45.694,274] <dbg> lorawan: lorawan_join: Network join request sent!
[00:00:52.243,133] <dbg> lorawan: mlme_confirm_handler: Received MlmeConfirm (for MlmeRequest 1)
[00:00:52.243,164] <err> lorawan: MlmeConfirm failed : Rx 2 timeout
[00:00:52.245,269] <err> lorawan_class_a: lorawan_join_network failed: -116
[00:00:53.245,361] <inf> lorawan_class_a: Joining network over OTAA with devnonce 0
[00:00:53.260,620] <dbg> lorawan: lorawan_join: Network join request sent!
[00:00:59.809,448] <dbg> lorawan: mlme_confirm_handler: Received MlmeConfirm (for MlmeRequest 1)
[00:00:59.809,478] <err> lorawan: MlmeConfirm failed : Rx 2 timeout
[00:00:59.811,584] <err> lorawan_class_a: lorawan_join_network failed: -116
[00:01:00.811,676] <inf> lorawan_class_a: Joining network over OTAA with devnonce 0
[00:01:00.826,904] <dbg> lorawan: lorawan_join: Network join request sent!
[00:01:07.033,355] <dbg> lorawan: mlme_confirm_handler: Received MlmeConfirm (for MlmeRequest 1)
[00:01:07.033,386] <err> lorawan: MlmeConfirm failed : Rx 2 timeout
[00:01:07.035,491] <err> lorawan_class_a: lorawan_join_network failed: -116
[00:01:08.035,583] <inf> lorawan_class_a: Joining network over OTAA with devnonce 0
[00:01:08.050,842] <dbg> lorawan: lorawan_join: Network join request sent!
[00:01:14.599,670] <dbg> lorawan: mlme_confirm_handler: Received MlmeConfirm (for MlmeRequest 1)
[00:01:14.599,700] <err> lorawan: MlmeConfirm failed : Rx 2 timeout
[00:01:14.601,806] <err> lorawan_class_a: lorawan_join_network failed: -116
[00:01:15.601,898] <inf> lorawan_class_a: Joining network over OTAA with devnonce 0
[00:01:15.617,156] <dbg> lorawan: lorawan_join: Network join request sent!
[00:01:22.165,985] <dbg> lorawan: mlme_confirm_handler: Received MlmeConfirm (for MlmeRequest 1)
[00:01:22.166,015] <err> lorawan: MlmeConfirm failed : Rx 2 timeout
[00:01:22.168,121] <err> lorawan_class_a: lorawan_join_network failed: -116
[00:01:23.168,212] <inf> lorawan_class_a: Joining network over OTAA with devnonce 0
[00:01:23.183,471] <dbg> lorawan: lorawan_join: Network join request sent!
[00:01:29.732,330] <dbg> lorawan: mlme_confirm_handler: Received MlmeConfirm (for MlmeRequest 1)
[00:01:29.732,360] <err> lorawan: MlmeConfirm failed : Rx 2 timeout
[00:01:29.734,466] <err> lorawan_class_a: lorawan_join_network failed: -116
[00:01:30.734,558] <inf> lorawan_class_a: Joining network over OTAA with devnonce 0
[00:01:30.749,816] <dbg> lorawan: lorawan_join: Network join request sent!

```

**Gateway Logs**

As you can see only 2 of the 14 joins are seen, it's because of FSB mentioned above.

[image]

** TN LOG**

[image]

**Additional context**

We have some RAK3172 (STM32WL5) we are using with mbed for long time and we tried on this one to check our infrastructure and also tried with EU868

- MBED + RAK3172 + US915 -> Works 
- MBED + RAK3172 (LP Version) + EU868 -> Works 
- Zephyr + NRF52840 + semtech_sx1262mb2das shield + EU868 -> Works 
- Zephyr + NRF52840 + semtech_sx1262mb2das shield + US915 -> Not working
- Zephyr + RAK4630 + EU868 -> Works 
- Zephyr + RAK4630 + US915 -> Not working

As you can see the issue is only happening on US915 with Zephyr.

We also tried to increase RX error from 20 to 100 just in case with  `CONFIG_LORAWAN_SYSTEM_MAX_RX_ERROR=100` not solving the issue.

Any help would be greatly appreciated @mluis1 @martinjaeger @carlescufi , FTI we already made some first investigations with @kartben

Thanks

Ashu12Aj, 2024-03-29:
can you help me to make grove LCD Display program 
i face configuration essue 
can you give me overlay file for this

kartben, 2024-03-29:
> can you help me to make grove LCD Display program i face configuration essue can you give me overlay file for this

@Ashu12Aj please ask your question on Discord (https://chat.zephyrproject.org/) -- this is an entirely different issue you've commented on, so I doubt anyone will be able to help you here :)

mluis1, 2024-04-02:
@hallard The LoRaWAN US915 region has been designed/specified to use 64x125kHz+8x500kHz channels. Which means that so called 64 channels gateways are are expected to be used. The reasons for this requirement are related to the FCC rules.
The FCC allows to use less channels but in this case the RF output levels must be decreased.
For further information please refer to the following Regional Parameters [RP2-1.0.4](https://resources.lora-alliance.org/technical-specifications/rp002-1-0-4-regional-parameters) specification note:
[image]

The issue being that most network providers in US based regions have decided to deploy 8x125kHz+1x500kHz channels gateways for different reasons (mainly costs).

By doing so one needs to accept the compromises that must taken in consideration when such choice happens.

From a certification point of view as well as network operation all end-devices must support the 64+8 channels, The reason being that your today's network makes use of 8 channels gateways however in the future this may change i.e. 64 channels gateways are added to the network or for some reason the 8 channels gateways are changed to use a different bank of 8 channels or your end-device is at some point connected to a network using the 2nd block of 8 channels and then moves to a network using another bank of channels.

The drawback of using 8 channel gateways is that we must accept that an end-device must potentially do several trials before being heard by a gateway and finalize the join procedure.
However once the join procedure is finalized the network server instructs the end-device to only activate the required channels using the CFList field of the Join Accept frame.

To speed up and reduce the amount of join trials the  [RP2-1.0.4](https://resources.lora-alliance.org/technical-specifications/rp002-1-0-4-regional-parameters) specification recommends the below algorithm.
[image]

As an example lets assume that the network uses the 8th bank of 8 channels. This means that at least 8 trials will be necessary prior to join the network.

In conclusion I would say that your observed behavior looks to be normal.
Another point being that RF signals can be interfered by other signals which if you are unlucky and your first join request supposed to be received isn't then 9 other trials will take place before being joined and this process may repeat up until the join procedure is successful.

As a side comment the RSSI values shown in your screenshots are way to high. Your end-device is shouting out loud on the receiver. I would recommend to move your end-device away from the gateway (at least 10 meters and a wall). For good conditions try to get RSSI values between -70dBm and 90dBm.
By doing so I am confident that you will observe a lot less issues joining the network.

hallard, 2024-04-02:
@mluis1, thank you very much for this crystal clear explanation about joining process on 64+8 channels. As I said, it's a deal (longer join process) that we can handle because once joined all should be fine.

I'm surprised that the issue may comes from end device too close to gateway (that's indeed the case) because it's far for the first time we're using US 915 devices in same situation (and lot of devices are different, even one already made from Elsys or other manufacturers) and it is the first time we're just unable to join. We already join with same device (with mbed lorawan stack) without any issue and even sometime with RSSI signal near -40dB (because as you say, we're close from GW). So of course I will test going far away from the GW but I'm pretty sure it will not change anything because it is something we handle day by day.

Here the logs and capture I've just tested few minutes ago in same situation, same location, same sensor, same gateway, mbed stack US915 (instead of zephyr one), as you can see works immediately even with -56 RSSI, same process with lorawan class a zephyr sample just never join.

```
[LoRaWAN] initialize: Success
[LoRaWAN] set callbacks: Success
[LoRaWAN] device set to class A: Success
[LoRaWAN] LoRa set_confirmed_msg_retries(3) : Success
MBED  join wait  : 3 seconds (random wait)
LoRa connect() joining...
Reset if no join in 24 h
LoRa Event joined
[LoRaWAN] enable adaptive datarate: Success
LoRa scheduled payload[1] on port:1 => 00 
LoRa cb_set_battery_level(3048mV) for 2x1.5V AA battery => F8 (97%)
LoRa Event TX done
```

[image]

[image]

So if someone would be able to do same test trying to connect nrf52840dk+semtech SX1262 shield on TTN it would be awesome we can confirm it just works (or not)

hallard, 2024-04-02:
I've done other tests according @mluis1 recommendation, I'm able going further because now TTN accept join and send downlink join but device still does not join

here TTN log
[image]

and Serial console

[image]

please note to enforce testing `LORAWAN_SYSTEM_MAX_RX_ERROR` is set to `100` instead of `20`

hallard, 2024-04-03:
Hi 

@mluis1 just to give more informations to investigate, today I decided to give a try with nrf52840dk + semtech shield with mbed (so same hardware, same infrastructure) and guess what, it worked 1st time without any issue.

[image]

Martdur, 2024-09-19:
I'm facing the same issue here do you managed to make it works @hallard  ?

hallard, 2024-09-19:
@Martdur sorry, no solution, unable to get this to work. Not sure it has been tested, but looks like team does not handle this as an issue @kartben @mluis1

martinjaeger, 2024-09-19:
@hallard The issue just became stale and was closed automatically by the bot, which does not mean that the Zephyr team treats this as a non-issue. However, sometimes issues are difficult to reproduce or maintainers just don't have the time to try and find the root cause of an issue themselves. I'm currently doing anything LoRa-related in my free time. And as I'm in Europe, I can't even test any LoRaWAN stuff with US settings. I can only say that I have successfully used LoRAWAN on an nrf52840+sx1262 board with TTN in the past, but that must have been about 2 years ago. Now I'm mostly using STM32WL and ChirpStack.

I'm re-opening this issue so that people are aware of it. Feel free to post any updates here. Hopefully someone will finally get to the bottom of it.

hallard, 2024-09-19:
@martinjaeger thanks for your prompt answer, I'm living also in EU but got some US testing infrastructure so I can help to figure this out but I don't have sufficient Zephyr LoRaWAN stack skills (and time to dig into mostly).
Anyway I may switch back to this project sooner or later so I can ping a dev team so we can work together to fix it. Just let me know

Martdur, 2024-09-19:
Hey thanks for the quick answer I'm too in EU developing US and AU product.
I've faced the same issue on stm32wleX, I've managed to join the network by using a mask that fit my gateway FSB.
I now this is not compliant with the lorawan alliance but it works for now.
The only things is that I'm struggling sending confirmed message as I get the Rx2 timeout error.
Messages seems to be received by the server.
I've done those tests in a hurry, I'll investigate further next week and keep you updated.

Martdur, 2024-09-23:
As I've said last week I've managed to connect to the lorawan network and even send a empty buffer on DR0.
But when we want to send larger buffer on the new datarate we get Rx2 Timeout even though we get the  `mcps_confirm_handler: Received McpsConfirm (for McpsRequest 1) `

**see the log :** 
```
[00:00:03.477,000] <inf> App: NetworkSearchState
[00:00:03.477,000] <dbg> lorawan: lorawan_set_region: Selected region 8
[00:00:03.519,000] <dbg> lorawan: lorawan_start: LoRaMAC Initialized
[00:00:03.523,000] <dbg> lorawan_nvm: lorawan_nvm_data_restore: Restoring LoRaWAN settings
[00:00:03.524,000] <dbg> lorawan_nvm: on_setting_loaded: Key: ClassB
[00:00:03.525,000] <dbg> lorawan_nvm: on_setting_loaded: Key: RegionGroup2
[00:00:03.525,000] <dbg> lorawan_nvm: on_setting_loaded: Key: RegionGroup1
[00:00:03.526,000] <dbg> lorawan_nvm: on_setting_loaded: Key: SecureElement
[00:00:03.526,000] <dbg> lorawan_nvm: on_setting_loaded: Key: MacGroup2
[00:00:03.527,000] <dbg> lorawan_nvm: on_setting_loaded: Key: MacGroup1
[00:00:03.527,000] <dbg> lorawan_nvm: on_setting_loaded: Key: Crypto
[00:00:03.527,000] <dbg> lorawan_nvm: lorawan_nvm_data_restore: Crypto version: 16777472, DevNonce: 7, JoinNonce: 5
[00:00:03.533,000] <dbg> lorawan_nvm: lorawan_nvm_data_restore: LoRaWAN context restored
[00:00:03.533,000] <inf> LorawanCommunication: Joining network over OTAA
[00:00:03.544,000] <dbg> lorawan: lorawan_join: Network join request sent!
[00:00:09.036,000] <dbg> lorawan: mlme_confirm_handler: Received MlmeConfirm (for MlmeRequest 1)
[00:00:09.036,000] <inf> lorawan: Joined network! DevAddr: 260b528a
[00:00:09.039,000] <dbg> lorawan_nvm: lorawan_nvm_save_settings: Saving LoRaWAN settings
[00:00:09.039,000] <dbg> lorawan_nvm: lorawan_nvm_save_settings: Crypto version: 16777472, DevNonce: 8, JoinNonce: 6
[00:00:09.039,000] <dbg> lorawan_nvm: lorawan_nvm_save_settings: Saving configuration lorawan/nvm/Crypto
[00:00:09.042,000] <dbg> lorawan_nvm: lorawan_nvm_save_settings: Saving configuration lorawan/nvm/MacGroup1
[00:00:09.044,000] <dbg> lorawan_nvm: lorawan_nvm_save_settings: Saving configuration lorawan/nvm/MacGroup2
[00:00:09.051,000] <dbg> lorawan_nvm: lorawan_nvm_save_settings: Saving configuration lorawan/nvm/SecureElement
[00:00:09.058,000] <dbg> lorawan_nvm: lorawan_nvm_save_settings: Saving configuration lorawan/nvm/RegionGroup1
[00:00:09.060,000] <inf> AppLorawan: New Datarate: DR_0, Max Payload 11
[00:00:09.060,000] <inf> lorawan: Datarate changed: DR_0
[00:00:12.358,000] <inf> App: Sending temperature 1046313
[00:00:12.358,000] <inf> LorawanCommunication: We try to send 0, Max payload 11
[00:00:13.742,000] <dbg> lorawan: mcps_indication_handler: Received McpsIndication 0
[00:00:13.742,000] <inf> AppLorawan: New Datarate: DR_2, Max Payload 125
[00:00:13.742,000] <inf> lorawan: Datarate changed: DR_2
[00:00:13.742,000] <inf> AppLorawan: Port 0, Pending 0, RSSI -42dB, SNR 7dBm
[00:00:13.742,000] <dbg> lorawan: mcps_confirm_handler: Received McpsConfirm (for McpsRequest 1)
[00:00:13.742,000] <dbg> lorawan: mcps_confirm_handler: McpsRequest success!
[00:00:13.744,000] <dbg> lorawan_nvm: lorawan_nvm_save_settings: Saving LoRaWAN settings
[00:00:13.744,000] <dbg> lorawan_nvm: lorawan_nvm_save_settings: Crypto version: 16777472, DevNonce: 8, JoinNonce: 6
[00:00:13.745,000] <dbg> lorawan_nvm: lorawan_nvm_save_settings: Saving configuration lorawan/nvm/Crypto
[00:00:13.748,000] <dbg> lorawan_nvm: lorawan_nvm_save_settings: Saving configuration lorawan/nvm/MacGroup1
[00:00:13.779,000] <dbg> lorawan_nvm: lorawan_nvm_save_settings: Saving configuration lorawan/nvm/MacGroup2
[00:00:13.785,000] <dbg> lorawan_nvm: lorawan_nvm_save_settings: Saving configuration lorawan/nvm/RegionGroup1
[00:00:13.786,000] <inf> App: Temperature message send succesfully
[00:00:13.786,000] <dbg> AppLorawan: SendDataLoggingBuffers: Max message size = 123
[00:00:13.786,000] <dbg> AppLorawan: SendDataLoggingBuffers: data size = 12
[00:00:13.786,000] <dbg> AppLorawan: SendDataLoggingBuffers: Message size = 12
[00:00:13.787,000] <inf> LorawanCommunication: We try to send 12, Max payload 123
[00:00:17.600,000] <dbg> lorawan: mcps_confirm_handler: Received McpsConfirm (for McpsRequest 1)
[00:00:17.600,000] <err> lorawan: McpsRequest failed : Rx 2 timeout
[00:00:17.603,000] <dbg> lorawan_nvm: lorawan_nvm_save_settings: Saving LoRaWAN settings
[00:00:17.603,000] <dbg> lorawan_nvm: lorawan_nvm_save_settings: Crypto version: 16777472, DevNonce: 8, JoinNonce: 6
[00:00:17.603,000] <dbg> lorawan_nvm: lorawan_nvm_save_settings: Saving configuration lorawan/nvm/Crypto
[00:00:17.605,000] <dbg> lorawan_nvm: lorawan_nvm_save_settings: Saving configuration lorawan/nvm/MacGroup1
[00:00:17.608,000] <dbg> lorawan_nvm: lorawan_nvm_save_settings: Saving configuration lorawan/nvm/RegionGroup1
[00:00:17.609,000] <wrn> LorawanCommunication: lorawan_send failed: -116. Continuing...
[00:00:21.343,000] <dbg> lorawan: mcps_confirm_handler: Received McpsConfirm (for McpsRequest 1)
[00:00:21.343,000] <err> lorawan: McpsRequest failed : Rx 2 timeout
[00:00:21.346,000] <dbg> lorawan_nvm: lorawan_nvm_save_settings: Saving LoRaWAN settings
[00:00:21.346,000] <dbg> lorawan_nvm: lorawan_nvm_save_settings: Crypto version: 16777472, DevNonce: 8, JoinNonce: 6
[00:00:21.346,000] <dbg> lorawan_nvm: lorawan_nvm_save_settings: Saving configuration lorawan/nvm/Crypto
[00:00:21.349,000] <dbg> lorawan_nvm: lorawan_nvm_save_settings: Saving configuration lorawan/nvm/MacGroup1
[00:00:21.351,000] <dbg> lorawan_nvm: lorawan_nvm_save_settings: Saving configuration lorawan/nvm/RegionGroup1
[00:00:21.353,000] <wrn> LorawanCommunication: lorawan_send failed: -116. Continuing...
[00:00:21.353,000] <err> LorawanCommunication: Failed to send lorawan message after 2 tries
[00:00:21.353,000] <inf> AppLorawan: Failed to send message we re-integrate the data into the LIFO

```

hallard, 2024-09-23:
@Martdur, interesting how dit you set FSB on zephyr? Do you hard code frequencies or use an hidden API call?

Martdur, 2024-09-23:
> @Martdur, interesting how did you set FSB on zephyr? Do you hard code frequencies or use an hidden API call? Anyway, thanks for follow up

My gateway only support the first frequency bands so I use : 
```
    uint16_t channelMask[6]={0};
    channelMask[0]=0xffff;
    lorawan.SetLorawanChannelMask(
        channelMask, 6);
```

There is a lot of magic number but it just for the sake of testing.

hallard, 2024-09-23:
Thanks,

No magic, same usage in mbed-os, but this one has some quite good [explanations](https://github.com/ARMmbed/mbed-os/blob/master/connectivity/lorawan/FSB_Usage.txt)

For US with your case it is 

```json
    "lora.fsb-mask": "{0x00FF, 0x0000, 0x0000, 0x0000, 0x0001}"
```
for my case, US with TTN (second frequencies) I'm using 
```json
    "lora.fsb-mask": "{0xFF00, 0x0000, 0x0000, 0x0000, 0x0002}"
```

Not sure how to translate my case for Zephyr, notation looks slightly different but could be something like that

```cpp
channelMask[1]=0xffff;
```

Martdur, 2024-09-23:
Ok I've found something interesting.
When the rx1 datarate is below rx10 the transmissions is working fine otherwise I get the rx2 timeout.
So setting the "Rx1 data channel offset" to 3 (on TTN) improve the behaviour but don't fix the issues.
see : 
[image]

Martdur, 2024-09-23:
And yes your right my mask should be : 
`uint16_t channelMask[6]= {0x00FF, 0x0000, 0x0000, 0x0000, 0x0001,0x0000}`

I've found out that I'm never reaching DR4 when on adaptative data-rate, when I forced it to DR4 and above I get timeout error.
Maybe is due to the gateway it self (I'll try to contact the constructor tomorow).

For the moment setting the rx1 channel offset to 3 fix the issue since it never reach DR4. But it lack some informations to validate the current behaviour.

closed 2024-12-07 by github-actions[bot]

imanpakii, 2024-12-24:
Is there any update on this issue? I have the same problem.

Martdur, 2025-02-19:
I've made some testing on a Milesight UG65 and got the same result.

It's seems to be some timings issues since downstream data rate superior to DR10 (SF10) aren't received by the device (Rx2 timeout).

I've done some research on loramac node  GitHub, but didn't get any answers.

Has someone done further testing ? 

I think it would be nice to re-open the issue.

reissjason, 2025-02-24:
Depending on your system clock the RX windows may not be opening at the correct times.

I have had success modifying the following parameters in the board config to open RX windows earlier and keep open longer than the defaults that are set for very precise timing of the host MCU.

CONFIG_LORAWAN_SYSTEM_MAX_RX_ERROR=30
CONFIG_LORAWAN_SYSTEM_MIN_RX_SYMBOLS=10

LoRaMac-node has details of how these parameters affect the RX window timing.
https://github.com/Lora-net/LoRaMac-node/blob/dcbcfb329b4a343ab007bc19ac43a8dc952b3354/src/mac/region/Region.h#L966

lorawan_start configures MAX_RX_ERROR in mainline.
https://github.com/zephyrproject-rtos/zephyr/blob/ee8266797c176b67867c1012d441141885bc9e4c/subsys/lorawan/lorawan.c#L770

I added handling for MIN_RX_SYMBOLS in my testing.

```
mib_req.Type = MIB_MIN_RX_SYMBOLS;
mib_req.Param.MinRxSymbols = CONFIG_LORAWAN_SYSTEM_MIN_RX_SYMBOLS;
LoRaMacMibSetRequestConfirm(&mib_req);
```

reissjason, 2025-02-24:
Also add to MIN_RX_SYMBOLS to Kconfig
https://github.com/zephyrproject-rtos/zephyr/blob/ee8266797c176b67867c1012d441141885bc9e4c/subsys/lorawan/Kconfig

```
config LORAWAN_SYSTEM_MIN_RX_SYMBOLS
	int "LoRaWAN System Min Rx Symbols"
	default 6
	help
	  System Min Rx symbols error value in ms to be used by LoRaWAN stack
	  for calculating the RX1/RX2 window timeout.
```

Martdur, 2025-03-04:
After tweekings some clocks parameters I'm now able to comunicate with any datarate.

hallard, 2025-03-04:
> After tweekings some clocks parameters I'm now able to comunicate with any datarate.

Awesome, do you thinks it's possible to summarize changes made to fix this issue?

FARLY7TII, 2025-03-07:
I am also facing this same issue.

- nRF53-DK
- [Semtech sx1262mb2das LoRa shield](https://docs.zephyrproject.org/latest/boards/shields/semtech_sx1262mb2das/doc/index.html) 
- Running `lorawan/class_a` sample
- EU868
- Gateway running Chirpstack

[image]

I have only tried increasing both `CONFIG_LORAWAN_SYSTEM_MAX_RX_ERROR` and `CONFIG_LORAWAN_SYSTEM_MIN_RX_SYMBOLS` (using the code in https://github.com/zephyrproject-rtos/zephyr/issues/70889#issuecomment-2679181598) but no success.

Any ideas? I'm not exactly sure what the root cause is. I don't think it has been identified in this thread?

FARLY7TII, 2025-03-07:
@hallard I notice you mentioned in your case:

> * Zephyr + NRF52840 + semtech_sx1262mb2das shield + EU868 -> Works

This would roughly be the same setup as mines, but isn't working. Could you share if you made any special config different from the default values in the sample application?

Martdur, 2025-03-18:
@FARLY7TII  I don't think it is related to our issue. 
We both succeed to communicate for the EU868 frequency band.
Moreover you get both RX1 and RX2 timeout which isn't our case.

It looks like your server isn't sending the join request confirmation.

Martdur, 2025-04-03:
After quite some testing in different conditions we find out that a too large MAX_RX_TIMEOUT_ERROR can lead to unexpected behavior.
Please see [this](https://github.com/Lora-net/LoRaMac-node/issues/1637) for more informations

FARLY7TII, 2025-04-04:
> [@FARLY7TII](https://github.com/FARLY7TII) I don't think it is related to our issue. We both succeed to communicate for the EU868 frequency band. Moreover you get both RX1 and RX2 timeout which isn't our case.
> 
> It looks like your server isn't sending the join request confirmation

Hi @Martdur, it is perhaps related - our error messages are the same. The gateway is indeed receiving the JoinRequest message and sending the JoinAccept. However, the device is not seeing this JoinAccept message.

I did get the class_a sample working with Chirpstack and TheThingsStack gateways using:
Zephyr + **NRF52DK** + semtech_sx1262mb2das shield + EU868.

However, running the same project on nRF53-DK, nRF52840-DK, and nRF52833-DK did not work.

Very confused 😄

umamaheshrouthu, 2025-04-07:
@FARLY7TII 
may i know if are successful with nRF53-DK, nRF52840-DK, and nRF52833-DK boards ?

I am also having same issue.
working on latest Zephyr + NRF52DK + semtech_sx1262mb2das shield + EU868.

umamaheshrouthu, 2025-04-29:
the reason for this failures are that PSI clock frequencies are much higher than the mx-frequency ocnfigured for these SPI peripherals in the board.

reduce the spi frequency to a value lesser than 8Mhz using a overlay and then the lorawan join procedure will be successful.
it atleast worked for me for all nrf52840dk, nrf5340dk and nrf54l15dk

hallard, 2025-04-29:
Thansk, remind me some fixes in the begening of LoRa.

What is strange is that Semtech says max SPI Frequency is 10MHz and as far as I remember Zephyr is setup to 8Mhz no?

What speed did you setup 4mHz or 6Mhz?

[image]

cfunes-pragma, 2025-04-29:
> Hi I'm facing the same issue with RAK3172 module in AUS915 band. I'm running `class_a` example: `west build -p always -b rak3172 -d build_rak ./samples/subsys/lorawan/class_a`
> 
> And this is the log:
> 
> *** Booting Zephyr OS build v4.1.0-2923-gb469fc188b4f ***
> [00:00:00.088,000] <inf> lorawan_class_a: Joining network over OTAA
> [00:00:06.578,000] <err> lorawan: MlmeConfirm failed : Rx 2 timeout
> [00:00:08.182,000] <err> lorawan_class_a: lorawan_join_network failed: -116
> [00:00:08.182,000] <inf> lorawan_class_a: Sending data...
> [00:00:08.182,000] <err> lorawan: LoRaWAN Send failed: No network joined
> [00:00:08.182,000] <err> lorawan_class_a: lorawan_send failed: -128
> [00:00:08.182,000] <inf> lorawan_class_a: Data sent!
> [00:00:18.182,000] <err> lorawan: LoRaWAN Send failed: No network joined
> [00:00:18.182,000] <err> lorawan_class_a: lorawan_send failed: -128
> [00:00:18.182,000] <inf> lorawan_class_a: Data sent!

Ok a friend of mine found the issue and it was basically that my gateway is an 8channels GW and this stack enables 72 channels on AU915 freq. So that is why it can never connects. Is there a way of setting up which channels to enable on the device to match gateway's configuration? if not, it could be a nice feature to add.

reissjason, 2025-04-29:
https://docs.zephyrproject.org/latest/doxygen/html/group__lorawan__api.html

```
uint16_t ch_mask[LORAWAN_CHANNELS_MASK_SIZE_US915] = { 0x00ff, 0, 0, 0, 0x0001 };
lorawan_set_channels_mask(ch_mask, LORAWAN_CHANNELS_MASK_SIZE_US915);
```

chrismoos, 2025-09-26:
I've been experiencing a similar issue, using a XIAO NRF52840 with the Wio-SX1262 (for XIAO) module. I have a SX1302 based gateway and receive the `JoinRequest`, but the NRF52840 doesn't seem to get the `JoinAccept`. I'm using the Class A example in Zephyr.

After trying everything to get it to work, I decided to try swapping the MCU and, using the same code, hooked up the same Wio-SX1262 to a RP2040. It worked fine! 

After this I decided to look at the SPI signals on an oscilloscope to compare. SCK didn't look great at 4Mhz on the NRF so I decided to go as low as possible and went down to **125Khz**. After doing this I was able to join the network and send data.

Not sure what the underlying reason is here but the only thing so far that makes it work is SPI @ 125Khz. Maybe someone smarter than me has an idea why.