zephyrproject-rtos/zephyr #70899: STM32F072B-DISCO USART setup error -ENOTSUP?

Spago123, 2024-03-29:
**USART setup**
Hello, I am trying to setup the USART3 channel on the STM32F072B-DISCO channel. I am able to successfully retrieve the setup infos from the overlay.file using uart_config_get() function. But I am unable to set the callback function using uart_callback_set() function, I am getting -ENOTSUP error.

**Possible problems**
I have looked into the uart.h file and thr CONFIG_UART_ASYNC_API is not defined so the program does not get into the IF part and that is the possible reason why I am getting the error :
static inline int uart_callback_set(const struct device *dev,
				    uart_callback_t callback,
				    void *user_data)
{
#ifdef CONFIG_UART_ASYNC_API
	const struct uart_driver_api *api =
			(const struct uart_driver_api *)dev->api;

	if (api->callback_set == NULL) {
		return -ENOSYS;
	}

	return api->callback_set(dev, callback, user_data);
#else
	ARG_UNUSED(dev);
	ARG_UNUSED(callback);
	ARG_UNUSED(user_data);
	return -ENOTSUP;
#endif
}

**prj.conf**
Here is my prj.conf file**
CONFIG_SERIAL=y
CONFIG_UART_ASYNC_API=y
CONFIG_UART_USE_RUNTIME_CONFIGURE=y
CONFIG_UART_INTERRUPT_DRIVEN=y

**app.overlay**
Here is my app.overlay:
&usart3 {
	pinctrl-0 = <&usart3_tx_pc4 &usart3_rx_pc11>;
	pinctrl-names = "default";
	current-speed = <9600>;
	status = "okay";
};

**main.c**
This is the main.c file
/*
 * Copyright (c) 2016 Open-RnD Sp. z o.o.
 * Copyright (c) 2020 Nordic Semiconductor ASA
 *
 * SPDX-License-Identifier: Apache-2.0
 */

#include <zephyr/kernel.h>
#include <zephyr/device.h>
#include <zephyr/drivers/gpio.h>
#include <zephyr/sys/util.h>
#include <zephyr/sys/printk.h>
#include <inttypes.h>
#include <zephyr/drivers/uart.h>

#define SLEEP_TIME_MS	1
#define RX_TIMEOUT 10000
#define CONFIG_UART_ASYNC_API 1
/*
 * Get button configuration from the devicetree sw0 alias. This is mandatory.
 */
#define SW0_NODE	DT_ALIAS(sw0)
#if !DT_NODE_HAS_STATUS(SW0_NODE, okay)
#error "Unsupported board: sw0 devicetree alias is not defined"
#endif

const struct device *uart4 = DEVICE_DT_GET(DT_NODELABEL(usart3));

static const struct gpio_dt_spec button = GPIO_DT_SPEC_GET_OR(SW0_NODE, gpios,
							      {0});
static struct gpio_callback button_cb_data;

struct uart_config konf;

/*
 * The led0 devicetree alias is optional. If present, we'll use it
 * to turn on the LED whenever the button is pressed.
 */
static struct gpio_dt_spec led = GPIO_DT_SPEC_GET_OR(DT_ALIAS(led5), gpios,
						     {0});

void button_pressed(const struct device *dev, struct gpio_callback *cb,
		    uint32_t pins)
{
	printk("Button pressed at %" PRIu32 "\n", k_cycle_get_32());
}

static  void uart_cb(const struct device *dev, struct uart_event *evt, void *user_data)
{
	printk("Transmit completed\n");
	

}

int main(void)
{
	int ret;

	if (!gpio_is_ready_dt(&button)) {
		printk("Error: button device %s is not ready\n",
		       button.port->name);
		return 0;
	}

	ret = gpio_pin_configure_dt(&button, GPIO_INPUT);
	if (ret != 0) {
		printk("Error %d: failed to configure %s pin %d\n",
		       ret, button.port->name, button.pin);
		return 0;
	}

	ret = gpio_pin_interrupt_configure_dt(&button,
					      GPIO_INT_EDGE_TO_ACTIVE);
	if (ret != 0) {
		printk("Error %d: failed to configure interrupt on %s pin %d\n",
			ret, button.port->name, button.pin);
		return 0;
	}

	gpio_init_callback(&button_cb_data, button_pressed, BIT(button.pin));
	gpio_add_callback(button.port, &button_cb_data);
	printk("Press the button\n");

	uint8_t arr[5] = {1,2,3,4,5};
	arr[0] = 3;
	uint8_t* point;
	int err = uart_callback_set(uart4, uart_cb, (void*)point);
	if(err!=0) printk("ERROR SETTING: %d\n", err);

	k_sleep(K_MSEC(1000));

	err = uart_config_get(uart4, &konf);
	if(err!=0) printk("ERROR CONFIG: %d\n", err);

	printk("Set up button at %s pin %d\n", button.port->name, button.pin);

	if (led.port && !gpio_is_ready_dt(&led)) {
		printk("Error %d: LED device %s is not ready; ignoring it\n",
		       ret, led.port->name);
		led.port = NULL;
	}
	if (led.port) {
		ret = gpio_pin_configure_dt(&led, GPIO_OUTPUT);
		if (ret != 0) {
			printk("Error %d: failed to configure LED device %s pin %d\n",
			       ret, led.port->name, led.pin);
			led.port = NULL;
		} else {
			printk("Set up LED at %s pin %d\n", led.port->name, led.pin);
		}
	}

	printk("Press the button\n");
	if (led.port) {
		while (1) {

			int val = uart_tx(uart4, arr, 5, 1000);
			if (val!=0) printk("ERROR! ,%d\n", val);
			for(uint8_t i=0; i<5; i++) arr[i]=arr[i]+1;
			k_sleep(K_MSEC(1000));
		}
	}
	return 0;
}

erwango, 2024-04-02:
Using callback using UART_ASYNC_API requires enabling DMA, for which you're missing configuration. For instance:
```
&uart4 {
	dmas = <&dma2 3 2 (STM32_DMA_PERIPH_TX | STM32_DMA_PRIORITY_HIGH)>,
		<&dma2 5 2 (STM32_DMA_PERIPH_RX | STM32_DMA_PRIORITY_HIGH)>;
	dma-names = "tx", "rx";
};
```

See `tests/drivers/uart/uart_async_api/boards` for DMA configuration examples

erwango, 2024-04-02:
I'm closing this issue. Please open a discussion if you're facing issues to provide a working configuration

closed 2024-04-02 by erwango