zephyrproject-rtos/zephyr #70906: tests/subsys/llext/simple fails with stm32h7

FRASTM, 2024-03-29:
**Describe the bug**
tests/subsys/llext/simple/  test execution fails with stm32h7 board on item llext.simple.readonly 

Please also mention any information which could help others to understand
the problem you're facing:
 - stm32h7b3i_dk or nucleo_h743zi
 -  regression after PR #70100

**To Reproduce**
Steps to reproduce the behavior:
1.west build -p auto -b stm32h7b3i_dk tests/subsys/llext/simple/ -- -DCONFIG_ARM_MPU=n -DCONFIG_LLEXT_STORAGE_WRITABLE=n
2. west flash
3. See error

**Expected behavior**
```
------ TESTSUITE SUMMARY START ------                                                                                                                   
                                                                                                                                                        
SUITE PASS - 100.00% [llext]: pass = 3, fail = 0, skip = 0, total = 3 duration = 1.130 seconds                                                          
 - PASS - [llext.test_load_unload_hello_world] duration = 0.425 seconds                                                                                 
 - PASS - [llext.test_load_unload_logging] duration = 0.704 seconds                                                                                     
 - PASS - [llext.test_printk_exported] duration = 0.001 seconds                                                                                         
                                                                                                                                                        
------ TESTSUITE SUMMARY END ------  
```

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

**Logs and console output**
```

===================================================================                                                                                     
START - test_load_unload_relative_jump                                                                                                                  
D: Loading relocatable or shared elf                                                                                                                    
D: Finding ELF tables...                 
...                                                                                                               
I: Loaded extension relative_jump                                                                                                                       
E: ***** MPU FAULT *****                                                                                                                                
E:   Instruction Access Violation                                                                                                                       
E: r0/a1:  0x00000003  r1/a2:  0x0800a064  r2/a3:  0x24001758                                                                                           
E: r3/a4:  0x00000002 r12/ip:  0x240014e0 r14/lr:  0x24001797                                                                                           
E:  xpsr:  0x80000000                                                                                                                                   
E: Faulting instruction address (r15/pc): 0xbf00bd80                                                                                                    
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0                                                                                                     
E: Current thread: 0x240000d0 (test_load_unload_relative_jump)                                                                                          
E: Halting system         

```

**Environment (please complete the following information):**
 - OS: ubuntu
 - Toolchain  SDK 0.16.3
 - Commit SHA dbfc1aaec697b78573c18d83fd40ba66ff63c0b3

FRASTM, 2024-03-29:
It appears that increasing the CONFIG_ZTEST_STACK_SIZE to 8192 is a possible WA

FRASTM, 2024-03-29:
CONFIG_ZTEST_STACK_SIZE=7168  (??) can fix both error for the stm32h7 serie

FRASTM, 2024-04-02:
has this issue some link with https://github.com/zephyrproject-rtos/zephyr/issues/70906 ?

pillo79, 2024-04-02:
@FRASTM in the last post you linked this issue with itself (and made me click at least 3 times before realizing I need coffee :rofl:). What other one were you referring to? 

I will try to replicate here in the next few days, but can you clarify the current status? Does increasing the stack reliably fix all issues for you? Thanks!

pillo79, 2024-04-03:
@FRASTM I was able to reproduce those issues on the STM32H747, they changed with different stack sizes but couldn't find a number that actually "fixed" it. With a debugger I could see the CPU was _not_ executing the expected code (which otherwise looked fine when probed). The icache flush in the PR above does reliably fix the issue for me.

FRASTM, 2024-04-03:
> @FRASTM I was able to reproduce those issues on the STM32H747, they changed with different stack sizes but couldn't find a number that actually "fixed" it. With a debugger I could see the CPU was [...]

Oh yes, sorry, for sure it has a link with itself but the issue Iwas https://github.com/zephyrproject-rtos/zephyr/issues/70317  indeed

FRASTM, 2024-04-03:
> @FRASTM in the last post you linked this issue with itself (and made me click at least 3 times before realizing I need coffee 🤣). What other one were you referring to?
> 
> I will try to replicate here in the next few days, but can you clarify the current status? Does increasing the stack reliably fix all issues for you? Is "7168" a magic number or any large one will do? Thanks!

Yes, I found that more than 8192 was too big for a stm32 soc and 4096 too low. 
Going to check with https://github.com/zephyrproject-rtos/zephyr/pull/71056/files

PR #71056 (2024-04-05): llext: invalidate instruction cache after loading extension
In addition to flushing the dcache, to ensure that the new code is actually read by the CPU when loading an extension, the icache must be invalidated as well.
Without this, some tests are failing in hardware because the CPU is executing stale code from its instruction cache.

Fixes #70906 on `arduino_giga_r1/stm32h747xx/m7`.
commit 4867c56a (2024-04-05): llext: invalidate instruction cache after loading extension

In addition to flushing the dcache, to ensure that the new code is
actually read by the CPU when loading an extension, the icache must
be invalidated as well.

Without this, some tests are failing in hardware because the CPU is
executing stale code from its instruction cache.

Fixes #70906 on arduino_giga_r1/stm32h747xx/m7.
closed 2024-04-05 by carlescufi