zephyrproject-rtos/zephyr #70915: Take semaphore when fatal error occur in coredump with CONFIG_DEBUG_COREDUMP_BACKEND_FLASH_PARTITION on

liutingyue, 2024-03-30:
When enable CONFIG_COREDUMP and use flash partition as backend, the flash backend output call partition_open() and take semaphore .

If  CONFIG_ASSERT enabled，the assertion log is shown as follows:

ASSERTION FAIL [((arch_is_in_isr() == 0) || ((timeout).ticks == (((k_timeout_t) {0})).ticks))] @ WEST_TOPDIR/zephyr/kernel/sem.c:136

```
/**
 * @brief Open the flash partition.
 *
 * @return Same as flash_area_open().
 */
static int partition_open(void)
{
	int ret;

	(void)k_sem_take(&flash_sem, FLASH_BACKEND_SEM_TIMEOUT);

	ret = flash_area_open(FLASH_PARTITION_ID, &backend_ctx.flash_area);
	if (ret != 0) {
		LOG_ERR("Error opening flash partition for coredump!");

		backend_ctx.flash_area = NULL;
		k_sem_give(&flash_sem);
	}

	return ret;
}
```

jennifervlaar, 2024-06-11:
I have recently encountered this issue while working with Zephyr, is there an ETA of when the bug will be fixed or a workaround that anyone has found?

liutingyue, 2024-06-12:
The official team does not seem to have any plans to fix the problem in a short time. My personal suggestion is that the conditions for determining whether it is in ISR can be processed separately.

mrkhldn, 2024-06-12:
There's some discussion of the same issue here, https://github.com/zephyrproject-rtos/zephyr/issues/59116, but I don't think we arrived at a good answer.

arbrauns, 2024-08-12:
Hasn't fixed itself.

dawidwojtas-mudita, 2024-10-14:
I've also encountered this issue when `CONFIG_DEBUG_COREDUMP_BACKEND_FLASH_PARTITION=y` and `CONFIG_ASSERT=y` are enabled. 
To make it work I need to disable `CONFIG_ASSERT` but I don't think this is the best solution, only a workaround for now.

closed 2024-12-29 by github-actions[bot]