zephyrproject-rtos/zephyr #70920: custom mpu regions for imxrt

GrixaYrev, 2024-03-30:
File `zephyr\soc\nxp\imxrt\mpu_regions.c` is included  to sources unconditionally. So i can't use SRAM size more than 16M and less than 32M, because error in `zephyr\include\zephyr\arch\arm\cortex_m\arm_mpu_mem_cfg.h:72` is rised: "_Unsupported sram size configuration_".

I think additional Kconfig option need for exclude `zephyr\soc\nxp\imxrt\mpu_regions.c` from sources.

I use custom board based on iMX.RT1052 with external SDRAM. Part SDRAM used as FLASH (for bootloader and application), other space as RAM.

Version of Zephyr – 3.6.99 (SHA 0714052c74bfe4e74ae6fca705da4d7d66c92681)

prj.conf:
```
CONFIG_FLASH_SIZE=11264 
CONFIG_FLASH_BASE_ADDRESS=0x80100000
CONFIG_SRAM_SIZE=20480
CONFIG_SRAM_BASE_ADDRESS=0x80C00000
```

GrixaYrev, 2024-04-10:
I suggest to add configuration option `NXP_IMX_RT_USER_MPU_REGIONS` to **zephyr\soc\nxp\imxrt\Kconfig**.
I can prepare PR.

GrixaYrev, 2024-04-12:
@teburd, @dleach02, what do you think about my suggestion?

GrixaYrev, 2024-07-07:
@dleach02, no, first two items of `mpu_regions[]` is included unconditional:
```
	/* Region 0 */
	MPU_REGION_ENTRY("FLASH_0",
			 CONFIG_FLASH_BASE_ADDRESS,
			 REGION_FLASH_ATTR(REGION_FLASH_SIZE)),
	/* Region 1 */
	MPU_REGION_ENTRY("SRAM_0",
			 CONFIG_SRAM_BASE_ADDRESS,
			 REGION_RAM_ATTR(REGION_SRAM_SIZE)),
```

If i set `CONFIG_FLASH_SIZE=11264` and `CONFIG_SRAM_SIZE=20480` i have error message "Unsupported sram size configuration".

dleach02, 2024-07-07:
Please prepare a PR so I can route it appropriately for internal review.

PR #75604 (2024-11-07): soc: nxp: imxrt: exclude mpu_regions.c when ARM_MPU disabled
Option NXP_IMX_RT_USER_MPU_REGIONS exclude soc file mpu_regions.c and allow user define custom mpu_regions in custom file. 

Fixes #70920

For test this PR i use:

- file prj.conf:
```
...
CONFIG_FLASH_SIZE=5120 
CONFIG_FLASH_BASE_ADDRESS=0x80100000
CONFIG_SRAM_SIZE=26624
CONFIG_SRAM_BASE_ADDRESS=0x80600000
...
```

- file with custom MPU regions:
```
#include <zephyr/devicetree.h>
#include <zephyr/arch/arm/mpu/arm_mpu.h>

#define REGION_RAM_EXEC_ATTR(size) \
{ \
	(NORMAL_OUTER_INNER_WRITE_BACK_WRITE_READ_ALLOCATE_NON_SHAREABLE | \
	 [... cut]
commit 91a59e7e (2024-11-07): soc: nxp: imxrt: exclude mpu_regions.c when ARM_MPU disabled

When option ARM_MPU is disabled exclude soc\nxp\imxrt\mpu_regions.c.
It is needed to remove constraints of SRAM and FLASH size.
Fixes #70920
closed 2024-11-07 by dkalowsk