zephyrproject-rtos/zephyr #70931: Time become incorrect when enable bluetooth of  board nucleo_wba52cg.

2024markLuo, 2024-04-01:
**Is your feature request related to a problem? Please describe.**
I use board nucleo_wba52cg, When I call bt_enable(),  then all k_msleep() become not correct.
for example: k_msleep(2000), it was supposed to delay for 2 seconds, but in fact it delayed for 12 seconds.

**Describe the solution you'd like**
After enabling bluetooth on board nucleo_wba52cg, k_msleep can be accurate and systick can also be accurate.

**Describe alternatives you've considered**
1. bt_disable() always failed, cannot set bluetooth enter low power consumption.
2. When enable extended advertising, it will report stack overflow.
3. When use bt scan, the rssi  always 0.
4. Use BT_LE_ADV_OPT_USE_IDENTITY,  all devices mac address is 00:00:00:00:00:04, Not the only address we're looking for.

**Additional context**

carlescufi, 2024-04-11:
@erwango any ideas?

asm5878, 2024-04-17:
Dear @2024markLuo, 
Just for info I have made a test with nucleo_wba55cg board and peripheral_hr.
I have used a GPIO to test a k_sleep behavior and it was as expected.

Here a snippet of while loop:
```
	while (1) {
		k_sleep(K_SECONDS(1));
		gpio_pin_configure_dt(&tp1, GPIO_OUTPUT_ACTIVE);
		/* Heartrate measurements simulation */
		hrs_notify();
		/* Battery level simulation */
		bas_notify();
		gpio_pin_set_dt(&tp1, 0);
		gpio_pin_configure_dt(&tp1, GPIO_DISCONNECTED);
	}
```

Right now I haven't in my hands any nucleo_wba52cg to do the same test. 
Maybe a mismatch between nucleo_wba52cg and nucleo_wba55cg clock configuration?

Thanks,
Alex

erwango, 2024-04-22:
@2024markLuo Did you try to use clock configuration from nucleo_wba55 to nucleo_wba52?
Note that currently BLE support on nucleo_wba52 is not enabled in tree, so this is not technically a "bug".

erwango, 2024-04-22:
Note that currently BLE support on nucleo_wba52 is not enabled in tree, so this is not technically a "bug".

Moving to discussion.