zephyrproject-rtos/zephyr #70933: The rssi in the bluetooth scan results is always 0 on stm32wba52cg

2024markLuo, 2024-04-01:
This issue list is only for bugs in the main Zephyr code base

**Describe the bug**
bt_le_scan_cb_register(&scan_cb);
static struct bt_le_scan_cb scan_cb = {
  .recv = scan_recv,
};
static void scan_recv(const struct bt_le_scan_recv_info *info,
          struct net_buf_simple *ad)

info->rssi always 0, test on board nucleo_wba52cg.

Please also mention any information which could help others to understand
the problem you're facing:
 - Target platform is stm32wba52cg

**To Reproduce**
Steps to reproduce the behavior:
1. cd samples/bluetooth/central
2. west build -b nucleo_wba52cg
3. west flash
4. See the log RSSI, it always 0.

**Expected behavior**
The rssi should be correct.

**Impact**

**Logs and console output**
If applicable, add console logs or other types of debug information
e.g Wireshark capture or Logic analyzer capture (upload in zip archive).
copy-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue. (if unable to obtain text log, add a screenshot)

**Environment (please complete the following information):**
 - OS: ubuntu22.04
 - Toolchain: zephyr-sdk-0.16.3
 - 2ca179c96e3eca078ed13db05b5f9ee7338840da

**Additional context**
Add any other context that could be relevant to your issue, such as pin setting,
target configuration, ...

asm5878, 2024-05-02:
Hi @2024markLuo,

Even in this case I have tested in the context of nucleo_wba55cg.
"samples/bluetooth/central" has been used.

Please have a look at   bt_le_scan_start function at this snippet:

```
static void start_scan(void)
{
	int err;

	/* This demo doesn't require active scan */
	err = bt_le_scan_start(BT_LE_SCAN_PASSIVE, device_found);
	if (err) {
		printk("Scanning failed to start (err %d)\n", err);
		return;
	}

	printk("Scanning successfully started\n");
}
```
A callback function "device_found" is directly called for every found device.
From a BLE standard point of view  a meta LE event is generated (BT_HCI_EVT_LE_ADVERTISING_REPORT, LE sub event code 0x02).  See Bluetooth spec. v.5.4 [Vol 4, Part E, 7.7.65.2].

The mechanism is working fine and RSSI is properly reported.

I suggest this sample with this extra configuration:

CONFIG_BT_HCI_TX_STACK_SIZE=2048
CONFIG_BT_HCI_TX_STACK_SIZE_WITH_PROMPT=y

Definitively no need to use the  bt_le_scan_cb_register function and modify the source code of the sample.

Thanks,
Alex

erwango, 2024-06-26:
@2024markLuo Is the answer fine for you ?

asm5878, 2024-07-03:
Hello @2024markLuo,

Could you please let us know if this solution is suitable for you ?

erwango, 2024-07-30:
Closing as fixed in V3.7.0. Please reopen if required.

closed 2024-07-30 by erwango