zephyrproject-rtos/zephyr #70935: Stack overflow when start bluetooth extended adv on board nucleo_wba52cg

2024markLuo, 2024-04-01:
This issue list is only for bugs in the main Zephyr code base

**Describe the bug**
00> [00:00:12.339,000] <inf> bt_hci_core: Identity: 00:00:00:00:00:40 (public)
00> [00:00:12.339,000] <inf> bt_hci_core: HCI: version 5.4 (0x0d) revision 0x800b, manufacturer 0x0030
00> [00:00:12.339,000] <inf> bt_hci_core: LMP: version 5.4 (0x0d) subver 0x500b
00> [00:00:13.040,000] <err> os: ***** USAGE FAULT *****
00> [00:00:13.040,000] <err> os:   Stack overflow (context area not valid)
00> [00:00:13.040,000] <err> os: r0/a1:  0x2000edd8  r1/a2:  0x2000eed7  r2/a3:  0xffffffff
00> [00:00:13.040,000] <err> os: r3/a4:  0x00000000 r12/ip:  0x2000edef r14/lr:  0x0801c77b
00> [00:00:13.040,000] <err> os:  xpsr:  0x81000200
00> [00:00:13.040,000] <err> os: Faulting instruction address (r15/pc): 0x080386ba
00> [00:00:13.040,000] <err> os: >>> ZEPHYR FATAL ERROR 2: Stack overflow on CPU 0
00> [00:00:13.040,000] <err> os: Current thread: 0x20007eb8 (unknown)
00> [00:00:13.048,000] <err> os: Halting system

The error happened at link_layer:
/home/mluo/tools/zephyr-sdk-0.16.3/arm-zephyr-eabi/bin/arm-zephyr-eabi-addr2line -e build/zephyr/zephyr.elf 801c77b
C:\dev\STM32WBA\WBA_SOW05PatchV6\Firmware\Middlewares\ST\STM32_WPAN\link_layer\ll_cmd_lib\src\ble_controller\src\ll_intf.c:?

Please also mention any information which could help others to understand
the problem you're facing:
 - Target platform: stm32wba52cg.

**To Reproduce**
Steps to reproduce the behavior:
1. cd zephyr/samples/bluetooth/peripheral
2. add CONFIG_BT_EXT_ADV=y in prj.conf
3. west build -b nucleo_wba52cg
4. west flash
5. see log

**Expected behavior**
When we use bluetooth extended advertising, it can be ok.

**Impact**

**Logs and console output**
00> [00:00:12.339,000] <inf> bt_hci_core: Identity: 00:00:00:00:00:40 (public)
00> [00:00:12.339,000] <inf> bt_hci_core: HCI: version 5.4 (0x0d) revision 0x800b, manufacturer 0x0030
00> [00:00:12.339,000] <inf> bt_hci_core: LMP: version 5.4 (0x0d) subver 0x500b
00> [00:00:13.040,000] <err> os: ***** USAGE FAULT *****
00> [00:00:13.040,000] <err> os:   Stack overflow (context area not valid)
00> [00:00:13.040,000] <err> os: r0/a1:  0x2000edd8  r1/a2:  0x2000eed7  r2/a3:  0xffffffff
00> [00:00:13.040,000] <err> os: r3/a4:  0x00000000 r12/ip:  0x2000edef r14/lr:  0x0801c77b
00> [00:00:13.040,000] <err> os:  xpsr:  0x81000200
00> [00:00:13.040,000] <err> os: Faulting instruction address (r15/pc): 0x080386ba
00> [00:00:13.040,000] <err> os: >>> ZEPHYR FATAL ERROR 2: Stack overflow on CPU 0
00> [00:00:13.040,000] <err> os: Current thread: 0x20007eb8 (unknown)
00> [00:00:13.048,000] <err> os: Halting system

**Environment (please complete the following information):**
 - OS: Ubuntu22.04
 - Toolchain: zephyr-sdk-0.16.3
 - Commit: 2ca179c96e3eca078ed13db05b5f9ee7338840da

**Additional context**

pepe2k, 2024-05-27:
Hi @2024markLuo, I think your issue might be related to the one reported by me (https://github.com/zephyrproject-rtos/zephyr/issues/73345). Could you try the possible fix I mentioned there (increase HCI TX stack size)? Thanks!

erwango, 2024-05-27:
@asm5878 ^^

asm5878, 2024-06-13:
In latests updates we have introduced this fix:
In modules\hal\stm32\lib\stm32wba\hci\app_conf.h we need to be sure:

`#define RADIO_SW_LOW_INTR_PRIO              (14)           /* 2.4GHz RADIO low ISR `

In the next updates we will be sure to integrate this change:
`CONFIG_BT_HCI_TX_STACK_SIZE=2048`

With these changes problem is solved (tested with samples\bluetooth\extended_adv\advertiser and nucleo_wba55cg)

asm5878, 2024-06-27:
Dear All, 
After RC1 I have discovered a new bug for extended advertising feature.
So a new version of hal_stm32 (update in STM32WBA55) is required.
Also according the suppression of "BT TX", the new config to be modified is CONFIG_SYSTEM_WORKQUEUE_STACK_SIZE=2048 (change required to cover others BLE use case according  current STM32WBA55 HCI driver).

PR will be created to fix this problem.

asm5878, 2024-07-03:
Issue has been properly addressed in this PR  [zephyrproject-rtos/zephyr/pull/75091]
[](https://github.com/zephyrproject-rtos/zephyr/pull/75091)
Already merged.

closed 2024-07-03 by erwango