zephyrproject-rtos/zephyr #70966: doc: Documentation issue in 'services/debugging/gdbstub'

kwangbaeklee99, 2024-04-01:
**Describe the bug**

Is this feature, gdbstub, supported on Zephyr 3.6.99?
Since I'm not able to find a directory under "samples/subsys/debug/gdbstub".
Thank you,

Kevin

<< Please describe the issue here >>
<< You may also want to update the automatically generated issue title above. >>

**Environment**

* Page: `services/debugging/gdbstub`
* Version: main
* SHA-1: fb19d532edb177dda56fb0f37c89f361004f41c0

golowanow, 2024-04-01:
@kwangbaeklee99 

> Is this feature, gdbstub, supported on Zephyr 3.6.99? Since I'm not able to find a directory under "samples/subsys/debug/gdbstub"
> 

it is moved to `tests/subsys/debug/gdbstub`
https://github.com/zephyrproject-rtos/zephyr/blob/main/doc/services/debugging/gdbstub.rst#example

nashif, 2024-04-02:
@golowanow  will you submit a fix for the documentation?

golowanow, 2024-04-02:
> @golowanow will you submit a fix for the documentation?

the documentation page is already updated, unless there is some other problem there in regard of the move. 
@kwangbaeklee99 is your question answered ?

kartben, 2024-04-02:
> @kwangbaeklee99 
> 
> > Is this feature, gdbstub, supported on Zephyr 3.6.99? Since I'm not able to find a directory under "samples/subsys/debug/gdbstub"
> > 
> 
> it is moved to `tests/subsys/debug/gdbstub` with #63672 see also
> https://github.com/zephyrproject-rtos/zephyr/blob/main/doc/services/debugging/gdbstub.rst#example

That page you linked actually seems to have several stale references
https://github.com/search?q=repo%3Azephyrproject-rtos%2Fzephyr%20samples%2Fsubsys%2Fdebug%2Fgdbstub&type=code

golowanow, 2024-04-03:
> @golowanow That page you linked actually seems to have several stale references

right, I'll update it with some other issues now I see there, thank you.

closed 2024-04-11 by carlescufi