zephyrproject-rtos/zephyr #70967: How build cross-compile native_posix app

LingaoM, 2024-04-02:
Hi,

I what run Zephyr samples on Raspberry Pi 4B(Which is arm linux os),

How i can build some zephyr sample run directly run Raspberry Pi?

i'm test native_posix, but this board only generate host exe not arm linux.

aescolar, 2024-04-10:
This request was created in the context of building for native_posix/sim in one target (a development host) to then run the executable in a Linux embedded target.
This is indeed not supported. But it was never meant to be supported. 
It is actually very strongly discouraged.
native_{sim,posix} is meant as an aid to allow developers to run and test Zephyr in their development machine (or a CI machine). It is, in no way, meant to allow running Zephyr apps in an embedded Linux machine just like if it was running Zephyr. It's drivers and the way everything else is built do not attempt to provide a real time execution environment but a test environment.
Similarly there would be severe security concerns in doing so, as native_posix/sim will by default generate reproducible runs with the same random numbers, and therefore, for crypto reasons would be very problematic.
Therefore closing as it is not a intended use case.

(Note this does not preclude the possibility of "crosscompiling" in the sense of producing 32bit binaries in a 64 bit host to run locally in a multilib like setup)

closed 2024-04-10 by aescolar

Frank-Buss, 2024-08-09:
How would you compile it to 32 bit? I have a 64 ARM host and would like to compile it to a 32 bit binary. I tried it, but get an error:
```
frank@ubuntu-4gb-arm:~/zephyrproject/zephyr/samples/hello_world$ west build -b native_posix
-- west build: generating a build system
Loading Zephyr default modules (Zephyr base).
-- Application: /home/frank/zephyrproject/zephyr/samples/hello_world
-- CMake version: 3.28.3
-- Found Python3: /usr/bin/python3 (found suitable version "3.12.3", minimum required is "3.8") found components: Interpreter 
-- Cache files will be written to: /home/frank/.cache/zephyr
-- Zephyr version: 3.7.99 (/home/frank/zephyrproject/zephyr)
-- Found west (found suitable version "1.2.0", minimum required is "0.14.0")
-- Board: native_posix, qualifiers: native
-- Found toolchain: host (gcc/ld)
-- Found Dtc: /usr/bin/dtc (found suitable version "1.7.0", minimum required is "1.4.6") 
-- Found BOARD.dts: /home/frank/zephyrproject/zephyr/boards/native/native_posix/native_posix.dts
-- Generated zephyr.dts: /home/frank/zephyrproject/zephyr/samples/hello_world/build/zephyr/zephyr.dts
-- Generated devicetree_generated.h: /home/frank/zephyrproject/zephyr/samples/hello_world/build/zephyr/include/generated/zephyr/devicetree_generated.h
-- Including generated dts.cmake file: /home/frank/zephyrproject/zephyr/samples/hello_world/build/zephyr/dts.cmake
Parsing /home/frank/zephyrproject/zephyr/Kconfig
Loaded configuration '/home/frank/zephyrproject/zephyr/boards/native/native_posix/native_posix_defconfig'
Merged configuration '/home/frank/zephyrproject/zephyr/samples/hello_world/prj.conf'
Configuration saved to '/home/frank/zephyrproject/zephyr/samples/hello_world/build/zephyr/.config'
Kconfig header saved to '/home/frank/zephyrproject/zephyr/samples/hello_world/build/zephyr/include/generated/zephyr/autoconf.h'
-- Found GnuLd: /usr/bin/ld.bfd (found version "2.42") 
-- The C compiler identification is GNU 13.2.0
-- The CXX compiler identification is unknown
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/gcc
CMake Error at /home/frank/zephyrproject/zephyr/arch/posix/Linux.aarch64.cmake:30 (message):
  CONFIG_64BIT=n but this Aarch64 machine has a 64-bit userspace.

  If you were targeting native_sim, target native_sim/native/64 instead.

  Otherwise, be sure to define CONFIG_64BIT appropriately.

Call Stack (most recent call first):
  /home/frank/zephyrproject/zephyr/arch/posix/CMakeLists.txt:44 (include)

-- Configuring incomplete, errors occurred!
FATAL ERROR: command exited with status 1: /usr/bin/cmake -DWEST_PYTHON=/usr/bin/python3 -B/home/frank/zephyrproject/zephyr/samples/hello_world/build -GNinja -DBOARD=native_posix -S/home/frank/zephyrproject/zephyr/samples/hello_worl
```

aescolar, 2024-08-10:
@Frank-Buss are you asking on the context of running it as a 32 bit binary in the same host were you built it? (if so it may be easier to talk in discord in the #native-sim-posix channel)

Frank-Buss, 2024-08-10:
No, I have the same use case as LingaoM: we are using a Nordic chip for BLE, but we are transitioning to a Linux module. The Nordic BLE implementation is really nice, and works fine in my tests over HCI on a Linux host as well. I agree, there might be many problems when running Zephyr on Linux, but it is just for a proof of concept project. Later we might port it all to native Linux, even if bluez kinda sucks, but maybe I find a better raw HCI library. Meanwhile I got it to work with the help of the not merged PR.

In general I think Zephyr shouldn't artificially limit how things can be cross compiled. It should just state native_posix is not suited for production use, and then let the developers decide how to use it.