zephyrproject-rtos/zephyr #70978: Communicate with SD card connected to SPI using DMA

Paul-Avo, 2024-04-02:
**Is your enhancement proposal related to a problem? Please describe.**

I would like to continuously acquire data at a minimum frequency of 1.05 MHz. For the moment, I can reach this frequency but not continuously. Indeed, the acquisition of data in the RAM memory is carried out at a frequency slightly higher than 1.05 MHz, but the writing of the buffer (of 120kB) in the SD card is only carried out "in 0.3s" this which blocks the acquisition of data and I then no longer respect my specifications which require continuous acquisition.

My objective is therefore to be able to communicate with my SD card connected to SPI using DMA to make this writing time on the card invisible for the microprocessor. And it’s on this last point that I’m stuck. Indeed, I have not yet seen an example of such a project.

**Describe the solution you'd like**

I would like to create the program that will allow this DMA storage to be carried out. However, several questions came to me. First of all, is it possible to do DMA with SPI connected to an SD card (I think so)?

Then, I tried to find how the zephyr library communicates with the SD card, but I can't go further than the file: sdhc.h and the functions:

```c
__syscall int sdhc_request(const struct device *dev, struct sdhc_command *cmd,
struct sdhc_data *data);

static inline int z_impl_sdhc_request(const struct device *dev,
struct sdhc_command *cmd,
struct sdhc_data *data)
{
const struct sdhc_driver_api *api = (const struct sdhc_driver_api *)dev->api;

if (!api->request) {
return -ENOSYS;
}

return api->request(dev, cmd, data);
}
``` 

That seems to be the terminal functions that communicate with the SD card (with a __syscall). To see further I thought I could see the code by obtaining the sources after the pre-processor with the equivalent during the build  of instruction  gcc -E. But I don't know if this is the right strategy or if there is a similar command for zephyr. The goal in doing this was to see the SPI communications to be able to pass them to DMA.

**Describe alternatives you've considered**

Another alternative is not to look at how the Zephyr libraries work and to write to the SD card directly.

**Additional context**

So, if you know a function that achieves what I'm trying to achieve or if you know the protocols to be able to communicate with the SD card via SPI (header → data → footer), I am at your disposal for your help

**Card** : NucleoF411re

danieldegrasse, 2024-04-05:
SPI SD transfers are implemented in the SPI SDHC driver: https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/sdhc/sdhc_spi.c.

This driver uses the SPI API directly, so if you enable DMA with the SPI instance on your SOC, you should be able to read/write from the card using DMA without issues.

Paul-Avo, 2024-04-08:
Thank you very much for the response,

So, if I understood correctly, I just need to activate the dma on the card. So I'm wondering if the code in the following files is sufficient:

main.c : 

`#include <zephyr/drivers/dma.h>`

devicetree :

```
&spi1 {
	pinctrl-0 = <&spi1_nss_pa4 &spi1_sck_pa5
		     &spi1_miso_pa6 &spi1_mosi_pa7>;
	pinctrl-names = "default";
	status = "okay";
	cs-gpios = <&gpiob 6 GPIO_ACTIVE_LOW>;
	
	sdhc0: sdhc@0 {
                compatible = "zephyr,sdhc-spi-slot";
                reg = <0>;
                status = "okay";
                mmc {
                    compatible = "zephyr,sdmmc-disk";
                    status = "okay";
                };
                spi-max-frequency = <80000000>;
        };

	dmas = <&dma1 3 3 0x400 0x03>, <&dma1 2 3 0x400 0x03>;
    dma-names = "tx", "rx";

};

&dma1 {
    status = "okay";
};
```
Kconfig :

```
CONFIG_DMA=y
CONFIG_SPI_STM32_DMA=y
CONFIG_SPI=y
```

I ask this because, when I add:

`CONFIG_SPI_STM32_DMA=y`

in the Kconfig file, the program no longer works and tells me:

```
*** Booting Zephyr OS build v3.6.0 ***
[00:00:00.001,000] <err> dma_stm32: Memcopy not supported for device dma@40026000
[00:00:00.001,000] <err> sdhc_spi: Card SCLK init sequence failed
[00:00:00.001,000] <err> sd: Could not disable card power via SDHC
[00:00:00.001,000] <err> main: Storage init ERROR!
[00:00:00.001,000] <inf> main: Block count 0
Sector size 0
Memory Size(MB) 0
[00:00:00.002,000] <err> dma_stm32: Memcopy not supported for device dma@40026000
[00:00:00.002,000] <err> sdhc_spi: Card SCLK init sequence failed
[00:00:00.002,000] <err> sd: Could not disable card power via SDHC
[00:00:00.002,000] <err> fs: fs mount error (-5)
Error mounting disk.
Error -5 :failed to configure SD card 
```

And when I use the dma2, I get:
```
*** Booting Zephyr OS build v3.6.0 ***
[00:00:00.205,000] <err> sd: Card error on CMD0
[00:00:00.205,000] <err> main: Storage init ERROR!
[00:00:00.205,000] <inf> main: Block count 0
Sector size 0
Memory Size(MB) 0
[00:00:00.411,000] <err> sd: Card error on CMD0
[00:00:00.411,000] <err> fs: fs mount error (-5)
Error mounting disk.
Error -5 :failed to configure SD card 
```

Am I doing something wrong ?

danieldegrasse, 2024-04-08:
> Am I doing something wrong ?

Unfortunately I'm not familiar with ST's DMA implementation. Could you run a test like the SPI loopback sample (`tests/drivers/spi/spi_loopback/` with DMA enabled? You can connect the SPI MOSI and MISO pins for that test, and it will verify if data is being sent and received correctly

Paul-Avo, 2024-04-09:
With the spi_loopback test with the dma activated, the test is conclusive:

```
*** Booting Zephyr OS build v3.6.0 ***
Running TESTSUITE spi_loopback
===================================================================
START - test_spi_loopback
I: SPI test on buffers TX/RX 0x200002e0/0x200002c0, frame size = 8, DMA enabled (without CONFIG_NOCACHE_)
I: SPI test slow config
I: Start complete multiple
I: Passed
I: Start complete loop
I: Passed
Temps écoulé = 0ns
I: Start null tx
I: Passed
I: Start half start
I: Passed
I: Skip half end
I: Skip every 4
I: Skip rx bigger than tx
I: Start async call
I: Passed
I: SPI test fast config
I: Start complete multiple
I: Passed
I: Start complete loop
I: Passed
Temps écoulé = 0ns
I: Start null tx
I: Passed
I: Start half start
I: Passed
I: Skip half end
I: Skip every 4
I: Skip rx bigger than tx
I: Start async call
I: Passed
I: Start complete loop
I: Passed
Temps écoulé = 0ns
I: Start complete loop
I: Passed
Temps écoulé = 0ns
I: All tx/rx passed
 PASS - test_spi_loopback in 0.073 seconds
===================================================================
TESTSUITE spi_loopback succeeded

------ TESTSUITE SUMMARY START ------

SUITE PASS - 100.00% [spi_loopback]: pass = 1, fail = 0, skip = 0, total = 1 duration = 0.073 seconds
 - PASS - [spi_loopback.test_spi_loopback] duration = 0.073 seconds

------ TESTSUITE SUMMARY END ------

===================================================================
PROJECT EXECUTION SUCCESSFUL

```

So, I don't see where the problem could come from... Do you have any ideas where to look ?

danieldegrasse, 2024-04-12:
> So, I don't see where the problem could come from... Do you have any ideas where to look ?

I'd start by enabling debug logging for the SD subsystem and SDHC SPI driver. Set these Kconfigs:

```
CONFIG_SDHC_LOG_LEVEL_DBG=y
CONFIG_SD_LOG_LEVEL_DBG=y
```

Also, based on the error message in your logs, it seems that DMA might be using memory to memory mode (which seems wrong). I see that the DMA cells for STM32 in the SPI loopback test contain some additional specifiers for the DMA channel-config field, maybe look there?

https://github.com/zephyrproject-rtos/zephyr/blob/89493ee56b6cbd51b1117796b5dc2a2666bc22e8/tests/drivers/spi/spi_loopback/overlay-stm32-spi-16bits.overlay#L12

Paul-Avo, 2024-04-18:
Thanks a lot for your help. I just received my nucleoh723zg card, which allows me to be faster. I would also like to experiment with dma on this card, however, I am stuck again, this time I have not activated dma, and therefore the program should work normally but I have the logs at the initialization level : 

```
*** Booting Zephyr OS build v3.6.0 ***
D: Host controller support 3.3V max
D: Resetting power to card
D: cmd0 arg 0x0
D: cmd8 arg 0x1aa
D: Found SDHC with CMD8 support
D: cmd59 arg 0x1
D: cmd58 arg 0x0
D: cmd55 arg 0x0
D: cmd41 arg 0x40000000
D: cmd55 arg 0x0
D: cmd41 arg 0x40000000
D: SDMMC responded to ACMD41 after 1 attempts
D: cmd58 arg 0x40000000
D: cmd10 arg 0x0
D: cmd12 arg 0x0
D: CMD10 failed: -5
D: Card MID: 0x0, OID: 
D: cmd9 arg 0x0
D: cmd12 arg 0x0
D: CMD9 failed: -5
D: Card block count 0, block size 512
I: Maximum SD clock is under 25MHz, using clock of 20000000Hz
D: cmd55 arg 0x0
```

I could see that the problem seemed to come from the sd_ops.c file and more specifically from shc_request() :

static int sdmmc_spi_read_cxd(struct sd_card *card, uint32_t opcode, uint32_t *cxd)

```c
{
	struct sdhc_command cmd;
	struct sdhc_data data;
	int ret, i;
	/* Use internal card buffer for data transfer */
	uint32_t *cxd_be = (uint32_t *)card->card_buffer;

	cmd.opcode = opcode;
	cmd.arg = 0;
	cmd.response_type = SD_SPI_RSP_TYPE_R1;
	cmd.retries = CONFIG_SD_CMD_RETRIES;
	cmd.timeout_ms = CONFIG_SD_CMD_TIMEOUT;

	/* CID/CSD is 16 bytes */
	data.block_addr = 0;        /* Unused set to 0 */
	data.block_size = 16;
	data.blocks = 1U;
	data.data = cxd_be;
	data.timeout_ms = CONFIG_SD_CMD_TIMEOUT;

	ret = sdhc_request(card->sdhc, &cmd, &data);

	if (ret) {
		LOG_DBG("CMD%d failed: %d", opcode, ret);
	}
	/* Swap endianness of CXD */
	for (i = 0; i < 4; i++) {
		cxd[3 - i] = sys_be32_to_cpu(cxd_be[i]);
	}
	return 0;
}
```

Do you know where this error comes from and how I should correct it knowing that I use exactly the same code on my nucleoF411re card and on it the code works perfectly ?

danieldegrasse, 2024-04-19:
> Do you know where this error comes from and how I should correct it knowing that I use exactly the same code on my nucleoF411re card and on it the code works perfectly

A few things to check:
- are you using the same SD card?
- is pinmux setup correctly?

It looks like the card is failing to respond to the CID request, but does respond to prior requests from the SD subsystem.

danieldegrasse, 2024-04-23:
Looking at the MMC implementation for reading the CID, I don't see any logic to retry reads, so it does not seem like this command should be failing on a card. I do see that your `cs-gpios` property has a pull-up resistor enabled. I don't expect that would cause an issue, but maybe it is causing the SPI bus to have errors? I typically have not set up a SPI bus with a pullup on the CS line.

You could also try raising the value of `CONFIG_SD_CMD_RETRIES` to issue the read CID command multiple times before the SD subsystem reports an error.

Paul-Avo, 2024-04-23:
Thank you for the answer,

I removed the pull up resistor and added the config (with five attempts) but I get:

```
*** Booting Zephyr OS build v3.6.0 ***
D: Host controller support 3.3V max
D: Resetting power to card
D: cmd0 arg 0x0
D: cmd8 arg 0x1aa
D: Found SDHC with CMD8 support
D: cmd59 arg 0x1
D: cmd58 arg 0x0
D: cmd55 arg 0x0
D: cmd41 arg 0x40000000
D: cmd55 arg 0x0
D: cmd41 arg 0x40000000
D: SDMMC responded to ACMD41 after 1 attempts
D: cmd58 arg 0x40000000
D: cmd10 arg 0x0
D: cmd12 arg 0x0
D: cmd10 arg 0x0
D: cmd12 arg 0x0
D: cmd10 arg 0x0
D: cmd12 arg 0x0
D: cmd10 arg 0x0
D: cmd12 arg 0x0
D: cmd10 arg 0x0
D: cmd12 arg 0x0
D: cmd10 arg 0x0
D: cmd12 arg 0x0
D: CMD10 failed: -5
D: Card MID: 0x0, OID: 
D: cmd9 arg 0x0
D: cmd12 arg 0x0
D: cmd9 arg 0x0
D: cmd12 arg 0x0
D: cmd9 arg 0x0
D: cmd12 arg 0x0
D: cmd9 arg 0x0
D: cmd12 arg 0x0
D: cmd9 arg 0x0
D: cmd12 arg 0x0
D: cmd9 arg 0x0
D: cmd12 arg 0x0
D: CMD9 failed: -5
D: Card block count 0, block size 512
I: Maximum SD clock is under 25MHz, using clock of 20000000Hz
D: cmd55 arg 0x0
```

If you have other tests that I can try, I'm interested

danieldegrasse, 2024-04-29:
You're running these tests on a `nucleoh723zg` board? That is a cortex M7 MCU, so the DMA *could* be introducing some cache coherency issues. You could try declaring the `struct sd_card` that you pass to as part of the `nocache` section by adding `_attribute__((__section__(".nocache")))` to the variable declaration. That would move the SD card transfer buffer to nocache memory, and hopefully resolve any issues with cache that could affect the init process.

As an aside, it seems that the `STM32H723ZG` part has an SDHC interface. If you are looking for improved performance (and able to spare the pins) I would suggest looking at using the SDHC interface over the SPI to interface with the SD card. ST's SDHC interface is supported by `drivers/disk/sdmmc_stm32.c` (although I don't know if this specific SOC is supported)