zephyrproject-rtos/zephyr #71010: threads load from uninitialized memory on start

snematbakhsh, 2024-04-02:
I will cite arch/riscv code here but I see the same pattern in other arches.

[`arch_new_thread()`](https://github.com/zephyrproject-rtos/zephyr/blob/main/arch/riscv/core/thread.c#L21) does not entirely initialize its exception stack frame, rather it only initializes the [subset of esf data](https://github.com/zephyrproject-rtos/zephyr/blob/main/arch/riscv/core/thread.c#L37) necessary to get to [`z_thread_entry()`](https://github.com/zephyrproject-rtos/zephyr/blob/main/lib/os/thread_entry.c#L35) on return from the exception handler.

Not initializing irrelevant esf data makes perfect sense for optimization (for example, no need to initialize argument registers past the first four used by `z_thread_entry()`), but it means that we may be loading from uninitialized memory when we [restore from esf in the exception handler](https://github.com/zephyrproject-rtos/zephyr/blob/main/arch/riscv/core/isr.S#L699).

So we're loading from uninitialized memory into registers that we don't care about, what's the big deal? On platforms with ECC SRAM, if SRAM is in some random uninitialized state at power-on, ECC will usually not match on a load from uninitialized SRAM, triggering an ECC fault.

The fix is to fully initialize esf, e.g.:
```
--- a/arch/riscv/core/thread.c
+++ b/arch/riscv/core/thread.c
@@ -33,6 +33,7 @@ void arch_new_thread(struct k_thread *thread, k_thread_stack_t *stack,
        stack_init = (struct __esf *)Z_STACK_PTR_ALIGN(
                                Z_STACK_PTR_TO_FRAME(struct __esf, stack_ptr)
                                );
+       memset(stack_init, 0xAA, sizeof(*stack_init));
```

I see [comments like this one](https://github.com/zephyrproject-rtos/zephyr/blob/main/arch/arc/core/thread.c#L202), so I realize that leaving esf partially-uninitialized was intentional and probably thought-through by the designers, so maybe this issue is "WontFix". But in any case, I opened this bug to document in case future users run into related ECC issues.

peter-mitsis, 2024-04-10:
@snematbakhsh - If I am following the stack initialisation code correctly, having the board enable CONFIG_INIT_STACKS ought to suffice to prevent the scenario you described.

snematbakhsh, 2024-04-11:
> @snematbakhsh - If I am following the stack initialisation code correctly, having the board enable CONFIG_INIT_STACKS ought to suffice to prevent the scenario you described.

Yes, you are correct, CONFIG_INIT_STACKS will work fine. Thanks for the pointer!

closed 2024-04-11 by snematbakhsh