zephyrproject-rtos/zephyr #71018: Nominate myself for contributor role

dnltz, 2024-04-03:
# Background

The [TSC Project Roles] defines the main roles for the Zephyr Project, including
Maintainer, Collaborator, and Contributor.

By default anyone that contributes code or documentation is a Contributor, but
with the lowest [GitHub Permission Level] of Read. For example, Contributors
with Read permission do not have the permission to add reviewers to a pull
request.

Use this template to nominate a GitHub user for the Contributor role with
Triage permission level, which allows the user to add reviewers to a pull
request and be added as a reviewer by other users.

# Nomination

## GitHub User

Provide the following information about the GitHub user:

1. Daniel Schultz
1. dnltz
1. PHYTEC

## Supporting Documents

https://github.com/zephyrproject-rtos/zephyr/pull/62462
https://github.com/zephyrproject-rtos/zephyr/pull/60603
https://github.com/zephyrproject-rtos/zephyr/pull/57802

[TSC Project Roles]: <https://docs.zephyrproject.org/latest/project/project_roles.html>
[GitHub Permission Level]: <https://docs.github.com/en/organizations/managing-access-to-your-organizations-repositories/repository-roles-for-an-organization>

stephanosio, 2024-04-03:
@dnltz has been invited as a contributor with "Triage" permission level.

closed 2024-04-03 by stephanosio

dnltz, 2024-04-16:
@stepanosio I'm now collaborator for the TI K3 Platforms (https://github.com/zephyrproject-rtos/zephyr/commit/b96b91c965ca48f2dc24f5e8dd8a0451b95a85cb). Should I open a new issue or can we re-use this one to increase my role?

stephanosio, 2024-04-22:
@dnltz I have added you as a collaborator now.

dnltz, 2024-10-08:
@stephanosio I'm maintainer of the `PHYTEC Platforms` now to manage PHYTEC boards. Is that already enough to grant me the maintainer role?

https://github.com/zephyrproject-rtos/zephyr/commit/836370f67cd0064817deacb3c4231d9976bfd54e

stephanosio, 2024-10-10:
> @stephanosio I'm maintainer of the `PHYTEC Platforms` now to manage PHYTEC boards. Is that already enough to grant me the maintainer role?

Yes, I have added you as a maintainer now.