zephyrproject-rtos/zephyr #71062: Bluetooth: Increased RAM usage due to removal of `CONFIG_BT_RECV_BLOCKING`

cvinayak, 2024-04-03:
**Describe the bug**
Removal of `CONFIG_BT_RECV_BLOCKING` has increased RAM use leading to failure to build mesh_demo samples on nRF51 SoCs which have smaller/restricted RAM/resource. PR https://github.com/zephyrproject-rtos/zephyr/pull/70510 addressed this by incorrectly reducing the Controller's receive thread stack size. This reduced receive thread stack causes HCI Controller hangs in hci_ipc, hci_usb and hci_uart samples.
 
**To Reproduce**
Steps to reproduce the behavior:
1. mkdir -p build/hci_usb; cd build/hci_usb
2. cmake -GNinja -DBOARD=nrf52840dongle_nrf52840 -DEXTRA_CONF_FILE=../hci_uart/overlay-all-bt_ll_sw_split.conf ../../samples/bluetooth/hci_usb
3. ninja
4. ninja flash
5. Use the dongle in a Linux OS distribution
6. Use `bluetoothctl`
7. run `scan le` to start scanning
8. Observe that after a duration, randomly the dongle hangs and requires to be reset

**Expected behavior**
The HCI controller can be used without hangs.

**Impact**
showstopper

**Logs and console output**
the stack size is reduced from 768 bytes to 448 bytes and is near overflow:
```
  BT RX               : STACK: unused 16 usage 432 / 448 (96 %); CPU: 0 %
      : Total CPU cycles used: %llu
  BT RX pri           : STACK: unused 280 usage 168 / 448 (37 %); CPU: 0 %
      : Total CPU cycles used: %llu
```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain Zephyr SDK
 - Commit SHA or Version used: https://github.com/zephyrproject-rtos/zephyr/commit/bba8641354bfbf83743c032c378a791b941ad7fc

**Additional context**
NA

jhedberg, 2024-04-03:
I was a bit afraid that something like this would happen, but assumed that the more memory-constrained platforms (at least Nordic ones) would have been tested with the change. I've set high priority on this one for now.

jori-nordic, 2024-04-04:
> nRF51 SoCs

> at least Nordic ones

This SoC family is community-supported. NCS does not support it, the recommended Bluetooth stack is the nRF5 SDK + Softdevice

alwa-nordic, 2024-04-04:
The problem is in `subsys/bluetooth/controller`, so I added the 'controller' tag.

To clarify: This it's not a true RAM capacity problem. The hci_ipc, hci_usb and hci_uart samples don't use the host, so how can a change in the host affect them?

The problem here is my change discussed here: https://github.com/zephyrproject-rtos/zephyr/pull/70510/files#r1537492110.

The problem is that the controller was using `CONFIG_BT_RX_STACK_SIZE`, which was unnecessarily large in combination with the host. I lowered it without thinking about the hci samples. I did not imagine that they have a large stack requirement.

The problem should be fixed by figuring out some scheme to minimize the controller driver stack size without breaking the hci samples. Possibly by changing the hci samples. A quick fix could be to add a new kconfig for the controller driver stack size.

theob-pro, 2024-04-04:
WG discussion:
- create a controller-only Kconfig for `hci_driver.c`'s RX stack size
- profile and minimize that size on constrained platforms

PR #71043 (2024-04-12): Bluetooth: Controller: Fix regression in BT_RX_STACK_SIZE use
Fix regression in BT_RX_STACK_SIZE use due to
commit b91728619c05 ("Bluetooth: host: remove
`CONFIG_BT_RECV_BLOCKING`").

Fixes #71062.
closed 2024-04-12 by carlescufi