zephyrproject-rtos/zephyr #71068: "Missing #address-cells in interrupt provider" warnings when building certain platforms

de-nordic, 2024-04-03:
**Describe the bug**
This kind of warnings:
```
zephyr.dts:26.19-31.4: Warning (interrupt_provider): /arcv2-intc: Missing #address-cells in interrupt provider
```
appear when building tests for following platforms:
qemu_cortex_r5
qemu_cortex_a9
qemu_leon3
qemu_arc_qemu_arc_hs6x
qemu_arc/qemu_arc_hs      
qemu_arc/qemu_arc_em      
qemu_arc/qemu_arc_hs5x  
qemu_arc/qemu_arc_hs/xip  

**To Reproduce**
Steps to reproduce the behavior, pick a platform from a list, do:
1. twister -p <platform>   -T  zephyr/tests/subsys/fs/fs_api/
2. see a log

**Expected behavior**
No warning.

**Impact**
It warnings me.

**Environment (please complete the following information):**
 - OS: Any
 - Toolchain zephyr sdk 0.16.x
 - Commit 2c722f97b54f6479b176b2abb8b92fa8145277e9 on top

**Additional context**
Too tired to figure out the right values and where to put them, somebody else is fixing this.

nashif, 2024-04-09:
not able to repdouce this or get the warnings in the title, can you please provide some more details?
I built using twister/west and was not able to see the warnings anywhere, maybe I am missing something?

de-nordic, 2024-04-09:
So now I am using the Zephyr at sha 3b895064afaf7924a80def09bd1edeefa61ebc9f and doing this:
```
twister -p qemu_cortex_r5 -T tests/subsys/fs/fs_api/
```
from within the Zephyr dir.
(and after waiting for ages it takes to run test on my machine)
I got this:
```
$$ twister -p qemu_cortex_r5 -T tests/subsys/fs/fs_api/
Renaming output directory to <redacted>/zephyrproject-clean/zephyr/twister-out.5
INFO    - Using Ninja..
INFO    - Zephyr version: v3.6.0-2341-g3b895064afaf
INFO    - Using 'zephyr' toolchain.
INFO    - Building initial testsuite list...
INFO    - Writing JSON report <redacted>zephyrproject-clean/zephyr/twister-out/testplan.json
INFO    - JOBS: 6
INFO    - Adding tasks to the queue...
INFO    - Added initial list of jobs to queue
ERROR   - qemu_cortex_r5            tests/subsys/fs/fs_api/filesystem.api               FAILED : Unknown
ERROR   - see: <redacted>/zephyrproject-clean/zephyr/twister-out/qemu_cortex_r5/tests/subsys/fs/fs_api/filesystem.api/build.log
INFO    - Total complete:    1/   1  100%  skipped:    0, failed:    1, error:    0
INFO    - 1 test scenarios (1 test instances) selected, 0 configurations skipped (0 by static filter, 0 at runtime).
INFO    - 0 of 1 test configurations passed (0.00%), 1 failed, 0 errored, 0 skipped with 0 warnings in 69.81 seconds
INFO    - In total 11 test cases were executed, 0 skipped on 1 out of total 711 platforms (0.14%)
INFO    - 0 test configurations executed on platforms, 1 test configurations were only built.
INFO    - Saving reports...
INFO    - Writing JSON report <redacted>/zephyrproject-clean/zephyr/twister-out/twister.json
INFO    - Writing xunit report <redacted>/zephyrproject-clean/zephyr/twister-out/twister.xml...
INFO    - Writing xunit report <redacted>/zephyrproject-clean/zephyr/twister-out/twister_report.xml...
INFO    - -+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+
INFO    - The following issues were found (showing the top 10 items):
INFO    - 1) tests/subsys/fs/fs_api/filesystem.api on qemu_cortex_r5 failed (Unknown)
INFO    -
INFO    - To rerun the tests, call twister using the following commandline:
INFO    - west twister -p <PLATFORM> -s <TEST ID>, for example:
INFO    -
INFO    - west twister -p qemu_cortex_r5 -s tests/subsys/fs/fs_api/filesystem.api
INFO    - or with west:
INFO    - west build -p -b qemu_cortex_r5 tests/subsys/fs/fs_api -T filesystem.api
INFO    - -+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+
```

in the log that is listed above I have:
```
-- Found host-tools: zephyr 0.16.1 (<redacted>/zephyr-sdk)
-- Found toolchain: zephyr 0.16.1 (<redacted>/zephyr-sdk)
-- Found Dtc: /usr/bin/dtc (found suitable version "1.6.1", minimum required is "1.4.6")
-- Found BOARD.dts: <redacted>/zephyrproject-clean/zephyr/boards/qemu/cortex_r5/qemu_cortex_r5.dts
-- Generated zephyr.dts: <redacted>/zephyrproject-clean/zephyr/twister-out/qemu_cortex_r5/tests/subsys/fs/fs_api/filesystem.api/zephyr/zephyr.dts
-- Generated devicetree_generated.h: <redacted>/zephyrproject-clean/zephyr/twister-out/qemu_cortex_r5/tests/subsys/fs/fs_api/filesystem.api/zephyr/include/generated/devicetree_generated.h
-- Including generated dts.cmake file: <redacted>/zephyrproject-clean/zephyr/twister-out/qemu_cortex_r5/tests/subsys/fs/fs_api/filesystem.api/zephyr/dts.cmake
<redacted>/zephyrproject-clean/zephyr/twister-out/qemu_cortex_r5/tests/subsys/fs/fs_api/filesystem.api/zephyr/zephyr.dts:270.38-277.5: Warning (interrupt_provider): /soc/interrupt-controller@f9000000: Missing #address-cells in interrupt provider
```
and the last lines are:
[image]

nashif, 2024-04-10:
strange, the test passed for me (it built) and I do not get any warnings...

`grep interrupt_provider  twister-out/qemu_cortex_r5/tests/subsys/fs/fs_api/filesystem.api/build.log`

de-nordic, 2024-04-10:
> strange, the test passed for me (it built) and I do not get any warnings...
> 
> `grep interrupt_provider twister-out/qemu_cortex_r5/tests/subsys/fs/fs_api/filesystem.api/build.log`
> 
> yields nothing and the build log is clean:
> 
> ```
> -- Found host-tools: zephyr 0.16.4 (/home/nashif/bin/zephyr-sdk-0.16.4)
> -- Found toolchain: zephyr 0.16.4 (/home/nashif/bin/zephyr-sdk-0.16.4)
> -- Found Dtc: /home/nashif/bin/zephyr-sdk-0.16.4/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.6.0", minimum required is "1.4.6")
> -- Found BOARD.dts: /home/nashif/zephyrproject/zephyr/boards/qemu/cortex_r5/qemu_cortex_r5.dts
> -- Generated zephyr.dts: /home/nashif/zephyrproject/zephyr/twister-out/qemu_cortex_r5/tests/subsys/fs/fs_api/filesystem.api/zephyr/zephyr.dts
> -- Generated devicetree_generated.h: /home/nashif/zephyrproject/zephyr/twister-out/qemu_cortex_r5/tests/subsys/fs/fs_api/filesystem.api/zephyr/include/generated/devicetree_generated.h
> -- Including generated dts.cmake file: /home/nashif/zephyrproject/zephyr/twister-out/qemu_cortex_r5/tests/subsys/fs/fs_api/filesystem.api/zephyr/dts.cmake
> Parsing /home/nashif/zephyrproject/zephyr/Kconfig
> Loaded configuration '/home/nashif/zephyrproject/zephyr/boards/qemu/cortex_r5/qemu_cortex_r5_defconfig'
> Merged configuration '/home/nashif/zephyrproject/zephyr/tests/subsys/fs/fs_api/prj.conf'
> Merged configuration '/home/nashif/zephyrproject/zephyr/twister-out/qemu_cortex_r5/tests/subsys/fs/fs_api/filesystem.api/zephyr/misc/generated/extra_kconfig_options.conf'
> Configuration saved to '/home/nashif/zephyrproject/zephyr/twister-out/qemu_cortex_r5/tests/subsys/fs/fs_api/filesystem.api/zephyr/.config'
> Kconfig header saved to '/home/nashif/zephyrproject/zephyr/twister-out/qemu_cortex_r5/tests/subsys/fs/fs_api/filesystem.api/zephyr/include/generated/autoconf.h'
> -- Found GnuLd: /home/nashif/bin/zephyr-sdk-0.16.4/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/../../../../arm-zephyr-eabi/bin/ld.bfd (found version "2.38")
> -- The C compiler identification is GNU 12.2.0
> -- The CXX compiler identification is GNU 12.2.0
> -- The ASM compiler identification is GNU
> -- Found assembler: /home/nashif/bin/zephyr-sdk-0.16.4/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc
> -- Using ccache: /usr/bin/ccache
> -- Configuring done (5.5s)
> -- Generating done (0.0s)
> -- Build files have been written to: /home/nashif/zephyrproject/zephyr/twister-out/qemu_cortex_r5/tests/subsys/fs/fs_api/filesystem.api
> [1/126] Preparing syscall dependency handling
> ```
> 
> what sdk version are you using?

The zephyr-sdk-0.16.1.
I'll get this https://github.com/zephyrproject-rtos/sdk-ng/releases/tag/v0.16.5-1 and check again.

de-nordic, 2024-04-10:
Sorry, I still get the thing:
[image]

This is my system:
```
$$ uname -a
Linux local-bm 5.15.0-101-generic #111-Ubuntu SMP Tue Mar 5 20:16:58 UTC 2024 x86_64 x86_64 x86_64 GNU/Linux
```
sdk https://github.com/zephyrproject-rtos/sdk-ng/releases/tag/v0.16.5-1 , got it today.

This is Zephyr sha: 3b895064afaf7924a80def09bd1edeefa61ebc9f
The west update has been done.
I run `pip3 install -U -r ` on everything from scripts dir.
`apt update` gives no updates required.

nashif, 2024-04-10:
ok, will look on some other machine and another environment. Thanks.

de-nordic, 2024-04-10:
> ok, will look on some other machine and another environment. Thanks.

Sure no problem. If you get any more idea I should test, let me know.

nashif, 2024-05-16:
tried to reproduce again with no luck, cell-address is not required afaik, so not sure what is going on. Asked many other to try and it works.

ceolin, 2024-05-16:
I couldn't reproduce it either.  Using zephyr 0.16.6

teburd, 2024-05-17:
What version of python?

de-nordic, 2024-05-17:
> What version of python and cmake? does it happen if you build and run the test without twister involved?

```
$$ cmake --version
cmake version 3.27.7
$$ python3 --version
Python 3.10.12
```

give me a moment to check get back to this and check with west.

de-nordic, 2024-05-17:
OK, so with the Zephyr at adeb19b30cbfb3552cb2d19c6c6676a872afe6c0, using, for example:
```
rm -r builds/qcr5_fs_api 
west build -b qemu_cortex_r5 -d builds/qcr5_fs_api  zephyr/tests/subsys/fs/fs_api/
```
I get this:
[image]

west update was done on the branch with the sha as mentioned.

teburd, 2024-05-17:
I didn't run into any problems myself either

```sh
(env) ~/z/zephyr ❯❯❯ which python
/home/tburdick/zephyrtrees/env/bin/python
(env) ~/z/zephyr ❯❯❯ which cmake
/usr/bin/cmake
(env) ~/z/zephyr ❯❯❯ python --version
Python 3.12.3
(env) ~/z/zephyr ❯❯❯ cmake --version
cmake version 3.29.3

CMake suite maintained and supported by Kitware (kitware.com/cmake).
(env) ~/z/zephyr ❯❯❯ cat ~/zephyr-sdk/sdk_version
0.16.1
(env) ~/z/zephyr ❯❯❯ cd ..
(env) ~/zephyrtrees ❯❯❯ ls
bootloader  builds  env  modules  optional  tools  zephyr
(env) ~/zephyrtrees ❯❯❯ west build -p always -b qemu_cortex_r5 -d builds/qcr5_fs_api zephyr/tests/subsys/fs/fs_api
-- west build: making build dir /home/tburdick/zephyrtrees/builds/qcr5_fs_api pristine
-- west build: generating a build system
Loading Zephyr default modules (Zephyr base).
-- Application: /home/tburdick/zephyrtrees/zephyr/tests/subsys/fs/fs_api
-- CMake version: 3.29.3
-- Found Python3: /home/tburdick/zephyrtrees/env/bin/python (found suitable version "3.12.3", minimum required is "3.8") found components: Interpreter
-- Cache files will be written to: /home/tburdick/.cache/zephyr
-- Zephyr version: 3.6.99 (/home/tburdick/zephyrtrees/zephyr)
-- Found west (found suitable version "1.2.0", minimum required is "0.14.0")
-- Board: qemu_cortex_r5, qualifiers: zynqmp_rpu
-- ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
-- Found host-tools: zephyr 0.16.1 (/home/tburdick/zephyr-sdk)
-- Found toolchain: zephyr 0.16.1 (/home/tburdick/zephyr-sdk)
-- Found Dtc: /home/tburdick/zephyr-sdk/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.6.0", minimum required is "1.4.6")
-- Found BOARD.dts: /home/tburdick/zephyrtrees/zephyr/boards/qemu/cortex_r5/qemu_cortex_r5.dts
-- Generated zephyr.dts: /home/tburdick/zephyrtrees/builds/qcr5_fs_api/zephyr/zephyr.dts
-- Generated devicetree_generated.h: /home/tburdick/zephyrtrees/builds/qcr5_fs_api/zephyr/include/generated/devicetree_generated.h
-- Including generated dts.cmake file: /home/tburdick/zephyrtrees/builds/qcr5_fs_api/zephyr/dts.cmake
Parsing /home/tburdick/zephyrtrees/zephyr/Kconfig
Loaded configuration '/home/tburdick/zephyrtrees/zephyr/boards/qemu/cortex_r5/qemu_cortex_r5_defconfig'
Merged configuration '/home/tburdick/zephyrtrees/zephyr/tests/subsys/fs/fs_api/prj.conf'
Configuration saved to '/home/tburdick/zephyrtrees/builds/qcr5_fs_api/zephyr/.config'
Kconfig header saved to '/home/tburdick/zephyrtrees/builds/qcr5_fs_api/zephyr/include/generated/autoconf.h'
-- Found GnuLd: /home/tburdick/zephyr-sdk/arm-zephyr-eabi/arm-zephyr-eabi/bin/ld.bfd (found version "2.38")
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/tburdick/zephyr-sdk/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc
-- Using ccache: /usr/bin/ccache
-- Configuring done (6.7s)
-- Generating done (0.1s)
-- Build files have been written to: /home/tburdick/zephyrtrees/builds/qcr5_fs_api
-- west build: building application
[1/126] Preparing syscall dependency handling

[2/126] Generating include/generated/version.h
-- Zephyr version: 3.6.99 (/home/tburdick/zephyrtrees/zephyr), build: v3.6.0-4147-gadeb19b30cbf
[126/126] Linking C executable zephyr/zephyr.elf
Memory region         Used Size  Region Size  %age Used
           FLASH:          0 GB        32 MB      0.00%
             RAM:       72932 B        64 MB      0.11%
             OCM:          0 GB       256 KB      0.00%
        IDT_LIST:          0 GB        32 KB      0.00%
Generating files from /home/tburdick/zephyrtrees/builds/qcr5_fs_api/zephyr/zephyr.elf for board: qemu_cortex_r5
```

de-nordic, 2024-05-17:
I got the same dts file, diff shows 0... but I have run dtc on the DTS I have downloaded from https://github.com/zephyrproject-rtos/zephyr/issues/71068#issuecomment-2117567254 and get:
```
$$ dtc ~/tmp/probably_some_virus.dts  > /dev/null
/home/doer/tmp/probably_some_virus.dts:270.38-277.5: Warning (interrupt_provider): /soc/interrupt-controller@f9000000: Missing #address-cells in interrupt provider
```

I have
```
$$ dtc --version
Version: DTC 1.6.1
```

I do not know. Should I debug the python...

All requirements show no updates... I start suspect that my stderr outs this to annoy me.

teburd, 2024-05-17:
seems like a dtc difference is the culprit? I have 1.7

de-nordic, 2024-05-21:
> seems like a dtc difference is the culprit? I have 1.7

Yes, it seems so. I have built the dtc from latest code:
```
$$ dtc ~/tmp/probably_some_virus.dts  --version
Version: DTC 1.6.1
$$=>0 | 15:25:14 doer@local-bm:/repos/doer/dtc
$$ dtc ~/tmp/probably_some_virus.dts  1>/dev/null
/home/doer/tmp/probably_some_virus.dts:270.38-277.5: Warning (interrupt_provider): /soc/interrupt-controller@f9000000: Missing #address-cells in interrupt provider
$$=>0 | 15:25:22 doer@local-bm:/repos/doer/dtc
$$ ./dtc --version
Version: DTC 1.7.0-gae26223a
$$=>0 | 15:25:36 doer@local-bm:/repos/doer/dtc
$$ ./dtc  ~/tmp/probably_some_virus.dts 1>/dev/null
$$=>0 | 15:25:41 doer@local-bm:/repos/doer/dtc
```

In summary: the issue is with DTC compiler below 1.7. Closing.

closed 2024-05-21 by de-nordic