zephyrproject-rtos/zephyr #71069: Heap & Memory Allocation Statistics: add mallinfo support & better dynamic memory documentation

htcrane, 2024-04-03:
**Is your enhancement proposal related to a problem? Please describe.**
1) The documentation does not make it obviously clear if using `malloc()` or `k_malloc()` is the preferred method for dynamic memory allocation.
2) There currently is no clean way of getting heap statistics for the _standard library_, such as with `mallinfo()`.

**Describe the solution you'd like**
1) Provide the user with a standard library function to understand how much memory is left for allocation via `malloc()`.
    - This could be completed through added support for `mallinfo()` for structure statistics and `malloc_stats()` for printed statistics ([newlib C, section 3.25](https://sourceware.org/newlib/libc.html#mallinfo)).
    - Completion by adding the `_mallinfo()` and `_malloc_stats()` stubs for the standard library.
    - For newlib, this could be achieved in */lib/libc/newlib/libc-hooks.c* by using existing variables: `HEAP_BASE` and `heap_sz`.

2) Provide *clear* documentation on *why* a user may want to use `malloc()` directly or stick to `k_malloc()`.
    - To my understanding...
        - `malloc()` is general purpose and supported no matter the standard C library chosen to build with. The memory sourced may go through a virtualization step where memory may no longer be continuous regions.
        - `k_malloc()` requires additional setup through the `CONFIG_HEAP_MEM_POOL_SIZE` option and is useful if memory should be allocated from a common memory region.
    - If this is correct, this should be stated in the documentation directly so others do not need to sift through multiple function calls to figure out what the difference is. Why one may be better for a task than the other.

3) Provide a memory summary documentation page.
Looking through several issues, conversations about the heap talk about the heap available to the standard c libraries (malloc) and the system heap (k_alloc) interchangeably when *they do not source their memory from the same heap*.

**Describe alternatives you've considered**

- Currently no way to know how much memory is left for `malloc()` or other heap statistics available to the standard library.
- Previous issues of such discussions about stats for malloc:
    - #29735 
    - #23076
    - #57725 

**Additional context**

- Zephyr currently implements a heap size function for the user-defined system heap: `sys_heap_runtime_stats_get()` shown in this [sample](https://github.com/zephyrproject-rtos/zephyr/blob/2c722f97b54f6479b176b2abb8b92fa8145277e9/samples/basic/sys_heap/src/main.c#L41). This function does not appear anywhere in the latest Zephyr documentation.
- The kernel shell service that uses aforementioned function above `cmd_kernel_heap()`, [code link](https://github.com/zephyrproject-rtos/zephyr/blob/2c722f97b54f6479b176b2abb8b92fa8145277e9/subsys/shell/modules/kernel_service.c#L280C12-L280C27), also does not appear anywhere in the latest Zephyr documentation. It provides stats for the `_system_heap`, which is used for `k_malloc()`. This is useful for that single memory region, which is not the same memory region as `heap_base` where `malloc()` is sourcing its memory from (when using newlib).

povsel, 2025-03-13:
Are there any news regarding this? 
I am looking for a way to track / monitor heap usage (malloc/new).

merethan, 2025-11-25:
> Looking through several issues, conversations about the heap talk about the heap available to the standard c libraries (malloc) and the system heap (k_alloc) interchangeably _when they do not source their memory from the same heap_.

That was indeed one of the more insidious pitfalls I have faced.

```
#if defined(CONFIG_SYS_HEAP_RUNTIME_STATS)
    struct sys_memory_stats k_heapstat;
    int ret = sys_heap_runtime_stats_get(&_system_heap, &k_heapstat);
    size_t unused = K_HEAP_MEM_POOL_SIZE - k_heapstat.max_allocated_bytes;
    if(unused < lowmem_heap && ret >= 0)
        LOG_WRN("Kernel heap unused mem low: %u (size: %u, alloc: %u, free: %u)", unused, K_HEAP_MEM_POOL_SIZE, k_heapstat.allocated_bytes, k_heapstat.free_bytes);
#endif // CONFIG_SYS_HEAP_RUNTIME_STATS

#ifdef CONFIG_NEWLIB_LIBC
#define USED_RAM_END_ADDR POINTER_TO_UINT(&_end)
#define HEAP_BASE USED_RAM_END_ADDR
#define MAX_HEAP_SIZE (KB(CONFIG_SRAM_SIZE) - (HEAP_BASE - CONFIG_SRAM_BASE_ADDRESS))

    const struct mallinfo p_heapstats = mallinfo();

    if(p_heapstats.fordblks < lowmem_heap)
        LOG_WRN("Newlibc heap mem low: %u (size: %u, alloc: %u, usable: %u)", p_heapstats.fordblks, MAX_HEAP_SIZE, p_heapstats.uordblks, p_heapstats.arena);
#endif // CONFIG_NEWLIB_LIBC
```
When selecting picolibc the above fails, hence the `#ifdef CONFIG_NEWLIB_LIBC`.

Keith Packard did put `mallinfo()` in picolibc, so it not working with Zephyr is likely a quirck/build/packaging error specific to Zephyr. I not got into figuring out why yet.