zephyrproject-rtos/zephyr #71119: nRF ADC driver when used in single-ended mode can return negative values

nixward, 2024-04-05:
**Describe the bug**
When measuring voltages near 0V the nRF ADC driver when used in single-ended mode can return negative values.

See discussion: https://github.com/zephyrproject-rtos/zephyr/pull/70277#discussion_r1551337884

**Expected behavior**
Single-ended mode should only output values >= 0.

nixward, 2024-06-29:
Hi @anangl is there anyone at Nordic who can have a look at this?

bjarki-andreasen, 2024-07-03:
I'm on the case :) Managed to reproduce the issue quite easily, -10mV on the input, and I see negative values.

The following is near 0V
```
ADC sequence reading [161]:
- adc@40007000, channel 0, 5 sequence samples:
- - 80 = 0mV
- - 62 = 0mV
- - 101 = 0mV
- - 24 = 0mV
- - 70 = 0mV
```

The following is near -10mV:
```
ADC sequence reading [139]:
- adc@40007000, channel 0, 5 sequence samples:
- - 61440 = 60mV
- - 61440 = 60mV
- - 61440 = 60mV
- - 61440 = 60mV
- - 61440 = 60mV
```

bjarki-andreasen, 2024-07-05:
This PR should fix the issue, I am still testing it, but it seems to be functional :) #75420

nordic-piks, 2024-07-05:
Please notice that sample adc_sequence has this bug https://github.com/zephyrproject-rtos/zephyr/issues/75517
which will lead to invalid conversion to millivolts.

nordic-piks, 2024-07-05:
Let me add some comments here:
I checked adc samples: `adc_dt` and `adc_sequence`. None of them, nor the driver automatically, do the calibration.
Offset is not "removed" before sampling, thus values near 0 can easily flip to the negative side.
I did a simple change and enabled calibration at `adc_dt` sample. After that change, readings were much closer to 0 (if single-ended input was shorted to GND). Still, a flip into negative happens, but it was only one bit.
Maybe the approach should be to enable (or force calibration) and only then convert negative values into 0?
(converting negatives into 0 without calibration - i.e. without "removing" offset - will make accuracy worse)
@anangl @kl-cruz @nika-nordic What do you think?

bjarki-andreasen, 2024-07-06:
@nordic-piks Calibrating the ADC before reading can help, but noise, or simply applying a negative voltage to the positie input will produce a negative reading, the PR fixing the issue explains why :)

bjarki-andreasen, 2024-07-08:
PR is tested and ready for review :)

PR #75420 (2024-07-12): drivers: adc: adc_nrfx_saadc: Patch neg single ended readings
The NRF SAADC produces negative values in single ended mode if the positive input is below 0V (ground). This behavior does not match the ADC device driver API, which states that in single ended mode, the readings must be positive [0 .. 2^resolution - 1].

This commit extends the adc_nrfx_saadc device driver to track which channels in a sequence are configured to be single ended, to then adjust negative readings for these channels to be 0.

### Tests
Build adc_sequence sample with a modified board overlay which has one single ended a [... cut]
closed 2024-07-12 by aescolar