zephyrproject-rtos/zephyr #71121: Log warning when enabling mDNS

hjuul, 2024-04-05:
**Describe the bug**
I get the following log warning when booting my application:

- `<wrn> net_if: iface 0x20020bc0 is down`

This happens even when the network cable is not connected.

I first saw this is in my own application using custom hardware and Zephyr v3.5.0.
But I've been able to reproduce it using a sample application running on stm32f769i_disco board on Zephyr master branch.

**To Reproduce**
Steps to reproduce the behavior:
1. Starting with sample application `samples/net/dhcpv4_client`
2. Add the following to `prj.conf`:
   - `CONFIG_NET_HOSTNAME_ENABLE=y`
   - `CONFIG_MDNS_RESPONDER=y`
3. Build the sample application for stm32f769i_disco
4. Flash and run
5.  Look at UART output

The sample application is functional - it gets an IP address from DHCP server.
I can also ping it at `zephyr.local`, so the mDNS I added seems to work.
So it is just the warning message that is unexpected.

**Expected behavior**
Working application with no warnings.

**Impact**
Annoyance

**Logs and console output**

```
*** Booting Zephyr OS build v3.6.0-2090-gfec3cee0bbe6 ***
[00:00:00.012,000] <wrn> net_if: iface 0x20020bc0 is down
[00:00:00.012,000] <inf> net_dhcpv4_client_sample: Run dhcpv4 client
[00:00:00.012,000] <inf> net_dhcpv4_client_sample: Start on ethernet@40028000: index=1
[00:00:04.015,000] <inf> net_dhcpv4: Received: 10.42.68.105
[00:00:04.015,000] <inf> net_dhcpv4_client_sample:    Address[1]: 10.42.68.105
[00:00:04.015,000] <inf> net_dhcpv4_client_sample:     Subnet[1]: 255.255.255.0
[00:00:04.015,000] <inf> net_dhcpv4_client_sample:     Router[1]: 10.42.68.1
[00:00:04.015,000] <inf> net_dhcpv4_client_sample: Lease time[1]: 86400 seconds
```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: zephyr-sdk-0.16.5-1
 - Zephyr v3.5.0 and latest commit on master

**Additional context**
Some comments:
- The warning can originate from two functions in `net_if.c`. The one I get is from function `net_if_send_data()`
- In my custom application I'm seeing this bug with a different warning:
  - `<err> eth_stm32_hal: Failed to enqueue frame into RX queue: -115`
  - I'm not 100% sure it is the same issue, but seems like it (error -115 is `ENETDOWN`)

jukkar, 2024-04-06:
> I get the following log warning when booting my application:
> 
>     * `<wrn> net_if: iface 0x20020bc0 is down`
> 
> 
> This happens even when the network cable is not connected.

Do you mean that you get this when you have the cable connected? I mean the error is expected if the cable is not connected.

If you can send / receive data, then it implies that the network interface is working fine and is up. Possible reason for the warning is that when the device boots, it might take some time before the interface is actually up. If someone tries to send data at that small window, you will get that warning.

BTW, you do not really need to edit the sample `prj.conf` file, as you can give additional config options when building the app like this:
```
west build -p -b stm32f769i_disco samples/net/dhcpv4_client -d ../build/dhcpv4-client -- -DCONFIG_NET_HOSTNAME_ENABLE=y -DCONFIG_MDNS_RESPONDER=y
```

hjuul, 2024-04-06:
@jukkar 

> Do you mean that you get this when you have the cable connected? I mean the error is expected if the cable is not connected.

Yes, this happens when cable is connected also

>If someone tries to send data at that small window, you will get that warning.

I'm not sending or receiving any data per se, just pinging. And I start pinging well after the interface is up.

Could it be that the mDNS tries to broadcast something before the interface is up?

jukkar, 2024-04-06:
mDNS will join multicast group at startup, but it will only do that after network interface is up, so that is not the culprit here.
I tried with a nucleo board I have and can see same printout, I try to dig out deeper why that is printed.

jukkar, 2024-04-06:
Found the culprit, it was mDNS after all. It has a detection for interface up but there is something wrong with it and it tries to join multicast group before the interface is up.

hjuul, 2024-04-06:
Fantastic, let me know if you need any help with testing.

jukkar, 2024-04-06:
The problem is only the printout, the multicast group is tried to join when the mDNS responder inits itself and that fails because the interface is still down. But when the interface is brought up, the group joining is retried and the joining will succeed at that point.
We could certainly check if the interface is up before joining the multicast group, that would prevent the warning.

jukkar, 2024-04-06:
> We could certainly check if the interface is up before joining the multicast group, that would prevent the warning.

Hmm, we cannot actually do that because then mDNS responder would not be able to bind the multicast address.
This is somewhat bigger task to change the mDNS responder initialization logic. I not sure if this is worth fixing, we could remove the printout but it might be useful to have in some cases.

jukkar, 2024-04-06:
@hjuul please take a look the PR, I managed to find a way to avoid the warning.

hjuul, 2024-04-08:
@jukkar Your PR fixes the reported issue. Also, I don't get the error message even if the network cable is not attached - which is a good thing.

Thanks for fixing this!

PR #71184 (2024-04-10): net: mdns_responder: Avoid joining IPv4 mcast group if iface is down
Check if the interface is down before joining IPv4 mDNS multicast group. This avoids warning later when the network packet cannot be sent.

Fixes #71121
commit 85911bcf (2024-04-10): net: mdns_responder: Avoid joining IPv4 mcast group if iface is down

Check if the interface is down before joining IPv4 mDNS multicast
group. This avoids warning later when the network packet cannot be sent.

Fixes #71121
closed 2024-04-10 by carlescufi