zephyrproject-rtos/zephyr #71122: Hardware model v2: Improve board and overlay file name scheme

carlescufi, 2024-04-05:
Due to a series of hard constraints, the file name scheme introduced with hardware model v2 had to be reworked. The main reason behind this is the fact that initially hardware model v2 merged Kconfig configuration files hierarchically in a board, but that was quickly identified as a major issue breaking several boards and use cases.

In an effort to get over this problem as soon as possible, an initial fix has been proposed, discussed and, for the most part, approved. This is implemented in PR #70548, and essentially takes the thorough approach: both board files and overlay files need to contain the board name and SoC (and, when applicable, CPU cluster and variant). While this is bullet-proof, it also makes the file names longer than they were and a bit cumbersome to parse.

There is one fallback we could use, in the same PR #70548, in order to be able to shorten board definition file names (see presentation below). The issue is that it would only apply to board definition files, and not overlays, so shortening the board files at the cost of inconsistency with overlay files.

The proposal thus is to merge #70548 in order to "unbreak" the current issues, and shortly after introduce a mechanism that is able to shorten names. This  could be an alias system or an optimization by which boards that only have a single SoC, CPU cluster and variant are able to use the board name as the file name.

More context:

- [Presentation at the Architecture WG](https://docs.google.com/presentation/d/1BCeKKhs3BV5EciA5cQrbEfMexOjWsHyj-dgcB0yJrbQ/edit#slide=id.p)
- [Recording of the Architecture WG](https://zoom.us/rec/share/KHaMiI1O7r7vmQdSvXp-LnnSRIed2ikNeiUE7zIVUiRRgdJ-fGjbJUAKxq8awxGZ.p8AMB9sstvctuYJB)
- [Discord thread](https://discord.com/channels/720317445772017664/1224744062687641622/1224744064168235028)

tejlmand, 2024-04-05:
>  The issue is that it would only apply to board definition files, and not overlays, so shortening the board files at the cost of inconsistency with overlay files.

Not true. Overlays can be shortened too if we want to allow that. And #70548 supports short names.

The issue, as I have pointed out many times, is the merge feature, not the (short) names themselves.
The names purely comes back to our own desire to be able to be extremely precise in how to name an overlay.

Especially because the short form can only be allowed for boards with single SoC overlays / configs.

tejlmand, 2024-04-05:
But notice, the cost is the complexity for users in when the SoC name is required.
Remember, on command line, if the SoC is required when using `-DBOARD=<board>`, then user will be given the list of qualifiers, if SoC is required.
Whereas, a missing SoC, for a board with >1 SoC, will not tell you that a file exists but is not picked up, for example when naming:
`nrf52840dk.overlay`, when that board has t.wo SoCs defined (real: nrf52840, emulation mode: nrf52811).

carlescufi, 2024-04-05:
> Whereas, a missing SoC, for a board with >1 SoC, will not tell you that a file exists but is not picked up, for example when naming:
> `nrf52840dk.overlay`, when that board has t.wo SoCs defined (real: nrf52840, emulation mode: nrf52811).

Well, let's look for it and fail then. Because that is just a mistake on the user's part, clearly.

nordicjm, 2024-04-08:
A requirement for this being reintroduced is checking for files that have typos in the name as has been seen in boards which were ported to hwmv2. To do this, python code can be created that globs all board.yml files together so a list of boards and soc names can be generated, then a glob of all *.dts and *_defconfig files in the boards directory can be generated and checked to ensure that all filenames are valid.

tejlmand, 2024-04-17:
Closing as fixed, because https://github.com/zephyrproject-rtos/zephyr/pull/71149 supports shortened names.

closed 2024-04-17 by tejlmand