zephyrproject-rtos/zephyr #71130: shell: Crash with different UART instances in different configurations

joerchan, 2024-04-05:
**Describe the bug**

SHELL_BACKEND_SERIAL_API defaults to ASYNC_API when this is supported by the serial driver.
However the instance of UART can be configured to not support the ASYNC operations, and instead be polling or interrupt driven.

Please also mention any information which could help others to understand
the problem you're facing:
- Target platform: nrf9160
- Workaround: Explicitly configure the shell backend type ()
- Regression since shell uart backend async support was added, which changed the default to async over interrupt driven.

```
CONFIG_UART_0_INTERRUPT_DRIVEN=y
CONFIG_UART_3_ASYNC=y

CONFIG_SHELL=y
CONFIG_SHELL_BACKEND_SERIAL=y

/ {
    chosen {
        zephyr,shell-uart = &uart0;
    };
};
```

Will crash here:
```
uarte_nrfx_rx_enable: (data->async = NULL)

	if (data->async->rx_enabled || data->async->discard_rx_fifo) {
		return -EBUSY;
	}
```

**To Reproduce**
Steps to reproduce the behavior:
1. Set the above configuration.
2. Build for board nrf9160 (or any other than uses the nrfx uarte driver)
3. flash
4. Crashes in uarte_nrfx_rx_enable called from async_init

**Expected behavior**
Shell backend should default to the chosen uart instance configuration.
uarte_nrfx_rx_enable should return ENOTSUP

**Impact**
annoyance, need to find a Kconfig to override.

**Logs and console output**
TF-M is enabled, so null-pointer becomes securefault exception.
```
FATAL ERROR: SecureFault
Here is some context for the exception:
    EXC_RETURN (LR): 0xFFFFFFBD
    Exception came from non-secure FW in thread mode.
<omitted register information, not relevant>
    CFSR:  0x00000000
    BFSR:  0x00000000
    BFAR:  Not Valid
    MMFSR: 0x00000000
    MMFAR: Not Valid
    UFSR:  0x00000000
    HFSR:  0x00000000
    SFSR:  0x00000048
    SFAR: 0x000000CA
```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK
 - Commit SHA or Version used: Zephyr 3.5.0

GilDev, 2024-06-28:
Thanks a lot @joerchan!! I lost so much time and was going crazy with my app crashing after enabling BLE on my nRF9160 through HCI on my nRF52840.

It was hard to trace the issue, having currently no logs from the secure part. Using the Debug mode gave me a hint about the shell async init function that triggered the assert, and brought me to your issue.

bjarki-andreasen, 2024-07-11:
The issue is not related to the UARTE device driver :) As the UART API specifies, if a specific instance of a UART device does not support async operations, the call to `uart_callback_set()` shall return -ENOTSUP, which the device driver indeed does:
https://github.com/zephyrproject-rtos/zephyr/blob/f320535f84db3b9bef8c1e975aa2fe61673c95c0/drivers/serial/uart_nrfx_uarte.c#L966-L974

The shell can't dynamically adapt to a UART instance supporting async or interrupt APIs, so it uses asserts to ensure the call to `uart_callback_set()` succeeds.
https://github.com/zephyrproject-rtos/zephyr/blob/f320535f84db3b9bef8c1e975aa2fe61673c95c0/subsys/shell/backends/shell_uart.c#L254-L256
if you have not enabled asserts, instead of getting a clear error message, you get an explosion (crash/hard fault) :)

Enabling asserts and configuring UART0 to be interrupt driven, and the shell to be expect async results in the following:
```
uart:~$ ASSERTION FAIL [err == 0] @ WEST_TOPDIR/zephyr/subsys/shell/backends/shell_uart.c:256
```
In KConfig terms:
```
CONFIG_UART_0_INTERRUPT_DRIVEN=y
CONFIG_SHELL_BACKEND_SERIAL_API_ASYNC=y
```

bjarki-andreasen, 2024-07-11:
Since the specific API supported by any single instance of a UART has to be checked at runtime, (nRF is an outlier here as each individual UART can be configured explicitly to support it), the `shell_uart.c` backend would need to be updated to support either API dynamically (like what was done for https://github.com/zephyrproject-rtos/zephyr/blob/main/include/zephyr/modem/backend/uart.h which I hope we can start to use more places than modem drivers).

closed 2024-07-11 by bjarki-andreasen