zephyrproject-rtos/zephyr #71134: USB Next Mass Storage not working

arnebdr, 2024-04-05:
**Describe the bug**
When running the USB Mass Storage example with USB Next enabled, connection of a Host results in Zehyr Kernel Panic and System Hold.

Tested with STM32U585 on ST BU585-IOT02A board. 
Without USB Next enabled, Mass storage is working fine.

**To Reproduce**
Steps to reproduce the behavior:
1. west build -p always -b b_u585i_iot02a samples/subsys/usb/mass -DEXTRA_DTC_OVERLAY_FILE="ramdisk.overlay" -DCONFIG_APP_MSC_STORAGE_RAM=y -DCONF_FILE=usbd_next_prj.conf
2. west flash
3. Connect USB Host (Windows 11)
4. MSC is not detected. Zephyr crashes. 

**Expected behavior**
Mass Storage device is mounted in windows and write/readable. Zephyr does not crash.

**Impact**
USB Next can't be used with MSC, at least with STM32U585

**Logs and console output**
Console Output:
`*** Booting Zephyr OS build v3.6.0-935-g42c8b7fb00a8 ***
Mount /RAM:: 0
/RAM:: bsize = 512 ; frsize = 512 ; blocks = 158 ; bfree = 158
/RAM: opendir: 0
End of files
[00:00:00.052,000] <inf> main: The device is put in USB mass storage mode.

[00:00:00.058,000] <wrn> usbd_core: SUSPEND event
[00:00:05.858,000] <wrn> udc: Spurious suspend/resume event
[00:00:05.858,000] <wrn> usbd_core: SUSPEND event
[00:00:06.021,000] <wrn> udc: Spurious suspend/resume event
[00:00:06.021,000] <wrn> usbd_core: SUSPEND event
[00:00:06.171,000] <wrn> usbd_core: RESET event
[00:00:06.171,000] <wrn> usbd_core: Bus reset event
[00:00:06.259,000] <wrn> usbd_core: RESET event
[00:00:06.259,000] <wrn> usbd_core: Bus reset event
[00:00:06.467,000] <inf> usbd_msc: Enable
[00:00:06.467,000] <inf> usbd_msc: Bulk-Only Mass Storage Reset
[00:00:06.524,000] <wrn> usbd_core: RESET event
[00:00:06.524,000] <wrn> usbd_core: Bus reset event
[00:00:06.524,000] <inf> usbd_msc: Disable
[00:00:06.524,000] <wrn> usbd_msc: request ep 0x01, len 0 cancelled
[00:00:06.611,000] <wrn> usbd_core: RESET event
[00:00:06.611,000] <wrn> usbd_core: Bus reset event
[00:00:06.655,000] <inf> usbd_msc: Enable
[00:00:06.655,000] <inf> usbd_msc: Bulk-Only Mass Storage Reset
[00:00:06.711,000] <wrn> usbd_core: RESET event
[00:00:06.711,000] <wrn> usbd_core: Bus reset event
[00:00:06.711,000] <inf> usbd_msc: Disable
[00:00:06.711,000] <wrn> usbd_msc: request ep 0x01, len 0 cancelled
[00:00:06.799,000] <wrn> usbd_core: RESET event
[00:00:06.799,000] <wrn> usbd_core: Bus reset event
[00:00:06.844,000] <inf> usbd_msc: Enable
[00:00:06.844,000] <inf> usbd_msc: Bulk-Only Mass Storage Reset
[00:00:06.901,000] <wrn> usbd_core: RESET event
[00:00:06.901,000] <wrn> usbd_core: Bus reset event
[00:00:06.901,000] <inf> usbd_msc: Disable
[00:00:06.901,000] <wrn> usbd_msc: request ep 0x01, len 0 cancelled
[00:00:06.989,000] <wrn> usbd_core: RESET event
[00:00:06.989,000] <wrn> usbd_core: Bus reset event
[00:00:07.032,000] <inf> usbd_msc: Enable
[00:00:07.032,000] <inf> usbd_msc: Bulk-Only Mass Storage Reset
[00:00:07.088,000] <wrn> usbd_core: RESET event
[00:00:07.088,000] <wrn> usbd_core: Bus reset event
[00:00:07.088,000] <inf> usbd_msc: Disable
[00:00:07.088,000] <wrn> usbd_msc: request ep 0x01, len 0 cancelled
[00:00:07.176,000] <wrn> usbd_core: RESET event
[00:00:07.176,000] <wrn> usbd_core: Bus reset event
[00:00:07.220,000] <inf> usbd_msc: Enable
[00:00:07.220,000] <inf> usbd_msc: Bulk-Only Mass Storage Reset
[00:00:07.277,000] <wrn> usbd_core: RESET event
[00:00:07.277,000] <wrn> usbd_core: Bus reset event
[00:00:07.277,000] <inf> usbd_msc: Disable
[00:00:07.277,000] <wrn> usbd_msc: request ep 0x01, len 0 cancelled
[00:00:07.365,000] <wrn> usbd_core: RESET event
[00:00:07.365,000] <wrn> usbd_core: Bus reset event
[00:00:07.409,000] <inf> usbd_msc: Enable
[00:00:07.409,000] <inf> usbd_msc: Bulk-Only Mass Storage Reset
[00:00:07.465,000] <wrn> usbd_core: RESET event
[00:00:07.465,000] <wrn> usbd_core: Bus reset event
[00:00:07.466,000] <inf> usbd_msc: Disable
[00:00:07.466,000] <wrn> usbd_msc: request ep 0x01, len 0 cancelled
[00:00:07.554,000] <wrn> usbd_core: RESET event
[00:00:07.554,000] <wrn> usbd_core: Bus reset event
[00:00:07.597,000] <inf> usbd_msc: Enable
[00:00:07.597,000] <inf> usbd_msc: Bulk-Only Mass Storage Reset
[00:00:07.655,000] <wrn> usbd_core: RESET event
[00:00:07.655,000] <wrn> usbd_core: Bus reset event
[00:00:07.655,000] <inf> usbd_msc: Disable
[00:00:07.655,000] <wrn> usbd_msc: request ep 0x01, len 0 cancelled
[00:00:07.743,000] <wrn> usbd_core: RESET event
[00:00:07.743,000] <wrn> usbd_core: Bus reset event
[00:00:07.787,000] <err> udc_stm32: Unable to allocate FIFO for 0x81
[00:00:07.787,000] <err> usbd_iface: Failed to handle op 1, ep 0x81, bm 0x00000000, -12
[00:00:07.787,000] <err> usbd_ch9: Failed to set configuration 1, -12
[00:00:07.787,000] <err> usbd_core: unrecoverable error -12, ep 0x00, buf 0x2001bd08
[00:00:07.787,000] <err> os: r0/a1:  0x00000004  r1/a2:  0x00000000  r2/a3:  0x20000938
[00:00:07.787,000] <err> os: r3/a4:  0x20000938 r12/ip:  0x00001e6b r14/lr:  0x08001a97
[00:00:07.787,000] <err> os:  xpsr:  0x41000000
[00:00:07.787,000] <err> os: Faulting instruction address (r15/pc): 0x08003b66
[00:00:07.787,000] <err> os: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0
[00:00:07.787,000] <err> os: Current thread: 0x20000938 (unknown)
[00:00:07.896,000] <err> os: Halting system`

**Environment (please complete the following information):**
 - OS:  Windows 11
 - STM32U585

jfischer-no, 2024-04-10:
I can confirm that UDC STM32 shim driver does not work with MSC using HS controller, not sure about FS only.

arnebdr, 2024-04-19:
#71702

arnebdr, 2024-04-19:
Tested again with the latest Zephyr Main Branch. The Kernel Panic is gone now, but enumeration still fails:

`[00:00:00.052,000] <inf> main: The device is put in USB mass storage mode.

[00:00:00.432,000] <inf> usbd_msc: Enable
[00:00:00.432,000] <inf> usbd_msc: Bulk-Only Mass Storage Reset
[00:00:00.489,000] <inf> usbd_msc: Disable
[00:00:00.489,000] <wrn> usbd_msc: request ep 0x01, len 0 cancelled
[00:00:00.620,000] <inf> usbd_msc: Enable
[00:00:00.620,000] <inf> usbd_msc: Bulk-Only Mass Storage Reset
[00:00:00.676,000] <inf> usbd_msc: Disable
[00:00:00.676,000] <wrn> usbd_msc: request ep 0x01, len 0 cancelled
[00:00:00.807,000] <inf> usbd_msc: Enable
[00:00:00.807,000] <inf> usbd_msc: Bulk-Only Mass Storage Reset
[00:00:00.863,000] <inf> usbd_msc: Disable
[00:00:00.863,000] <wrn> usbd_msc: request ep 0x01, len 0 cancelled
[00:00:00.994,000] <inf> usbd_msc: Enable
[00:00:00.994,000] <inf> usbd_msc: Bulk-Only Mass Storage Reset
[00:00:01.050,000] <inf> usbd_msc: Disable
[00:00:01.050,000] <wrn> usbd_msc: request ep 0x01, len 0 cancelled
[00:00:01.180,000] <inf> usbd_msc: Enable
[00:00:01.180,000] <inf> usbd_msc: Bulk-Only Mass Storage Reset
[00:00:01.236,000] <inf> usbd_msc: Disable
[00:00:01.236,000] <wrn> usbd_msc: request ep 0x01, len 0 cancelled
[00:00:01.367,000] <inf> usbd_msc: Enable
[00:00:01.367,000] <inf> usbd_msc: Bulk-Only Mass Storage Reset
[00:00:01.423,000] <inf> usbd_msc: Disable
[00:00:01.423,000] <wrn> usbd_msc: request ep 0x01, len 0 cancelled
[00:00:01.554,000] <inf> usbd_msc: Enable
[00:00:01.554,000] <inf> usbd_msc: Bulk-Only Mass Storage Reset
[00:00:01.611,000] <inf> usbd_msc: Disable
[00:00:01.611,000] <wrn> usbd_msc: request ep 0x01, len 0 cancelled
[00:00:01.742,000] <err> udc_stm32: Unable to allocate FIFO for 0x81
[00:00:01.742,000] <err> usbd_iface: Failed to handle op 1, ep 0x81, bm 0x00000000, -12
[00:00:01.742,000] <err> usbd_ch9: Failed to set configuration 1, -12`

I think the issue is in udc_stm32.c function udc_stm32_ep_mem_config(). The variable priv->occupied_mem is only counted up and not down on disable. This results in error "udc_stm32: Unable to allocate FIFO for 0x81".
With my fix from #71702 enumeration is now working, the flash drive is mounted and can files can be written.
Still I wonder why the Endpoint is Enabled/Disabled 22 Times before it is connected successfull.
See log output with the Fix below:
`[00:00:00.052,000] <inf> main: The device is put in USB mass storage mode.

[00:00:00.435,000] <inf> usbd_msc: Enable
[00:00:00.435,000] <inf> usbd_msc: Bulk-Only Mass Storage Reset
[00:00:00.491,000] <inf> usbd_msc: Disable
[00:00:00.491,000] <wrn> usbd_msc: request ep 0x01, len 0 cancelled
[00:00:00.622,000] <inf> usbd_msc: Enable
[00:00:00.622,000] <inf> usbd_msc: Bulk-Only Mass Storage Reset
[00:00:00.678,000] <inf> usbd_msc: Disable
[00:00:00.678,000] <wrn> usbd_msc: request ep 0x01, len 0 cancelled
[00:00:00.810,000] <inf> usbd_msc: Enable
[00:00:00.810,000] <inf> usbd_msc: Bulk-Only Mass Storage Reset
[00:00:00.867,000] <inf> usbd_msc: Disable
[00:00:00.867,000] <wrn> usbd_msc: request ep 0x01, len 0 cancelled
[00:00:00.998,000] <inf> usbd_msc: Enable
[00:00:00.998,000] <inf> usbd_msc: Bulk-Only Mass Storage Reset
[00:00:01.054,000] <inf> usbd_msc: Disable
[00:00:01.054,000] <wrn> usbd_msc: request ep 0x01, len 0 cancelled
[00:00:01.185,000] <inf> usbd_msc: Enable
[00:00:01.185,000] <inf> usbd_msc: Bulk-Only Mass Storage Reset
[00:00:01.241,000] <inf> usbd_msc: Disable
[00:00:01.241,000] <wrn> usbd_msc: request ep 0x01, len 0 cancelled
[00:00:01.373,000] <inf> usbd_msc: Enable
[00:00:01.373,000] <inf> usbd_msc: Bulk-Only Mass Storage Reset
[00:00:01.430,000] <inf> usbd_msc: Disable
[00:00:01.430,000] <wrn> usbd_msc: request ep 0x01, len 0 cancelled
[00:00:01.563,000] <inf> usbd_msc: Enable
[00:00:01.563,000] <inf> usbd_msc: Bulk-Only Mass Storage Reset
[00:00:01.620,000] <inf> usbd_msc: Disable
[00:00:01.620,000] <wrn> usbd_msc: request ep 0x01, len 0 cancelled
[... 33 lines cut ...]
[00:00:03.252,000] <inf> usbd_msc: Bulk-Only Mass Storage Reset
[00:00:03.327,000] <inf> usbd_msc: Disable
[00:00:03.327,000] <wrn> usbd_msc: request ep 0x01, len 0 cancelled
[00:00:03.458,000] <inf> usbd_msc: Enable
[00:00:03.458,000] <inf> usbd_msc: Bulk-Only Mass Storage Reset
[00:00:03.515,000] <inf> usbd_msc: Disable
[00:00:03.515,000] <wrn> usbd_msc: request ep 0x01, len 0 cancelled
[00:00:03.646,000] <inf> usbd_msc: Enable
[00:00:03.646,000] <inf> usbd_msc: Bulk-Only Mass Storage Reset
[00:00:03.716,000] <inf> usbd_msc: Disable
[00:00:03.716,000] <wrn> usbd_msc: request ep 0x01, len 0 cancelled
[00:00:03.847,000] <inf> usbd_msc: Enable
[00:00:03.847,000] <inf> usbd_msc: Bulk-Only Mass Storage Reset
[00:00:03.903,000] <inf> usbd_msc: Disable
[00:00:03.903,000] <wrn> usbd_msc: request ep 0x01, len 0 cancelled
[00:00:04.033,000] <inf> usbd_msc: Enable
[00:00:04.033,000] <inf> usbd_msc: Bulk-Only Mass Storage Reset
[00:00:04.109,000] <inf> usbd_msc: Disable
[00:00:04.109,000] <wrn> usbd_msc: request ep 0x01, len 0 cancelled
[00:00:04.240,000] <inf> usbd_msc: Enable
[00:00:04.240,000] <inf> usbd_msc: Bulk-Only Mass Storage Reset
[00:00:04.296,000] <inf> usbd_msc: Disable
[00:00:04.296,000] <wrn> usbd_msc: request ep 0x01, len 0 cancelled
[00:00:04.427,000] <inf> usbd_msc: Enable
[00:00:04.427,000] <inf> usbd_msc: Bulk-Only Mass Storage Reset`

PR #71702 (2024-04-23): #71134 USB: Fix for Mass storage not working with USB Next on STM32.
Fix for Issue 71134. USB MSC class enumeration failed with STM32 using USB Next stack, because in function udc_stm32_ep_mem_config(), priv->occupied_mem was only counted up on enable, but not down on endpoint disable.

Fixes: #71134
closed 2024-04-23 by fabiobaltieri