zephyrproject-rtos/zephyr #71142: drivers: can: can_stm32_bxcan.c: Failed to exit sleep mode

adityaganesh2k, 2024-04-05:
**Describe the bug**
For STMF4XX family microcontrollers in the file can_stm32_bxcan.c , inside the can_stm32_init() function the microcontroller always fails to exit the sleep mode when it wakes up due to a software reset .

Please also mention any information which could help others to understand
the problem you're facing:
 - I am using a custom board on STM32F446VET6 
 - Inside the can_stm32_init() function first we exit the sleep mode and the enter the initialization mode . This process works fine if the device has had a power reset . But when device has undergone a software reset the device wakes up with the CAN controller in sleep mode but does not exit the sleep mode and times out . Hence, CAN never gets initialized if you do a software reset.
 - Inside can_stm32_init I have made the following changes . The can_stm32_enter_init_mode(can) is called before can_stm32_leave_sleep_mode(can) instead of the other way around . NOTE: When I went through the HAL_CAN_Init function in the stm32cubeide HAL Drivers I noticed that they too first made the CAN controller go into initialization mode and then Exited the Sleep mode .
 - Is this a regression? No

**To Reproduce**
Steps to reproduce the behavior:
1. To reproduce this issue you can use a nucleo_f446re from the supported board 
2. Use the sample code from samples/drivers/can/babbling 
3. During execution of the code do a software reset using the stm32 debugger(note: in my case I am using CORTEX-M debug extension on vscode)
4. As soon as you reset the code on your zephyr console you should be seeing the following message "Failed to exit sleep mode".
I personally am using a custom board which is not added as a supported board in the zephyr codebase . But the above method should still reproduce the issue.

**Impact**
This issue basically disables CAN whenever a software reset happens and gets only fixed if a power reset is done to the controller

**Environment (please complete the following information):**
 - OS: Linux in a docker
 - Toolchain Zephyr SDK 
 - v3.6-branch
 
**Additional context**
This is the dts for CAN in my custom board
 &can1 {
	 // CAN1 Connection (CAN1_RX - PD0, CAN1_TX - PD1) to CAN Transceiver and Vehicle Network
	 pinctrl-0 = <&can1_rx_pd0 &can1_tx_pd1>;
	 pinctrl-names = "default";
	 bus-speed = <250000>;
	 status = "okay";
 };

erwango, 2024-04-05:
Adding @henrikbrixandersen to comment on the fix proposal:
> The can_stm32_enter_init_mode(can) is called before can_stm32_leave_sleep_mode(can) instead of the other way around

henrikbrixandersen, 2024-04-05:
Hi @adityaganesh2k! Thank you for this detailed bug report.

I will do some local testing and provide a fix based on your proposal - unless you wish to provide the fix PR yourself?

adityaganesh2k, 2024-04-05:
I have fixed the bug, will raise the PR myself.

adityaganesh, 2024-04-05:
Raised PR here #71164

PR #71164 (2024-04-11): drivers: can: can_stm32_bxcan.c: Failed to exit sleep mode
Fixes the issue of CAN failing to exit sleep mode inside can_stm32_init() when a software reset is done

Fixes: #71142
closed 2024-04-11 by aescolar