zephyrproject-rtos/zephyr #71154: samples: nrfx_prs: Failing after nrfx update

joerchan, 2024-04-05:
**Describe the bug**
The sample nrfx_prs appears to not have been updated after recent nrfx updates.
Following the sample I'm unable to communicate with my SPI device.

Sample was last working in NCS 2.4.x (nrfx 2.11.0).

In recent versions this fails even in init.
This looks to be just because the init function expects raw integer as frequency instead of NRF defines.
However with that change the transfer still fails, so probably more changes are needed in the sample.
There appears to be a different behavior in zephyr main branch compared to NCS 2.6.0 release.

**To Reproduce**
Steps to reproduce the behavior:
NCS 2.6.0
1. west build -b  nrf5340dk_nrf5340_cpuapp samples/boards/nrf/nrfx_prs
2. west flash --erase
3. Push button 2

```
*** Booting nRF Connect SDK v3.5.99-ncs1 ***
nrfx PRS example on nrf5340dk_nrf5340_cpuapp
-> press "Push button 1" to trigger a transfer
-> press "Push button 2" to switch the type of peripheral
Switched to SPIM
-- SPIM transfer --
Tx: 00 01 02 03 04 05 06 07 08 09
Rx: 00 00 00 00 00 00 00 00 00 00
nrfx_uarte_init() failed: 0x0bad0001
```

Zephyr commit cd685bdce82d.
```
*** Booting Zephyr OS build v3.6.0-2198-gcd685bdce82d ***
nrfx PRS example on nrf5340dk
-> press "Push button 1" to trigger a transfer
-> press "Push button 2" to switch the type of peripheral
nrfx_spim_init() failed: 0x0bad0004
```

**Expected behavior**
Sample should work.

**Impact**
SPI driver not working when sharing the serial box.

joerchan, 2024-04-08:
Found the second problem I was having. Since I'm using SPI mode 3 the SCK needs to be set, this is missing from the sample.

nashif, 2024-04-09:
This is not a zephyr upstream issue, please do not file issues of NCS upstream(zephyr), file those downstream (NCS).

joerchan, 2024-04-10:
> This is not a zephyr upstream issue, please do not file issues of NCS upstream(zephyr), file those downstream (NCS).

It is a zephyr upstream issue. I've reproduced it first in NCS, and then in zephyr commit cd685bdce82d1d3843bcd56b67a086278e59e3e5.

PR #71157 (2024-04-12): samples: nrfx_prs: Fix sample frequency and baudrate
Fix sample frequency and baudrate.
The nrfx drivers are now using raw integere values instead of nrf definitions.

Add handling of SCK initial value based on the SPI mode used.
If mode is using CPOL then SCK needs to be set.

Fixes: #71154
commit d97c780f (2024-04-12): samples: nrfx_prs: Add handling of SCK initial value based on mode

Add handling of SCK initial value based on the SPI mode used.
If mode is using CPOL then SCK needs to be set.

Fixes: #71154
commit e9cf74b5 (2024-04-12): samples: nrfx_prs: Fix sample frequency

Fix setting of sample frequency.
The nrfx drivers are now using raw integere values instead of nrf
definitions.

Fixes: #71154
closed 2024-04-12 by carlescufi