zephyrproject-rtos/zephyr #71190: Enable simulator to be selected from twister's command line

ithinuel, 2024-04-07:
**Is your enhancement proposal related to a problem? Please describe.**
-

Switching between QEMU and another simulator (such as Arm FVP) requires editing the board yaml file.

**Describe the solution you'd like**
-

I'd like to have an argument to twister to select the simulator, similarly to what `--west-runner` does.

**Describe alternatives you've considered**
-

For now the solution is to patch zephyr in CI before running the tests.
It may be possible to do something with CMake's property to alter a yaml file (like it is possible to configure tfm's hex file name before flashing with pyocd).

**Additional context**
-

-

henrikbrixandersen, 2024-04-07:
Can you give an example of a board and the needed patch for switching between the two? Perhaps a board variant could solve this.

ithinuel, 2024-04-07:
The gist of the patch more or less what https://docs.zephyrproject.org/latest/develop/test/twister.html#running-tests-on-custom-emulator.

For example, on Corstone-300 (aka `mps3/an547`) [this line](https://github.com/zephyrproject-rtos/zephyr/blob/main/boards/arm/mps3/board.cmake#L12C5-L12C28) does enable `run_armfvp` but without the changes to `simulation` and `simulation_exec` in [`boards/arm/mps3/mps3_an547.yaml`](https://github.com/zephyrproject-rtos/zephyr/blob/main/boards/arm/mps3/mps3_an547.yaml), `west twister` won't run those tests on FVP.
But with those changes, they can no longer be run on qemu.