zephyrproject-rtos/zephyr #71191: TCP over subg on BeagleconnectFreedom is unusable

Ayush1325, 2024-04-07:
**Describe the bug**
After investigating UDP over subg on beagleconnect freedom (https://github.com/zephyrproject-rtos/zephyr/issues/68674), I have moved to testing TCP. However, I cannot seem to get TCP reliably working on the `echo_server` and `echo_client`.

Note: The subg overlays present in `echo_server` and `echo_client` disable TCP by default, so to reproduce these results, it should be enabled.

Here are the logs from `echo_server`:
```
*** Booting Zephyr OS build v3.6.0-1935-gf87143d9198a ***
[00:00:00.045,104] <inf> net_config: Initializing network
[00:00:00.155,487] <inf> net_config: IPv6 address: 2001:db8::1
[00:00:00.157,012] <inf> net_echo_server_sample: Run echo server
[00:00:00.157,073] <inf> net_echo_server_sample: Network connected
[00:00:00.157,135] <inf> net_echo_server_sample: Starting...
[00:00:00.157,592] <inf> net_echo_server_sample: Waiting for TCP connection on port 4242 (IPv6)...
[00:00:03.049,682] <inf> net_echo_server_sample: TCP (IPv6): Accepted connection
[00:00:03.050,506] <inf> net_echo_server_sample: Waiting for TCP connection on port 4242 (IPv6)...
[00:00:03.173,492] <dbg> net_echo_server_sample: handle_data: TCP (IPv6): Received and replied with 65 bytes
[00:00:03.715,881] <dbg> net_echo_server_sample: handle_data: TCP (IPv6): Received and replied with 65 bytes
[00:00:03.739,776] <wrn> ieee802154_cc13xx_cc26xx_subg: Cannot allocate packet
[00:00:03.755,676] <wrn> ieee802154_cc13xx_cc26xx_subg: Cannot allocate packet
[00:00:03.755,676] <wrn> ieee802154_cc13xx_cc26xx_subg: Cannot allocate packet
[00:00:03.773,010] <wrn> ieee802154_cc13xx_cc26xx_subg: Cannot allocate packet
[00:00:03.773,010] <wrn> ieee802154_cc13xx_cc26xx_subg: Cannot allocate packet
[00:00:03.773,040] <wrn> ieee802154_cc13xx_cc26xx_subg: Cannot allocate packet
[00:00:03.790,039] <wrn> ieee802154_cc13xx_cc26xx_subg: Cannot allocate packet
[00:00:03.790,069] <wrn> ieee802154_cc13xx_cc26xx_subg: Cannot allocate packet
[00:00:03.790,100] <wrn> ieee802154_cc13xx_cc26xx_subg: Cannot allocate packet
[00:00:03.790,130] <wrn> ieee802154_cc13xx_cc26xx_subg: Cannot allocate packet
[00:00:03.804,321] <wrn> ieee802154_cc13xx_cc26xx_subg: Cannot allocate packet
[00:00:03.804,351] <wrn> ieee802154_cc13xx_cc26xx_subg: Cannot allocate packet
[00:00:03.804,351] <wrn> ieee802154_cc13xx_cc26xx_subg: Cannot allocate packet
[00:00:03.804,382] <wrn> ieee802154_cc13xx_cc26xx_subg: Cannot allocate packet
[00:00:03.809,295] <wrn> ieee802154_cc13xx_cc26xx_subg: Cannot allocate packet
[00:00:03.809,295] <wrn> ieee802154_cc13xx_cc26xx_subg: Cannot allocate packet
[00:00:03.809,326] <wrn> ieee802154_cc13xx_cc26xx_subg: Cannot allocate packet
[00:00:03.809,356] <wrn> ieee802154_cc13xx_cc26xx_subg: Cannot allocate packet
[00:00:03.809,356] <wrn> ieee802154_cc13xx_cc26xx_subg: Cannot allocate packet
[00:00:03.958,648] <dbg> net_echo_server_sample: handle_data: TCP (IPv6): Received and replied with 130 bytes
[00:00:04.676,635] <dbg> net_echo_server_sample: handle_data: TCP (IPv6): Received and replied with 8 bytes
[00:00:21.128,753] <err> net_echo_server_sample: TCP (IPv6): Connection error 116
[00:01:00.157,775] <inf> net_echo_server_sample: IPv6 TCP: Received 4 B/sec
```

And here are the logs from `echo_client`:
```
*** Booting Zephyr OS build v3.6.0-1935-gf87143d9198a ***
[00:00:00.046,569] <inf> net_config: Initializing network
[00:00:00.164,703] <inf> net_config: IPv6 address: 2001:db8::2
[00:00:00.164,825] <inf> net_echo_client_sample: Run echo client
[00:00:00.164,886] <inf> net_echo_client_sample: Network connected
[00:00:00.164,947] <inf> net_echo_client_sample: Starting...
[00:00:00.427,185] <dbg> net_echo_client_sample: send_tcp_data: IPv6 TCP: Sent 268 bytes
[00:00:01.144,378] <wrn> ieee802154_cc13xx_cc26xx_subg: Cannot allocate packet
[00:00:01.155,883] <wrn> ieee802154_cc13xx_cc26xx_subg: Cannot allocate packet
[00:00:01.155,883] <wrn> ieee802154_cc13xx_cc26xx_subg: Cannot allocate packet
[00:00:01.162,139] <wrn> ieee802154_cc13xx_cc26xx_subg: Cannot allocate packet
[00:00:01.162,170] <wrn> ieee802154_cc13xx_cc26xx_subg: Cannot allocate packet
[00:00:01.178,253] <wrn> ieee802154_cc13xx_cc26xx_subg: Cannot allocate packet
[00:00:01.178,283] <wrn> ieee802154_cc13xx_cc26xx_subg: Cannot allocate packet
[00:00:01.178,283] <wrn> ieee802154_cc13xx_cc26xx_subg: Cannot allocate packet
[00:00:01.202,453] <wrn> ieee802154_cc13xx_cc26xx_subg: Cannot allocate packet
[00:00:01.202,453] <wrn> ieee802154_cc13xx_cc26xx_subg: Cannot allocate packet
[00:00:01.202,484] <wrn> ieee802154_cc13xx_cc26xx_subg: Cannot allocate packet
[00:00:01.202,514] <wrn> ieee802154_cc13xx_cc26xx_subg: Cannot allocate packet
[00:00:01.220,947] <wrn> ieee802154_cc13xx_cc26xx_subg: Cannot allocate packet
[00:00:01.220,947] <wrn> ieee802154_cc13xx_cc26xx_subg: Cannot allocate packet
[00:00:01.220,977] <wrn> ieee802154_cc13xx_cc26xx_subg: Cannot allocate packet
[00:00:01.220,977] <wrn> ieee802154_cc13xx_cc26xx_subg: Cannot allocate packet
[00:00:01.221,008] <wrn> ieee802154_cc13xx_cc26xx_subg: Cannot allocate packet
[00:00:01.243,560] <wrn> ieee802154_cc13xx_cc26xx_subg: Cannot allocate packet
[00:00:01.243,560] <wrn> ieee802154_cc13xx_cc26xx_subg: Cannot allocate packet
[00:00:01.243,591] <wrn> ieee802154_cc13xx_cc26xx_subg: Cannot allocate packet
[00:00:01.243,591] <wrn> ieee802154_cc13xx_cc26xx_subg: Cannot allocate packet
[00:00:01.243,621] <wrn> ieee802154_cc13xx_cc26xx_subg: Cannot allocate packet
[00:00:01.243,652] <wrn> ieee802154_cc13xx_cc26xx_subg: Cannot allocate packet
[00:00:17.344,635] <inf> net_echo_client_sample: Stopping...
```

Some additional config options I tried adding to fix this are as follows:
```
CONFIG_NET_L2_IEEE802154_FRAGMENT_REASS_CACHE_SIZE=32
CONFIG_IEEE802154_CC13XX_CC26XX_SUB_GHZ_NUM_RX_BUF=6
```

- Target: beagleconnect_freedom

**To Reproduce**
Steps to reproduce the behavior:

- Enable TCP in both server and client subg overlay
- Build and flash echo_server to one beagleconnect freedom
```
west build -b beagleconnect_freedom zephyr/samples/net/sockets/echo_server -p -- -DEXTRA_CONF_FILE=overlay-802154-subg.conf && west flash
```
- Build and flash echo_client to the other beagleconnect freedom
```
west build -b beagleconnect_freedom zephyr/samples/net/sockets/echo_client -p -- -DEXTRA_CONF_FILE=overlay-802154-subg.conf && west flash
```
- Open serial connection.

**Expected behavior**
All packets to be received by the server.

**Environment (please complete the following information):**

- OS: Linux
- Toolchain Zephyr SDK 0.16.5
- Zephyr Commit SHA: f87143d9198a326da5d6cbc58290d24cd4546fec

fgrandel, 2024-04-07:
It would be useful to see sniffer data. Any TI Dev Board can be transformed into a sniffer via their PacketSniffer 2 firmware. Analyzing the wireshark protocols could help us identify the root cause.

Ayush1325, 2024-04-08:
> It would be useful to see sniffer data. Any TI Dev Board can be transformed into a sniffer via their PacketSniffer 2 firmware. Analyzing the wireshark protocols could help us identify the root cause.

Hi. Is there a way to use PacketSniffer 2 in Linux? I can only see stuff for windows.

fgrandel, 2024-04-08:
> Hi. Is there a way to use PacketSniffer 2 in Linux? I can only see stuff for windows.

I'm afraid that's true - several of TI's proprietary tools do not exist for the Linux platform. I use them in a Windows VM running on Linux which works well. As you're in the SubG range and TI's SUN SubG PHY settings are not 100% standard conforming, you're pretty much depending on their tools currently AFAIK. Of course you could use Zephyr's low-level IEEE 802.15.4 APIs to write your own Sniffer in Zephyr but that's certainly more work than setting up a quick Windows VM.

Ayush1325, 2024-04-09:
> Re your TCP problems - it might just be that you don't get responses reliably, especially if you use ALOHA without ACK switched off. CSMA/CA and ACK are there for a reason... You'll see traces of collisions and missing packages when sniffing or debugging the TCP protocol details directly.

The initial testing is being done with CSMA/CA and ACK since I was not really testing the performance in this issue (it has to work reliably before worrying about performance).

I will try sniffing and a git bisect to pin point the problem.

fgrandel, 2024-04-09:
> The initial testing is being done with CSMA/CA and ACK since I was not really testing the performance in this issue (it has to work reliably before worrying about performance).

Totally agree.

> I will try sniffing and a git bisect to pin point the problem.

Nice. You may of course contact me any time on Discord as well if there's something you think I could help you with. I'd really love to see this hardware to be better supported.

Ayush1325, 2024-05-17:
So did a git bisect and found that the commit responsible is most likely 128354ae17fb95ec6c61d73b9060f1a274715464 .

However, it seems that the problem is a bit different from what I initially considered. What seems to be happening is that for some reason the driver does not recover if it runs out of available rx and/or tx packets. To speed up testing, I had to set `CONFIG_NET_L2_IEEE802154_FRAGMENT_REASS_CACHE_SIZE=8`, but not setting it just delays the problem.

Ayush1325, 2024-05-22:
So #72977 does seem to prevent timeout errors, but it is not a complete fix. While it does not time out now, it still stops in under 2 mins. The error is as follows:
```
[00:01:30.354,797] <err> net_echo_server_sample: TCP (IPv6): Failed to send, closing socket
```

fgrandel, 2024-05-23:
@Ayush1325 Puh - sorry to hear that. Sounds like we had to go back to digging deeper in the code. Debugging with a breakpoint on the non-zero error code path would give you a good indication where this comes from as you see the stack then. Probably you'll have to do several rounds to track it across threads. Can I help something to make this easier for you? (And thanks again for investing all that debugging time here, very much appreciated!)

Ayush1325, 2024-06-14:
I am reopening the issue since as mentioned in the previous comment, while I am no longer geting receive timeout errors, TCP is still not usable due errors while sending.

Ayush1325, 2024-07-08:
I have found the reliability for long running tasks has recently improved https://github.com/zephyrproject-rtos/zephyr/pull/73488
Will close this once that is merged.

fgrandel, 2024-07-24:
@Ayush1325 Can you check please, whether this issue is still required and close otherwise?

closed 2024-07-24 by Ayush1325