zephyrproject-rtos/zephyr #71196: subsys:fs:fs.c : Add mounting option to fs_mount() for fatfs

adityaganesh2k, 2024-04-07:
**Is your enhancement proposal related to a problem? Please describe.**
The fatfs function f_mount has a parameter initialization option which describes the mounting option for the drive . This option as of now is hardcoded to "1" inside fatfs_mount (subsys/fs/fatfs.c). This does not give user the ability to choose between different mounting options . I am using a custom board based of stm32f446vet6 and we constantly face issues where mounting fails as the codebase does a forced mounting everytime . So we had no option but to physically hardcode it to "0" after which this issue got resolved

**Describe the solution you'd like**
The structure fs_mount_t (File system mount info structure) which is passed as a parameter to fs_mount should have a separate parameter named mounting option which can be then passed by the user through the application code . This parameter then should be passed to f_mount function inside fatfs_mount which resolves this issue.

**Describe alternatives you've considered**
For now I have hardcoded the mounting option value to 0 for my use case. 

**Additional context**
[image]
Here, you can see in the above image in line 428 the f_mount has mounting option set as 1 .

adityaganesh, 2024-04-08:
@de-nordic can I create a pr for this with the above stated method?

de-nordic, 2024-04-08:
> @de-nordic can I create a pr for this with the above stated method?

Yeah. Sure.

I am not sure whether extending the structure is enough as now what will happen is that every existing usage of the structure will land some default (probably 0) or a random value in the field that is not filled in by user.

we should probably consider of addition of fs_mount_ex type call that will understand the extra part of structure or take such parameter directly.
Note also that in this case you have to deal with int but other fa may take other type or even several parameters.

Think how can you address issues like mention above.

adityaganesh, 2024-04-10:
I will think of a solution keeping in mind the suggestions you have provided and create a PR