zephyrproject-rtos/zephyr #71203: Broken behaviour for boards with revision overlays

Oleh-Kravchenko, 2024-04-08:
Starting from commit 3eaab029ec6ad3180e0d7049dd38ad66595c3943
Zephyr doesn't collect all overlays as expected and described in documentation.

**To Reproduce**
1. create overlay for specific revision in <BOARD_DIR>, application 'boards' directory.
2. configure the project
3. see the issue

**Expected behavior**
https://docs.zephyrproject.org/latest/build/dts/howtos.html#set-devicetree-overlays

```
-- Found BOARD.dts: <PROJECT>/app/boards/arm/<BOARD>/<BOARD>.dts
-- Found devicetree overlay: <PROJECT>/app/boards/arm/<BOARD>/<BOARD>_<REVISION>.overlay
-- Found devicetree overlay: <PROJECT>/app/boards/<BOARD>.overlay
-- Found devicetree overlay: <PROJECT>/app/boards/<BOARD>_<REVISION>.overlay
```

**Impact**
showstopper

**Environment (please complete the following information):**
 - OS: MacOS
 - Toolchain Zephyr SDK
 - 3eaab029ec6ad3180e0d7049dd38ad66595c3943

nordicjm, 2024-04-08:
Cannot reproduce:
```
-- Found BOARD.dts: /tmp/aa/zephyr/boards/nordic/nrf9160dk/nrf9160dk_nrf9160.dts
-- Found devicetree overlay: /tmp/aa/zephyr/boards/nordic/nrf9160dk/nrf9160dk_nrf9160_0_14_0.overlay
-- Found devicetree overlay: /tmp/aa/zephyr/samples/hello_world/boards/nrf9160dk_nrf9160.overlay
-- Found devicetree overlay: /tmp/aa/zephyr/samples/hello_world/boards/nrf9160dk_nrf9160_0_14_0.overlay
```

Oleh-Kravchenko, 2024-04-08:
@nordicjm,
With latest main branch the issue is not reproduced,
but what to do with latest release?
Is there a fix for it?

nordicjm, 2024-04-08:
See if https://github.com/zephyrproject-rtos/zephyr/pull/71212 fixes the issue for you

Oleh-Kravchenko, 2024-04-08:
@nordicjm,
Change https://github.com/zephyrproject-rtos/zephyr/pull/71212 fixes the issue.
Please merge ASAP!

nordicjm, 2024-06-17:
Has been fixed

PR #71212 (2024-04-26): [v3.6-branch] cmake: modules: extensions: Fix missing board revision overlays
Fixes an issue whereby board overlays for board revisions were not included when configuring applications

Fixes #71203
closed 2024-06-17 by nordicjm