zephyrproject-rtos/zephyr #71211: RFC: drivers: can: change signature of `can_get_{min,max}_bitrate()`

henrikbrixandersen, 2024-04-08:
## Introduction

### Problem description

Since the minimum/maximum supported bitrates are now stored in the common CAN controller driver configuration struct, retrieving these can no longer fail, yet the function calls return whether the call succeeded (it always does) and requires the bitrate limit to be returned via a supplied pointer.

### Proposed change

Change the signatures of the can_get_min_bitrate() and can_get_max_bitrate() to directly return the minimum/maximum supported bitrate.

## Detailed RFC

Please see PR #71209.

### Proposed change (Detailed)

Please see PR #71209.

### Dependencies

None.

### Concerns and Unresolved Questions

None.

## Alternatives

Leave it as is.

henrikbrixandersen, 2024-04-08:
CC: @str4t0m, @martinjaeger, @alexanderwachter

martinjaeger, 2024-04-08:
Fine with me, makes sense.

nashif, 2024-04-09:
> ## Alternatives
> Leave it as is.

what can't this be deprecated, i.e. introduce a new API with a new name and deprecate the old one? This is also an alternative? Any reason it is not being listed?

nashif, 2024-04-09:
> That's an option too.
> 
> I didn't list it as I don't believe this change has a wide enough impact to have to go through a more or less artificial rename just to change the signature of a couple along driver API functions (in reality one, as `can_get_min_bitrate()` was introduced in this development cycle).
> 
> Naming stuff is hard enough without having to come up with new names when optimizing behavior.

Well, this should always be the first option to look at and we will go into a breaking change only and only when deprecation is not possible. 

> "...just to change the signature..."

We need to absorb this and bite the bullet and avoid breaking APIs as much as possible, nevermind how minor it looks.

> "... artificial rename ..."

how about `can_{min,max}_bitrate_get()`, not that artificial :)

carlescufi, 2024-04-09:
> > "... artificial rename ..."
> 
> how about `can_{min,max}_bitrate_get()`, not that artificial :)

We really really need to choose between <object>_<predicate> and <predicate>_<object> and stick to it. "Playing games"  with word order is not sustainable and will lead to confusion IMHO.

henrikbrixandersen, 2024-04-09:
> > > "... artificial rename ..."
> 
> > 
> 
> > how about `can_{min,max}_bitrate_get()`, not that artificial :)
> 
> 
> 
> We really really need to choose between <object>_<predicate> and <predicate>_<object> and stick to it. "Playing games"  with word order is not sustainable and will lead to confusion IMHO.

I agree.

I could introduce `can_get_bitrate_{min,max}()` and deprecate the existing, if that is preferred. I am not sure about the value added by doing so, though.

nashif, 2024-04-09:
"Playing games" ?
So your preference is breaking APIs instead? My main concern here is that we do not even consider deprecation at this point and we jump to breaking stable APIs just because we are shy of introducing a new name.

henrikbrixandersen, 2024-04-09:
> So your preference is breaking APIs instead? My main concern here is that we do not even consider deprecation at this point and we jump to breaking stable APIs just because we are shy of introducing a new name.

I believe we cannot make a one-size-fits-all statement for when to "rename and deprecate" and when to "break". It will always be a case-by-case decision. For this particular change, my preference was to to "break" since I don't expect many - if any - out-of-tree uses of the `can_get_max_bitrate()` API function (and the `can_get_min_bitrate()` function was introduced in this development cycle).

I have raised this proposal as an RFC. You responded to the RFC, stating you would prefer a "rename and deprecate" solution instead. Your review is appreciated. This is exactly why we have the RFC process; to collect comments and alternatives to the proposed change.

I will take your suggestions into consideration and rework the associated PR to add two new API functions (`can_get_bitrate_{min, max}()`) and deprecate the existing ones instead.

henrikbrixandersen, 2024-04-09:
> I will take your suggestions into consideration and rework the associated PR to add two new API functions (`can_get_bitrate_{min, max}()`) and deprecate the existing ones instead.

I have updated the PR (#71209).

closed 2024-04-11 by nashif