zephyrproject-rtos/zephyr #71235: RFC: Sensor YAML

teburd, 2024-04-08:
## Introduction

Sensors need structured metadata associated with them to be able to be able to do things
statically at build time.

### Problem description

Statically defining things like buffers that allow for direct reading of a subset of enabled sensor channels with varying sample bitwidths along with metadata requires an understanding at build time
of these attributes.

Some sensors may have a large number of possible channels and channel configurations.

### Proposed change

Akin to board.yml have each sensor driver provide a sensor.yml with metadata about available channels and sample sizings allowing for C defines and macros to be generated that provide the
metadata at build time for reading subsets of channels.

## Detailed RFC

An earlier attempt at doing this in devicetree failed to find a foothold as it..

* Required usage of subtree includes
* Still would have required custom device tree macros to obtain the desired information.

The goal is to be able to have a macro akin to 

SENSOR_BUF_SIZE(sensor_model, ...) which takes a set of tokens representing a channel and configuration and provide an exact buffer size needed to hold the data for that channel subset and configuration.

E.g.

```c 
#define BUF_SIZE SENSOR_BUF_SIZE(SENSOR_ICM42688, SENSOR_CHANNEL(SENSOR_CHAN_ACCEL_XYZ, 0 ICM42688_ACCEL_20BIT), SENSOR_CHANNEL(SENSOR_CHAN_GYRO_XYZ, 0, ICM42688_GYRO_20BIT), SENSOR_CHANNEL(SENSOR_CHAN_TEMP, 0, ICM42688_TEMP_16BIT))

static uint8_t icm42688_buf[BUF_SIZE]

sensor_read(icm42688_iodev, r, icm42688_buf, ARRAY_SIZE(icm42688_buf));
```

### Proposed change (Detailed)

A sensor.yml in each sensor driver directory, with a yet to be defined schema associated with it, containing all the necessary metadata to successfully provide, at compile time and optionally run time, buffer sizes for specific channel configurations. Along with a new python build script which takes said yaml file and generates appropriate C defines which a common set of macros to inspect the sensor metadata.

## Alternatives

Devicetree was initially tried for this task and had some unfortunately downsides due to limitations around both the ability to inject subtrees in dts as an inability to reuse C defines in dts schema yaml files.

https://github.com/zephyrproject-rtos/zephyr/pull/65830
https://github.com/zephyrproject-rtos/zephyr/pull/71198

teburd, 2024-04-08:
Created this RFC after a discussion in the sensor wg today, April 8 2024 with below attendees in agreement.
@ubieda @laurenmurphyx64 @yperess @MaureenHelm

nordicjm, 2024-04-09:
Would an expansion of/plugin for https://github.com/zephyrproject-rtos/zephyr/pull/68127 help for this? How would this work with sensors that support different levels of precision? E.g. an adc might support 8-bit, 10-bit and 12-bit output, maybe that is set at compile time with a Kconfig, but maybe it set at runtime (and an application might switch dynamically depending upon power/system state)

jukkar, 2024-04-09:
Hi @teburd, in #68127 I created a way to describe system network interfaces in a yaml file, then have some Cmake tooling to use that yaml file to generate a C header file to be used in network configuration library to setup network interfaces when device boots.

If you are interested, the tooling part in that PR could be used to generate the sensor configuration data from a yaml file. You would need to create a python script that could convert the yaml file to some other format usable in runtime.

teburd, 2024-04-09:
> Would an expansion of/plugin for #68127 help for this? How would this work with sensors that support different levels of precision? 

I definitely think that would be worth a look, @yperess hacked around and already got something working in less than a day doing most of what I think we want.

> E.g. an adc might support 8-bit, 10-bit and 12-bit output, maybe that is set at compile time with a Kconfig, but maybe it set at runtime (and an application might switch dynamically depending upon power/system state), or different precisions for the same adc device but on a per-channel basis

In that scenario you could allocate to the largest buffer size needed for the largest configuration you want at compile time and be done. Or allocate dynamically on the stack. Or allocate dynamically from a heap. Or dynamically allocate using the rtio mempool allocator. That's a feature of the new API, how memory is allocated for reading these things comes from the application now and not buried in the driver.

The most common use case I've had myself, and I'd guess many others have is likely an "on" mode and perhaps a low power/off mode. E.g. I already know my sample size and rate and the channels I want to read from at build time because the design of the firmware requires it. So being able to statically allocate the buffers for that to be the exact right size is very useful.

The above macro includes a device specific sample size format. I believe having a SENSOR_SAMPLE_SIZE() type macro here that takes the device type, channel, and a device specific configuration TOKEN.

The size of the buffer can then be sorted out at build time. There's other neat things I think we had in mind with this data as well.

The sensor shell/samples/tests ideally would know what channels are available for which sensors. That's not really possible today, we instead have a sensor shell sample that attempts to read all the channels, samples per sensor type, etc.

teburd, 2024-04-09:
> Hi @teburd, in #68127 I created a way to describe system network interfaces in a yaml file, then have some Cmake tooling to use that yaml file to generate a C header file to be used in network [...]

I definitely think its worth looking at, @yperess has volunteered to tackle this so its up to Yuval in the end how to approach it.

yperess, 2025-03-10:
I promised during the Sensor WG to provide a high level view of the short term benefits/impact of the sensor YAML work, so here it is :)

# Benefits
1. Remove conflicting attribute/channel/trigger values - currently each sensor can create new enums for their own attributes/channels/triggers. The issue is that any 2 sensors will end up with the same value which means that it's easy for the user to pass the channel along with the wrong `const struct device*`. The features in the sensor YAML work address this by flattening all the attributes/channels/triggers.
2. Static checks - sensors that opt into the sensor YAML will be able to run static checks on their attributes and capabilities. This means that when an app gets a sensor device using something like `DT_ALIAS(lid_accel)` it will also be able to assert that the same node actually provides `SENSOR_CHAN_ACCEL_XYZ`.

# Migration plan
1. Discussed in the sensor WG we don't want a repeat of the ICM driver. So we'll only start with the `bma4xx` and `bmi323` drivers. Why? These drivers are used on Chrome OS's trulo and fatcat projects so they'll be continuously tested.
2. Anyone that wants to also use this feature can opt in by just upstreaming a `sensor.yaml` to the driver they want to use.

## Enabling the feature
2 options exist for how this feature will be enabled: (1) using a kconfig or (2) just using it and having to manually roll back if we choose to go in a different direction.

I recommend the second option. The reason is that it adds less burden on the reviewers and the manual rollback is very simple (just copy the content of a generated file back into `sensor.h`.

### Kconfig
Using a kconfig seems like a natural way to gate a new feature, but since this is done at build time and crosses the boundary of C it will lead to an overhead. If used, it will look like:
1. Add a new `CONFIG_SENSORS_USE_YAML_DESCRIPTORS` which will enable the feature
2. When enabled, we'll generate the `zephyr/generated/sensor_constants.h` but the enum in `sensor.h` will have to still be there wrapped with `#ifdef`
   ```c
   #ifndef CONFIG_SENSORS_USE_YAML_DESCRIPTORS
   enum sensor_channel {
   ...
   };
   enum sensor_attribute {
   ...
   };
   enum sensor_trigger {
   ...
   };
   #else
   #include <zephyr/generated/sensor_constants.h>
   #endif
   ```

#### Benefits
1. If we want to abort, it's as easy as removing the `ifndef` wrapper`

#### Shortcomings
1. If we want to add a new attribute/channel/trigger they'll need to be added in 2 places: (1) in the sensor.h and (2) in the yaml descriptor. Reviewers will have to be vigilant to make sure that things are added in both places.

### all-in
The current model allows mixing drivers that don't provide a `sensor.yaml` file with ones that do. So "all-in" really just means that we don't replicated the generated enums. In this model we rely on the script to create the generated enums and require new enums to be added into the YAML specifier. Doing so is relatively simple, a new entry just needs to provide a `name`, `description`, and `units` (when applicable). These would normally be required anyway in the form of a new enum entry and doxygen comment.

#### Benefits
1. Simplicity. New changes are done in one place and reviewers don't have to keep track of this feature.
2. Better feature testing. We will get more feedback by having this be the de-facto way of doing things

#### Shortcomings
1. If we decide that the sensor.yaml solution wasn't right for Zephyr rolling back would involve building locally and copying the content of the generated `sensor_constants.h` back into `zephyr/drivers/sensor.h`.

# Next steps
- [ ] Update https://github.com/zephyrproject-rtos/zephyr/pull/83077 to only support `bma4xx` and `bmi323`.
- [ ] Write a migration guide so people know how to write the `sensor.yaml`
- [ ] Start using and adding features to the sensor.yamls
- [ ] During 4.3, decide if we want to keep this format.