zephyrproject-rtos/zephyr #71240: boards/m5stack/m5stack_atom_lite cannot be build - board file ending wrong

Trifunik, 2024-04-08:
**Describe the bug**
It is not possible to build any sample with m5stack_atom_lite board. 

**To Reproduce**
$ west build -b m5stack_atom_lite/esp32/procpu samples/hello_world 
-- west build: generating a build system
Loading Zephyr default modules (Zephyr base).
-- Application: /home/trifu/Projekte/ZephyrOS/zephyrproject/zephyr/samples/hello_world
-- CMake version: 3.25.1
-- Found Python3: /home/trifu/Projekte/ZephyrOS/zephyrproject/.venv/bin/python3 (found suitable version "3.11.2", minimum required is "3.8") found components: Interpreter 
-- Cache files will be written to: /home/trifu/.cache/zephyr
-- Zephyr version: 3.6.99 (/home/trifu/Projekte/ZephyrOS/zephyrproject/zephyr)
-- Found west (found suitable version "1.2.0", minimum required is "0.14.0")
No board named 'm5stack_atom_lite' found.

**Proposed change**
Change board.yaml to board.yml in zephyr/boards/m5stack/m5stack_atom_lite.

**NOTE**
This file ending slipped through the automatic check. If possible the automatic compliance check should be adapted to the new hardware model.

nordicjm, 2024-04-09:
Can you submit a fix for this?

Trifunik, 2024-04-09:
Yes, I will.

PR #71292 (2024-04-10): boards: m5stack: m5stack_atom_lite file ending typo fixed
Change board.yaml to board.yml in zephyr/boards/m5stack/m5stack_atom_lite. Fixes: #71240