zephyrproject-rtos/zephyr #71248: drivers: usb: stm32: Add explicit dts clocks property for otg hsp ulpi

erwango, 2024-04-09:
**Is your enhancement proposal related to a problem? Please describe.**
See https://github.com/zephyrproject-rtos/zephyr/pull/68431/
Configuring clocks for usb phy on STM32 is confusing due to:
- difference between various IP versions
- ulpi clock which should be disabled in low power more

**Describe the solution you'd like**
Add clear dts properties to explicit the possible clocks configuration which are expected from the IP.
Update both legacy and experimental drivers

00thirdeye00, 2024-07-08:
hej @erwango, I am new to zephyr and would like to work on this issue. Could you please guide me and give me some references/information for my understanding? :)

djiatsaf-st, 2024-07-31:
Hi , 

have you tried some examples? if so which ones? on what types of nucleo or dk boards?

The PR you specified, I could see that he used a board that just uses a stm32f479 chip (Adi_sdp_k1), but it is not strictly properly

a nucleo board (maybe it can influence the configuration of the clocks on the driver), which is why I am asking you these 

questions above in order to test on my side and provide more details if I find any.

Thank you

erwango, 2025-02-07:
cc: @kbidani

erwango, 2025-02-07:
Linked to: https://github.com/zephyrproject-rtos/zephyr/issues/76842
Should avoid issues such as https://github.com/zephyrproject-rtos/zephyr/issues/84934

mathieuchopstm, 2026-02-09:
Partially resolved:
- [x] Have `clocks` property on (applicable) PHYs
  - Same rules as other peripherals
    - `clocks[0]` = clock gate
    - `clocks[1]` = clock mux (if applicable)
- [x] Consume PHYs `clocks` for their configuration
  - Done as part of the PHY pseudo-drivers 
  - [ ] **TODO:** I think there are still some cases (DT files/series) where the PHY clock mux is configured via `clocks` on USB controller
- [ ] Expose `Sleep clock` configuration in Devicetree
  - Only needed for a few *legacy* series (F4/H7) with ULPI interface
    - `Sleep clock` must be *disabled* when using embedded FS PHY instead of HS ULPI
  - *Currently handled via a dedicated `if(SERIES_xxx)`-gated driver*
    - Could be sufficient given new products don't need this?

mathieuchopstm, 2026-09-25:
For series where the USB controller has only a clock gate, it is on the USB controller.

For series where the USB controller has a clock gate and a clock mux, both are now on the USB controller node.

For series where the USB controller has a clock gate only, and the OTGHS PHY has a clock gate, each gate goes on the respective controller. In addition, if there is a clock mux that affects the OTGHS PHY, then it is now configured on the OTGHS PHY node instead of the USB node.

This should be now consistent across all series, so this point is addressed. Closing issue.

closed 2026-09-25 by mathieuchopstm