zephyrproject-rtos/zephyr #71252: lib: smf: constant number of arguments for SMF_CREATE_STATE()

maass-hamburg, 2024-04-09:
## Introduction

Currently the number of arguments of `SMF_CREATE_STATE()` is dependent on the Kconfig options `CONFIG_SMF_ANCESTOR_SUPPORT` and `CONFIG_SMF_INITIAL_TRANSITION`, IMO this is bad, as it breaks existing code if a Kconfig option is enabled, that wasn't enabled when the existing code was written.

### Problem description

As the USB-C subsystem uses the SMF with `CONFIG_SMF_ANCESTOR_SUPPORT` enabled and `CONFIG_SMF_INITIAL_TRANSITION` disabled, so the user can’t use hierarchical state machine with support for initial transitions in the same project together with using the USB-C subsystem, as enabling `CONFIG_SMF_INITIAL_TRANSITION` would break the USB-C subsystem.

### Proposed change

Therefore `SMF_CREATE_STATE()` should always have 5 arguments and the unused arguments should be set to `NULL`

## Detailed RFC

NA

### Proposed change (Detailed)

NA

### Dependencies

at some places, where `SMF_CREATE_STATE()` is used, `, NULL, NULL`  or `, NULL` has to be added

### Concerns and Unresolved Questions

NA

## Alternatives

List any alternatives considered, and the reasons for choosing this option
over them.

keith-zephyr, 2024-04-19:
The proposal makes sense.  @sambhurst Do you have any objections?

glenn-andrews, 2024-04-21:
Thinking about it, if `CONFIG_SMF_ANCESTOR_SUPPORT` is not enabled, you don't want the `.parent` field, to save memory in flat state machines that don't use it.

sambhurst, 2024-04-21:
Seems like a good idea to me.

glenn-andrews, 2024-04-23:
> > Thinking about it, if `CONFIG_SMF_ANCESTOR_SUPPORT` is not enabled, you don't want the `.parent` field, to save memory in flat state machines that don't use it.
> 
> It should be no problem changing that in another PR after my proposal has been accepted und my PR been merged.

The current state machine omits it for flat state machines, so permanently adding it is a memory waste on resource-constrained devices (So I'm told).

maass-hamburg, 2024-04-23:
> The current state machine omits it for flat state machines, so permanently adding it is a memory waste on resource-constrained devices (So I'm told).

In the current main the `const struct smf_state *parent` it is a permanent member of `struct smf_state`.  My proposal would work with having and not having it as a permanent member.  For not having it as a permanent member feel free to open a new issue and PR to have it changed. By the way having a permanent member of a struct, that is not used in some configurations is not forbidden and the conditional inclusion is not mandatory, but allowed [see here](https://docs.zephyrproject.org/latest/contribute/coding_guidelines/index.html#rule-a-1-conditional-compilation).

keith-zephyr, 2024-04-23:
I think it was on oversight that the initial implementation of the SMF made  `const struct smf_state *parent` a permanent member.

glenn-andrews, 2024-04-23:
> In the current main the `const struct smf_state *parent` it is a permanent member of `struct smf_state`. 

https://github.com/zephyrproject-rtos/zephyr/blob/84e7107b32f9babdab7c74a7b2a3f96df1f5172b/include/zephyr/smf.h#L69

It's not there if `CONFIG_SMF_ANCESTOR_SUPPORT` is not set.

glenn-andrews, 2024-04-23:
> In the current main the `const struct smf_state *parent` it is a permanent member of `struct smf_state`. 

You are correct, my mistake.

carlescufi, 2024-04-30:
Architecture WG:

- No objections

carlescufi, 2024-05-06:
@maass-hamburg have you sent an email to the devel@ mailing list about this breaking change?

maass-hamburg, 2024-05-06:
> @maass-hamburg have you sent an email to the devel@ mailing list about this breaking change?

Yes, here https://lists.zephyrproject.org/g/devel/message/9472

PR #71250 (2024-05-07): lib: smf: make number of arguments constant for SMF_CREATE_STATE()
Currently the number of arguments of `SMF_CREATE_STATE()` is dependent on the Kconfig options `CONFIG_SMF_ANCESTOR_SUPPORT` and `CONFIG_SMF_INITIAL_TRANSITION`, IMO this is bad, as it breaks existing code if a Kconfig option is enabled, that wasn't enabled when the existing code was written. 

Explicitly as the USB-C subsystem uses the SMF with `CONFIG_SMF_ANCESTOR_SUPPORT` enabled and `CONFIG_SMF_INITIAL_TRANSITION` disabled, so the user can’t use hierarchical state machine with support for initial transitions in the same proj [... cut]
closed 2024-05-07 by carlescufi