zephyrproject-rtos/zephyr #71262: LE Audio: BAP unicast default audio start sine shell command fails when invoked multiple times

niym-ot, 2024-04-09:
**Describe the bug**
LE Audio: BAP unicast audio start sine shell command fails when invoked multiple times.

In the case of default streams, sequence seems to be wrong.

The right sequence shall be as given below
1. lc3 initialization
2. verify sine stream validity
3. start sine stream

Current sequence is 
1. verify sine stream validity (Validity check is toggled)
2. lc3 initialization
3. start sine stream

**Impact**
sine stream cannot be invoked multiple times in case of unicast default stream

**Environment (please complete the following information):**
 - OS: Zephyr
 - Toolchain  Zephyr SDK
 - Commit SHA or Version:  latest HEAD

henrikbrixandersen, 2024-04-09:
> Commit SHA or Version: latest HEAD

Please always state a version or a SHA. "Latest HEAD" looses its meaning as soon as a new commit is made.

niym-ot, 2024-04-09:
Closing the issue as in the latest , it is already fixed by the below commit

[f4c8020d45] Emil Gydesen <emil.gydesen@nordicsemi.no> (2/26/2024)

Bluetooth: BAP: Shell: Minor refactor of LC3 encoder

This commit renames a few variables and changes the type and adds some additional checks.

It moves stream specific data to the shell stream struct, but still keeps the global values that were used to initialize the LC3 encoder

The purpose of this is to better allow for a future LC3 decoder without any clashes in names or the like.

closed 2024-04-10 by Thalley