zephyrproject-rtos/zephyr #71263: LE Audio: cap_initiator unicast-start shell command fails because qos is not set for streams

niym-ot, 2024-04-09:
**Describe the bug**
LE Audio: cap_initiator unicast-start shell command fails because qos for the streams belonging to one group is not set

**Code Ref**
git/fw/rf/zephyr/subsys/bluetooth/audio/shell/cap_initiator.c

When bt_bap_unicast_group_create is invoked, it verifies whether the qos is configured for all streams in the group

In unicast start shell command, cmd_cap_initiator_unicast_start, qos is not configured for the streams and hence
unicast-start command fails.

**Environment**
OS: Zephyr
Toolchain Zephyr SDK
Commit SHA or Version: latest HEAD

henrikbrixandersen, 2024-04-09:
> Commit SHA or Version: latest HEAD

Please always state a version or a SHA. "Latest HEAD" looses its meaning as soon as a new commit is made.

Thalley, 2024-04-12:
@niym-ot since this is from the shell, can you provide steps to reproduce?

Thalley, 2024-04-30:
You can test with the `cap_inititator ac_1` command as well

PR #72333 (2024-05-08): Bluetoioth: CAP: Shell: Added qos config to unicast_start cmd
For cap_initiator unicast_start command, qos configuration is added.

Fixes #71263