zephyrproject-rtos/zephyr #71269: LwM2M block-wise transfer: add callback to notify upper layer that a transfer has restarted

marco85mars20, 2024-04-09:
When a Zephyr device receives data via Block-Transfer, each block is passed to the upper layers via a post-write callback. The receiving function knows:
- the size of the current block
- the total size of the data being transferred
- if the current block is the last one

However, if the transfer fails, depending on the CoAP configuration of the client (sending the data), the transfer might be restarted from scratch, ie. from block #0. Zephyr detects the reception of the first block for an already existing transfer, removes the old context for the block transfer and creates a new one. However, the application is not notified about it and keeps appending the incoming blocks for the new transfers to those from the old one.

**Describe the solution you'd like**
A possible solution could be to add a parameter to the post-write callback (eg. `first_block`) set to `true` when receiving block #0

SeppoTakalo, 2024-04-11:
As you have found a real issue here, my proposal would be even a third option: refactor the post-write callback
1. Remove any Block-Wise parameters from post-write. It should only receive a full payload, if that is available.
2. Define a new callback and resource type that can handle block-wise transfers. Even without having full memory for it.

This new resource type should allow both read&write using block-wise.

rlubos, 2024-04-11:
I'm not sure if I'm convinced about introducing yet another callback type (although I'm of course open for discussion), wouldn't that be confusing to have two different types of write callbacks, depending on whether you expect to receive resource data via block transfer or not? Especially, that post-write wouldn't be reliably called then, for example during FW updates.

For me the most appealing option would be to simply add the `offset` parameter to the callback. It would not only allow to detect when the transfer starts from scratch, but also detect data stream inconsistencies (although I believe that the LwM2M engine already guarantees that out-of-order blocks are not forwarded to the application).

I don't think though that we could drop `last_block` parameter - `total_size` is not always available, i.e. if the CoAP peer does not send the Size1/2 option, we have no information about the total resource size. Maybe we should group block parameters given to the callback function into some structure, and provide a pointer to the structure in the callback instead - that way apps could easily determine whether block transfer is used or not.

PR #72590 (2024-05-15): lwm2m: Add offset to post-write callback
This PR contains fix for #71351 and tests to verify that, but also new fixes for issues that I saw while implementing the tests.

## lwm2m: Add offset to post-write callback

When Block-Wise transfer restarts, the post-write callback should
receive some indication that the block is actually a beginning of new,
instead of part of previous transfer.

NOTE: This is API break.

Fixes #71351
Fixes #71269

## test: lwm2m: Add tests for Block-Wise transfers

Block-Wise GET, PUT and SEND

##  lwm2m: Allow Block-Wise write without callbacks
    
Allow Block-Wise [... cut]