zephyrproject-rtos/zephyr #71276: Xtensa xt-clang global merging causing issues with gen_kobject_list.py script

SaintNicholas, 2024-04-09:
We are using an Xtensa toolchain RI-2021.7, specifically xt-clang.

An example main.c build command is:
```
.../bin/xt-clang -DKERNEL -DK_HEAP_MEM_POOL_SIZE=0 -DLITTLEENDIAN_CPU -D__ZEPHYR__=1 -Idependencies/zephyr/include -Ibuilds/build/zephyr/include/generated -Idependencies/zephyr/drivers -Iapp/include -isystem dependencies/zephyr/lib/libc/minimal/include -isystem dependencies/zephyr/lib/libc/common/include -fno-strict-aliasing -Os -imacros builds/build/zephyr/include/generated/autoconf.h -ffreestanding -fno-common -g -gdwarf-4 -fcolor-diagnostics -fms-extensions -imacros dependencies/zephyr/include/zephyr/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wno-typedef-redefinition -Wdouble-promotion -Wno-pointer-sign -Wpointer-arith -Wno-sometimes-uninitialized -Wno-shift-overflow -Wno-missing-braces -Wno-self-assign -Wno-address-of-packed-member -Wno-unused-function -Wno-initializer-overrides -Wno-section -Wno-unknown-warning-option -Wno-unused-variable -Wno-format-invalid-specifier -Wno-gnu -Wno-tautological-compare -Werror=implicit-int -fno-asynchronous-unwind-tables -mno-global-merge -Wno-vla -fmacro-prefix-map=app=CMAKE_SOURCE_DIR -fmacro-prefix-map=dependencies/zephyr=ZEPHYR_BASE -ffunction-sections -fdata-sections -mlongcalls -std=c99 -o CMakeFiles/app.dir/src/main.c.obj -c app/src/main.c
```

In our `main.c` file, we have 2 threads defined.
```
struct k_thread u_thr;
struct k_thread poc_thr;
```

Even though `-mno-global-merge` is being passed, when we build, we still see a lot of warnings for `clang-10: warning: argument unused during compilation: '-mno-global-merge' [-Wunused-command-line-argument]`. When we look at our map file, we see `u_thr` and `poc_thr` were placed into a `.bss..L_MergedGlobals` section.

```
 .bss..L_MergedGlobals
                0x00029f90      0x160 ../app/libapp.a(main.c.obj)
                0x00029f90                u_thr
                0x0002a040                poc_thr
```

When we do a verbose build, we see that the gen_kobject_list.py script is detecting that `u_thr` and `poc_thr` are at the same address, which is causing a fatal error during runtime.

```
gen_kobject_list.py: symbol 'u_thr' at 0x29f90 contains 1 object(s)
gen_kobject_list.py: symbol 'poc_thr' at 0x29f90 contains 1 object(s)
```

Looking at the gen_kobject_list.py script, it appears to parse the DWARF debug information to determine the addresses of these variables. When I read the elf file that this script is parsing, with the command `./xt-readelf --debug-dump=info build/zephyr/zephyr_pre1.elf | grep "u_thr\|poc_thr" -B1 -A5`, I see the following information:

```
 <1><cb2>: Abbrev Number: 32 (DW_TAG_variable)
    <cb3>   DW_AT_name        : (indirect string, offset: 0xcaf): u_thr
    <cb7>   DW_AT_type        : <0x63f>
    <cbb>   DW_AT_external    : 1
    <cbb>   DW_AT_decl_file   : 25
    <cbc>   DW_AT_decl_line   : 27
    <cbd>   DW_AT_location    : 5 byte block: 3 90 9f 2 0       (DW_OP_addr: 29f90)
--
 <1><ce3>: Abbrev Number: 32 (DW_TAG_variable)
    <ce4>   DW_AT_name        : (indirect string, offset: 0xcc0): poc_thr
    <ce8>   DW_AT_type        : <0x63f>
    <cec>   DW_AT_external    : 1
    <cec>   DW_AT_decl_file   : 25
    <ced>   DW_AT_decl_line   : 31
    <cee>   DW_AT_location    : 8 byte block: 3 90 9f 2 0 23 b0 1       (DW_OP_addr: 29f90; DW_OP_plus_uconst: 176)
```

Looking further at the gen_kobject_list.py script, it's only looking at the `DW_OP_addr` part of the `DW_AT_location`, it's ignoring the `DW_OP_plus_uconst` part for `poc_thr`, so the addresses appear to be the same.

When I add the following diff to the gen_kobject_list.py script, so it uses `DW_OP_plus_uconst`:

```
diff --git a/scripts/build/gen_kobject_list.py b/scripts/build/gen_kobject_list.py
index 6e41adb797..6e461c1c56 100755
--- a/scripts/build/gen_kobject_list.py
+++ b/scripts/build/gen_kobject_list.py
@@ -171,6 +171,7 @@ def debug_die(die, text):
 # -- ELF processing
 
 DW_OP_addr = 0x3
+DW_OP_plus_uconst = 0x23
 DW_OP_fbreg = 0x91
 STACK_TYPE = "z_thread_stack_element"
 thread_counter = 0
@@ -622,6 +623,11 @@ def find_kobjects(elf, syms):
             addr = ((loc.value[1] << 0 ) | (loc.value[2] << 8)  |
                     (loc.value[3] << 16) | (loc.value[4] << 24))
 
+            # Handle a DW_FORM_exprloc that contains a DW_OP_addr, followed immediately by
+            # a DW_OP_plus_uconst.
+            if len(loc.value) >= 7 and loc.value[5] == DW_OP_plus_uconst:
+                addr += (loc.value[6])
+
         if addr == 0:
             # Never linked; gc-sections deleted it
             continue
```

Then the `u_thr` and `poc_thr` addresses are parsed correctly by the script:

```
gen_kobject_list.py: symbol 'u_thr' at 0x29f90 contains 1 object(s)
gen_kobject_list.py: symbol 'poc_thr' at 0x2a040 contains 1 object(s)
```

dcpleung, 2024-04-09:
Since you already have a diff, could you submit a PR?

SaintNicholas, 2024-04-09:
PR submitted and linked.

closed 2024-04-22 by nashif