zephyrproject-rtos/zephyr #71288: LLVM build fails with ASAN=y

swkim101, 2024-04-09:
**Describe the bug**
LLVM fails to build Zephyr under the ASAN option.  
The 'hello world' sample cannot be built.  

**To Reproduce**

zephyr/samples/hello_world/prj.conf:
```
CONFIG_ASAN=y
```

build:
```sh
ZEPHYR_TOOLCHAIN_VARIANT=llvm west -v build --pristine -b native_sim_64 samples/hello_world
```

**Expected behavior**
Successful build.

**Impact**

ASAN is not applicable for LLVM build.  
It may affect security research requiring both ASAN and LLVM-dependent features, such as llvm-pass.

**Logs and console output**

*(Truncated)*

```
[83/84] cd /home/sungwoo/hdd1/zephyrproject/zephyr/build/zephyr && /usr/bin/make -f /home/sungwoo/hdd1/zephyrproject/zephyr/scripts/native_simulator/Makefile all --warn-undefined-variables -r NSI_CONFIG_FILE=/home/sungwoo/hdd1/zephyrproject/zephyr/build/zephyr/NSI/nsi_config
FAILED: zephyr/CMakeFiles/native_runner_executable zephyr/zephyr.exe /home/sungwoo/hdd1/zephyrproject/zephyr/build/zephyr/CMakeFiles/native_runner_executable /home/sungwoo/hdd1/zephyrproject/zephyr/build/zephyr/zephyr.exe 
cd /home/sungwoo/hdd1/zephyrproject/zephyr/build/zephyr && /usr/bin/make -f /home/sungwoo/hdd1/zephyrproject/zephyr/scripts/native_simulator/Makefile all --warn-undefined-variables -r NSI_CONFIG_FILE=/home/sungwoo/hdd1/zephyrproject/zephyr/build/zephyr/NSI/nsi_config
/usr/bin/ld: /home/sungwoo/hdd1/zephyrproject/zephyr/build/zephyr/NSI/home/sungwoo/hdd1/zephyrproject/zephyr/build/zephyr/zephyr.elf.loc_cpusw.o: TLS transition from R_X86_64_TLSLD to R_X86_64_TPOFF32 against `_ZN6__asanL14fake_stack_tlsE' at 0xb in section `.text._ZN6__asan15SetTLSFakeStackEPNS_9FakeStackE' failed
/usr/bin/ld: failed to set dynamic section sizes: bad value
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make: *** [/home/sungwoo/hdd1/zephyrproject/zephyr/scripts/native_simulator/Makefile:134: /home/sungwoo/hdd1/zephyrproject/zephyr/build/zephyr/zephyr.exe] Error 1
ninja: build stopped: subcommand failed.
```

**Environment (please complete the following information):**
- LLVM version: tested on 14, 15, and 18
 - Commit SHA or Version used: 20f747bb630b4ce9e84d5e3e7f525df19f33cc43

**Additional context**

`/usr/bin/ld` is the reason.  
If I use 'ld.lld` instead, then it works. 

For some reason, `ZEPHYR_TOOLCHAIN_VARIANT=llvm` uses `ld`, not `ld.lld` as follows:
(I got this by adding a `-v` option in `NSI_CC` of `scripts/native_simulator/Makefile`)

```
"/usr/bin/ld" -z relro --hash-style=gnu --eh-frame-hdr -m elf_x86_64 -pie -dynamic-linker /lib64/ld-linux-x86-64.so.2 -o /home/sungwoo/hdd1/zephyrproject/zephyr/build/zephyr/zephyr.exe /lib/x86_64-linux-gnu/Scrt1.o /lib/x86_64-linux-gnu/crti.o /usr/lib/gcc/x86_64-linux-gnu/11/crtbeginS.o -L/mnt/hdd1/sungwoo/llvm-project-18/build/bin/../lib/x86_64-unknown-linux-gnu -L/mnt/hdd1/sungwoo/llvm-project-18/build/lib/clang/18/lib/x86_64-unknown-linux-gnu -L/usr/lib/gcc/x86_64-linux-gnu/11 -L/usr/lib/gcc/x86_64-linux-gnu/11/../../../../lib64 -L/lib/x86_64-linux-gnu -L/lib/../lib64 -L/usr/lib/x86_64-linux-gnu -L/usr/lib/../lib64 -L/lib -L/usr/lib --whole-archive /mnt/hdd1/sungwoo/llvm-project-18/build/lib/clang/18/lib/x86_64-unknown-linux-gnu/libclang_rt.asan_static.a --no-whole-archive --whole-archive /mnt/hdd1/sungwoo/llvm-project-18/build/lib/clang/18/lib/x86_64-unknown-linux-gnu/libclang_rt.asan.a --no-whole-archive --dynamic-list=/mnt/hdd1/sungwoo/llvm-project-18/build/lib/clang/18/lib/x86_64-unknown-linux-gnu/libclang_rt.asan.a.syms --whole-archive /home/sungwoo/hdd1/zephyrproject/zephyr/build/zephyr/NSI/home/sungwoo/hdd1/zephyrproject/zephyr/build/zephyr/zephyr.elf.loc_cpusw.o /home/sungwoo/hdd1/zephyrproject/zephyr/build/zephyr/NSI/runner.a --no-whole-archive --gc-sections -ldl -lm --no-as-needed -lpthread -lrt -lm -ldl -lresolv -lgcc --as-needed -lgcc_s --no-as-needed -lpthread -lc -lgcc --as-needed -lgcc_s --no-as-needed /usr/lib/gcc/x86_64-linux-gnu/11/crtendS.o /lib/x86_64-linux-gnu/crtn.o -T /home/sungwoo/hdd1/zephyrproject/zephyr/boards/native/common/natsim_linker_script.ld -T /home/sungwoo/hdd1/zephyrproject/zephyr/build/zephyr/NSI/linker_script.ld
```

It works if `/usr/bin/ld` is replaced by `ld.lld`.

Although now I can get `zephyr.exe`, but it crashes:

```
$ ./build/zephyr/zephyr.exe 
==1920160==Shadow memory range interleaves with an existing memory mapping. ASan cannot proceed correctly. ABORTING.
==1920160==ASan shadow was supposed to be located in the [0x00007fff7000-0x10007fff7fff] range.
==1920160==This might be related to ELF_ET_DYN_BASE change in Linux 4.12.
==1920160==See https://github.com/google/sanitizers/issues/856 for possible workarounds.
==1920160==Process memory map follows:
        0x00007fff7000-0x00008fff7000
        0x00008fff7000-0x02008fff7000
        0x02008fff7000-0x10007fff8000
        0x14d80ab72000-0x14d80af00000
        0x14d80af00000-0x14d80b700000
        0x14d80b700000-0x14d80b800000
        0x14d80b900000-0x14d80ba00000
        0x14d80bb00000-0x14d80bc00000
        0x14d80bc57000-0x14d80c900000
        0x14d80c900000-0x14d80d100000
        0x14d80d100000-0x14d80d200000
        0x14d80d300000-0x14d80d400000
        0x14d80d500000-0x14d80d800000
        0x14d80d825000-0x14d80dc09000
        0x14d80dc09000-0x14d80dc31000   /usr/lib/x86_64-linux-gnu/libc.so.6
        0x14d80dc31000-0x14d80ddc6000   /usr/lib/x86_64-linux-gnu/libc.so.6
        0x14d80ddc6000-0x14d80de1e000   /usr/lib/x86_64-linux-gnu/libc.so.6
        0x14d80de1e000-0x14d80de1f000   /usr/lib/x86_64-linux-gnu/libc.so.6
        0x14d80de1f000-0x14d80de23000   /usr/lib/x86_64-linux-gnu/libc.so.6
        0x14d80de23000-0x14d80de25000   /usr/lib/x86_64-linux-gnu/libc.so.6
        0x14d80de25000-0x14d80de32000
        0x14d80de32000-0x14d80de35000   /usr/lib/x86_64-linux-gnu/libgcc_s.so.1
        0x14d80de35000-0x14d80de4c000   /usr/lib/x86_64-linux-gnu/libgcc_s.so.1
        0x14d80de4c000-0x14d80de50000   /usr/lib/x86_64-linux-gnu/libgcc_s.so.1
        0x14d80de50000-0x14d80de51000   /usr/lib/x86_64-linux-gnu/libgcc_s.so.1
        0x14d80de51000-0x14d80de52000   /usr/lib/x86_64-linux-gnu/libgcc_s.so.1
        0x14d80de52000-0x14d80de55000   /usr/lib/x86_64-linux-gnu/libresolv.so.2
        0x14d80de55000-0x14d80de5f000   /usr/lib/x86_64-linux-gnu/libresolv.so.2
        0x14d80de5f000-0x14d80de62000   /usr/lib/x86_64-linux-gnu/libresolv.so.2
        0x14d80de62000-0x14d80de63000   /usr/lib/x86_64-linux-gnu/libresolv.so.2
        0x14d80de63000-0x14d80de64000   /usr/lib/x86_64-linux-gnu/libresolv.so.2
        0x14d80de64000-0x14d80de66000
        0x14d80de66000-0x14d80de74000   /usr/lib/x86_64-linux-gnu/libm.so.6
        0x14d80de74000-0x14d80def0000   /usr/lib/x86_64-linux-gnu/libm.so.6
        0x14d80def0000-0x14d80df4b000   /usr/lib/x86_64-linux-gnu/libm.so.6
        0x14d80df4b000-0x14d80df4c000   /usr/lib/x86_64-linux-gnu/libm.so.6
        0x14d80df4c000-0x14d80df4d000   /usr/lib/x86_64-linux-gnu/libm.so.6
        0x14d80df4e000-0x14d80df68000
        0x14d80df68000-0x14d80df6a000   /usr/lib/x86_64-linux-gnu/ld-linux-x86-64.so.2
        0x14d80df6a000-0x14d80df94000   /usr/lib/x86_64-linux-gnu/ld-linux-x86-64.so.2
        0x14d80df94000-0x14d80df9f000   /usr/lib/x86_64-linux-gnu/ld-linux-x86-64.so.2
        0x14d80df9f000-0x14d80dfa0000
        0x14d80dfa0000-0x14d80dfa2000   /usr/lib/x86_64-linux-gnu/ld-linux-x86-64.so.2
        0x14d80dfa2000-0x14d80dfa4000   /usr/lib/x86_64-linux-gnu/ld-linux-x86-64.so.2
        0x500000000000-0x503000000000
        0x503000000000-0x503000040000
        0x503000040000-0x503e00000000
        0x503e00000000-0x503e00040000
        0x503e00040000-0x506000000000
        0x506000000000-0x506000040000
        0x506000040000-0x506e00000000
        0x506e00000000-0x506e00040000
        0x506e00040000-0x507000000000
        0x507000000000-0x507000040000
        0x507000040000-0x507e00000000
        0x507e00000000-0x507e00040000
        0x507e00040000-0x524000000000
        0x524000000000-0x524000040000
        0x524000040000-0x524e00000000
        0x524e00000000-0x524e00040000
        0x524e00040000-0x540000000000
        0x540000000000-0x540000002000
        0x5556a9803000-0x5556a9891000   /mnt/hdd1/sungwoo/zephyrproject/zephyr/build/zephyr/zephyr.exe
        0x5556a9891000-0x5556a9ac9000   /mnt/hdd1/sungwoo/zephyrproject/zephyr/build/zephyr/zephyr.exe
        0x5556a9ac9000-0x5556a9acb000   /mnt/hdd1/sungwoo/zephyrproject/zephyr/build/zephyr/zephyr.exe
        0x5556a9acb000-0x5556a9ad6000   /mnt/hdd1/sungwoo/zephyrproject/zephyr/build/zephyr/zephyr.exe
        0x5556a9ad6000-0x5556aa53f000
        0x7fffb6be6000-0x7fffb6c08000   [stack]
        0x7fffb6d7d000-0x7fffb6d81000   [vvar]
        0x7fffb6d81000-0x7fffb6d83000   [vdso]
        0xffffffffff600000-0xffffffffff601000   [vsyscall]
==1920160==End of process memory map.
```

swkim101, 2024-04-09:
I found `ld.lld` option:

https://github.com/zephyrproject-rtos/zephyr/blob/b9f3d68863b9f75dcc1dfac7dd7ff269b99cd118/cmake/toolchain/llvm/target.cmake#L1-L8

However, setting `CONFIG_LLVM_USE_LLD=y` in `prj.conf` did not solve this problem.  
It still uses `/usr/bin/ld`.

swkim101, 2024-04-10:
The reason for the error "Shadow memory range interleaves with an existing memory mapping" is a duplicated ASAN initialization.  

As mentioned in https://forums.developer.nvidia.com/t/nvidia-driver-525-147-05-cause-asan-crash/286375, `AsanInitInternal()` uses a global variable ***asan_inited*** to track the initialization states and avoid duplicated initialization.  
However, `AsanInitInternal()` fails to trace duplicated initialization.  

As shown in gdb log, the address of `asan_inited` variable is not the same as the first `AsanInitInternal()`:  
```gdb
(gdb) r
Starting program: /mnt/hdd1/sungwoo/zephyrproject/zephyr/zephyr.exe 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".

Breakpoint 1, __asan::AsanInitInternal () at /home/sungwoo/hdd1/llvm-project-18/compiler-rt/lib/asan/asan_rtl.cpp:394
394       if (LIKELY(AsanInited()))
(gdb) p asan_inited
$8 = {val_dont_use = 0 '\000'}
(gdb) p &asan_inited
$9 = (__sanitizer::atomic_uint8_t *) 0x555555837d19 <__asan::asan_inited>
(gdb) c
Continuing.

Breakpoint 1, __asan::AsanInitInternal () at /home/sungwoo/hdd1/llvm-project-18/compiler-rt/lib/asan/asan_rtl.cpp:396
396       SanitizerToolName = "AddressSanitizer";
(gdb) c
Continuing.

Breakpoint 1, __asan::AsanInitInternal () at /home/sungwoo/hdd1/llvm-project-18/compiler-rt/lib/asan/asan_rtl.cpp:394
394       if (LIKELY(AsanInited()))
(gdb) p asan_inited
$10 = {val_dont_use = 0 '\000'}
(gdb) p &asan_inited
$11 = (__sanitizer::atomic_uint8_t *) 0x555555d6bbd9 <__asan::asan_inited>
(gdb) c
Continuing.

Breakpoint 1, __asan::AsanInitInternal () at /home/sungwoo/hdd1/llvm-project-18/compiler-rt/lib/asan/asan_rtl.cpp:396
396       SanitizerToolName = "AddressSanitizer";
(gdb) c
Continuing.
==3903196==Shadow memory range interleaves with an existing memory mapping. ASan cannot proceed correctly. ABORTING.
```
Thus, `AsanInitInternal ()` wrongly thinks that ASAN is not initialized yet, so the program calls InitializeShadowMemory again and causes the crash.

I don't believe it is related to https://github.com/google/sanitizers/issues/856.  
Instead, it is similar to these:
* https://github.com/google/sanitizers/issues/1630#issuecomment-1997292016
* https://forums.developer.nvidia.com/t/nvidia-driver-525-147-05-cause-asan-crash/286375

aescolar, 2024-04-10:
Note: For native_sim (32) the build succeeds with gnu ld. The runtime failure is the same though.

closed 2024-04-12 by aescolar