zephyrproject-rtos/zephyr #71305: drivers:led_strip Sample not working for MIMXRT1050 with DMA en DCACHE enabled

allemanm, 2024-04-10:
**Describe the bug**
I build and run `samples/drivers/led_strip/` on the MIMXRT1050_EVK with DMA enabled for LPSPI. However, no data is put on MOSI pin. When I disable DCACHE, the data is put on the bus properly. I checked out Zephyr on commit `f86f390baf27a48768b6bd3cd1285b54970c537c`

**To Reproduce**
Steps to reproduce the behavior:
1. Edit `prj.conf` by adding `CONFIG_SPI_MCUX_LPSPI_DMA=y`
2. Edit `mimxrt1050_evk.overlay` by changing `&lpspi3` to `&lpspi1`
3. Run `west build -p -b mimxrt1050_evk samples/drivers/led_strip/`
4. `west flash -r jlink`
5. No signal on the MOSI pin
6. Edit `prj.conf` by adding `CONFIG_DCACHE=n`
7. Build and flash
8. Signal is visible on MOSI pin

prj.conf
```
CONFIG_LOG=y
CONFIG_LED_STRIP=y
CONFIG_LED_STRIP_LOG_LEVEL_DBG=y
CONFIG_SPI_MCUX_LPSPI_DMA=y
```

mimxrt1050_evk.overlay
```
#include <zephyr/dt-bindings/led/led.h>

&lpspi1 {
	led_strip: ws2812@0 {
		compatible = "worldsemi,ws2812-spi";

		/* SPI */
		reg = <0>; /* ignored, but necessary for SPI bindings */
		spi-max-frequency = <6400000>;

		/* WS2812 */
		chain-length = <2>; /* arbitrary; change at will */
		spi-cpha;
		spi-one-frame = <0xf0>; /* 11110000: 625 ns high and 625 ns low */
		spi-zero-frame = <0xc0>; /* 11000000: 312.5 ns high and 937.5 ns low */
		color-mapping = <LED_COLOR_ID_GREEN
				 LED_COLOR_ID_RED
				 LED_COLOR_ID_BLUE>;
	};
};

/ {
	aliases {
		led-strip = &led_strip;
	};
};
```

**Expected behavior**
I would expect the lpspi driver to work properly when both `CONFIG_SPI_MCUX_LPSPI_DMA` and `CONFIG_DCACHE` are enabled.

**Logs and console output**
With DCACHE disabled there is a signal on the MOSI pin
[image]

**Environment (please complete the following information):**
 - Zephyr SDK: zephyr-sdk-0.16.5-1
 - Zephy version: `f86f390baf27a48768b6bd3cd1285b54970c537c`

dleach02, 2024-05-20:
@EmilioCBen, take a look at this when we get to feature freeze

DerekSnell, 2024-06-13:
Hi @allemanm ,

Since this issue depends on the DMA and is affected by the cache, I suspect the cause of this issue is that the buffer used by the SPI driver is cached.  On the RT1050, by default the linker places data in the external SDRAM, which is all cached.  The cache is only for the M7 CPU.  So it seems likely the issue here is that when the app writes to the `pixels` buffer, it is written to the cache.  But that buffer in SDRAM is not updated, and therefore the DMA does not read the same data.  

If using the DMA on a platform like this, the app should place the DMA buffers in non-cacheable memory.  The spi_loopback test does this by placing the buffers in the [.nocache region](https://github.com/zephyrproject-rtos/zephyr/blob/e2cdd92a522d48e1acf080981bf41ab4effc60ef/tests/drivers/spi/spi_loopback/src/spi.c#L63).

Best regards

allemanm, 2024-10-17:
I can still reproduce the issue with Zephyr commit 3c82cd6585b451fa0e25080b7f4c0ff4f86c0e16. I did indeed populate these with 0 Ohm resistors. The description of the steps to reproduce is still correct, however the build command has been changed to:
```west build -p -b mimxrt1050_evk samples/drivers/led/led_strip/```

EmilioCBen, 2024-10-17:
Thank you for the feedback, I will start taking a look into this!

DerekSnell, 2025-01-22:
I feel this issue is a known limitation of the LPSPI driver, since that driver does not manage cache coherency, as discussed above https://github.com/zephyrproject-rtos/zephyr/issues/71305#issuecomment-2166463001.  There is already an Enhancement Request https://github.com/zephyrproject-rtos/zephyr/issues/81125 to add cache coherency support to this driver.  I feel this bug report does not need to be kept open, since the enhancement already tracks this feature request.

decsny, 2025-01-31:
okay, I think I agree, we can close this issue as duplicate. Although, this issue does make it sound more like a bug than an enhancement request since it shows an in tree sample failing. I wonder if there are any other platforms in Zephyr with DMA based communication peripheral support and having a cache, and if they have this issue? Otherwise maybe LPSPI+EDMA is the first to expose a generic limitation of Zephyr?

closed 2025-01-31 by decsny