zephyrproject-rtos/zephyr #71312: ci: samples/drivers/counter/alarm/sample.drivers.counter.alarm build is failing on sam_e70_xplained/same70q21

ajarmouni-st, 2024-04-10:
**To Reproduce**
```
west build -p -b sam_e70_xplained/same70q21 samples/drivers/counter/alarm -T sample.drivers.counter.alarm
```
**Logs and console output**
```
-- west build: making build dir ~/zephyrproject/zephyr/build pristine
-- west build: generating a build system
Loading Zephyr default modules (Zephyr base).
-- Application: ~/zephyrproject/zephyr/samples/drivers/counter/alarm
-- CMake version: 3.22.1
-- Found Python3:~/zephyrproject/.venv/bin/python3 (found suitable version "3.10.12", minimum required is "3.8") found components: Interpreter 
-- Cache files will be written to: ~/.cache/zephyr
-- Zephyr version: 3.6.99 (~/zephyrproject/zephyr)
-- Found west (found suitable version "1.2.0", minimum required is "0.14.0")
-- Board: sam_e70_xplained, qualifiers: same70q21
-- ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
-- Found host-tools: zephyr 0.16.5 (~/zephyr-sdk-0.16.5)
-- Found toolchain: zephyr 0.16.5 (~/zephyr-sdk-0.16.5)
-- Found Dtc: ~/zephyr-sdk-0.16.5/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.6.0", minimum required is "1.4.6") 
-- Found BOARD.dts: ~/zephyrproject/zephyr/boards/atmel/sam/sam_e70_xplained/sam_e70_xplained_same70q21.dts
-- Found devicetree overlay: ~/zephyrproject/zephyr/samples/drivers/counter/alarm/boards/sam_e70_xplained_same70q21.overlay
devicetree error: 'clk' appears in /soc/tc@4000c000 in ~/zephyrproject/zephyr/build/zephyr/zephyr.dts.pre, but is not declared in 'properties:' in ~/zephyrproject/zephyr/dts/bindings/sensor/atmel,sam-tc-qdec.yaml
-- In: ~/zephyrproject/zephyr/build/zephyr, command: ~/zephyrproject/.venv/bin/python3;~/zephyrproject/zephyr/scripts/dts/gen_defines.py;--dts;~/zephyrproject/zephyr/build/zephyr/zephyr.dts.pre;--dtc-flags;'-Wno-unique_unit_address_if_enabled';--bindings-dirs;~/zephyrproject/zephyr/dts/bindings;--header-out;~/zephyrproject/zephyr/build/zephyr/include/generated/devicetree_generated.h.new;--dts-out;~/zephyrproject/zephyr/build/zephyr/zephyr.dts.new;--edt-pickle-out;~/zephyrproject/zephyr/build/zephyr/edt.pickle;--vendor-prefixes;~/zephyrproject/zephyr/dts/bindings/vendor-prefixes.txt
CMake Error at ~/zephyrproject/zephyr/cmake/modules/dts.cmake:295 (message):
  gen_defines.py failed with return code: 1
Call Stack (most recent call first):
  ~/zephyrproject/zephyr/cmake/modules/zephyr_default.cmake:132 (include)
  ~/zephyrproject/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:66 (include)
  ~/zephyrproject/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:92 (include_boilerplate)
  CMakeLists.txt:4 (find_package)

-- Configuring incomplete, errors occurred!
FATAL ERROR: command exited with status 1: /usr/bin/cmake -DWEST_PYTHON=~/zephyrproject/.venv/bin/python3 -B~/zephyrproject/zephyr/build -GNinja -DBOARD=sam_e70_xplained/same70q21 -S~/zephyrproject/zephyr/samples/drivers/counter/alarm
```
**Impact**
blocking CI in https://github.com/zephyrproject-rtos/zephyr/pull/71208

**Environment**
main 731f8d473610f04ec3d0cf53cb10dd373b467f31

ajarmouni-st, 2024-04-10:
The sam_e70_xplained/same70q21 overlay was introduced here [commit](https://github.com/zephyrproject-rtos/zephyr/pull/36195/commits/61d9951f00d8a99cfcdcd7661a07133bdc87adf7#diff-5c157640bfc682355bb870d327a39e0a166267e3f6cc26c38999927719ccaec3)

ajarmouni-st, 2024-04-10:
This change fixes the build issue, but no idea if it's the proper compatible @nandojve .
```
diff --git a/boards/atmel/sam/sam_e70_xplained/sam_e70_xplained_same70q21.dts b/boards/atmel/sam/sam_e70_xplained/sam_e70_xplained_same70q21.dts
index 64a26cecc8..01fdba6e5c 100644
--- a/boards/atmel/sam/sam_e70_xplained/sam_e70_xplained_same70q21.dts
+++ b/boards/atmel/sam/sam_e70_xplained/sam_e70_xplained_same70q21.dts
@@ -18,7 +18,7 @@
 
 &tc0 {
        status = "okay";
-       compatible = "atmel,sam-tc-qdec";
+       compatible = "atmel,sam-tc";
 
        pinctrl-0 = <&tc0_qdec_default>;
        pinctrl-names = "default";
```

ajarmouni-st, 2024-04-10:
The `atmel,sam-tc-qdec` binding does not have a `clk` property.

ajarmouni-st, 2024-04-10:
The `tc0`node in `sam_e70_xplained_same70q21.dts` was introduced in [PR](https://github.com/zephyrproject-rtos/zephyr/pull/55655/files#diff-43a0d9ccb1733e22576ca651e1fbddabd1e1f884eb0b33fb012066cc82fd4d9c).

ajarmouni-st, 2024-04-10:
> This change fixes the build issue, but no idea if it's the proper compatible @nandojve .
>
> ```
> diff --git a/boards/atmel/sam/sam_e70_xplained/sam_e70_xplained_same70q21.dts [...]

In [dts/arm/atmel/same70.dtsi](https://github.com/zephyrproject-rtos/zephyr/blob/7d45f3c92a1c9b1dc0e63526670bcf4f16e9fa33/dts/arm/atmel/same70.dtsi#L348), we can see that the correct compatible is indeed `atmel,sam-tc`.

tswaehn, 2024-04-17:
question: the line was there for more than 1year without changes

```
compatible = "atmel,sam-tc-qdec";
```

why was it not failing before?

ajarmouni-st, 2024-04-17:
@tswaehn The test was never executed in CI cz `sam_e70_xplained_same70q21` is not in the `platform_allow` list in `samples/drivers/counter/alarm/sample.yaml`.

ajarmouni-st, 2024-05-31:
Hi @nandojve, any updates regarding this issue?

tswaehn, 2024-06-01:
The timer controler can run in standard `sam-tc` mode or in `sam-tc-qdec` mode. depending on usage we load the correct driver. default mode is `sam-tc-qdec`, thus loading the` sam-tc-qdec` driver.

so for the example mentioned above we need the `sam-tc` mode. 

suggestion: we update the `samples/drivers/counter/alarm/boards/sam_e70_xplained_same70q21.overlay`

&tc0 {
        compatible = "atmel,sam-tc";
        clk = <4>;
        status = "okay";
};

that should do?

nandojve, 2024-06-02:
> Hi @nandojve, any updates regarding this issue?

Hi @ajarmouni-st , not yet. Let me try to fix for the next week.

Hi @tswaehn , as I already mentioned here #35683 the atmel,sam-tc is reserved to be used with the counter driver and atmel,sam-tc-qdec must be used when using the quadrature decoder.

My suggestion is that we apply the sub function from a timer a child node:

```c
&tc0 {
    qdec {
        compatible = ...
        pinctrl = ....
    };
    pwm {
    };
    input-capture {
    };
    etc {
    };
};
```

That allows us to specialize any sub function from the timer in a proper way without create this type of issue.

_Originally posted by @nandojve in https://github.com/zephyrproject-rtos/zephyr/issues/71329#issuecomment-2061284621_

About the below:
> The timer controller can run in sam-tc mode or in sam-tc-qdec mode. depending on usage we load the correct driver. default mode is sam-tc-qdec, thus loading the sam-tc-qdec driver.

My understand is that the default must be `sam-tc` always since the `qdec` is an special mode. However, the soc/board must be able to define the 'sam-tc-qdec' without broke the other functionality and that is the issue here that is fixed by above proposal.

nandojve, 2024-06-02:
Hi @tswaehn ,

Would you mind to check if #73619 works for you?

tswaehn, 2024-06-02:
> Hi @tswaehn ,
> 
> Would you mind to check if #73619 works for you?

sure, will try to implement. I am just curious if other sample demos will expect `qdec` by default => but they need to be adjusted then, .... no problem.

nandojve, 2024-06-03:
Hi @tswaehn ,

> sure, will try to implement. I am just curious if other sample demos will expect qdec by default => but they need to be adjusted then, .... no problem.

The #73619 is green at CI so I suppose that no new changes are needed. After your tests let me know if it will be necessary add an entry at `samples/sensor/qdec/sample.yaml` to run at CI each time.

tswaehn, 2024-06-07:
* board `sam_e70_xplained` qualifier `same70q21` ... works
* * board `sam_e70_xplained` qualifier `same70q21b` ... doesnt work (I guess its only missing the tc0 definitions in `sam_e70_xplained_same70q21b.dts`)

tswaehn, 2024-06-07:
tested the qdec0 and qdec3 

added to my overlay: 

```
&tc0 {
	status = "okay";

	qdec0:qdec {
		status = "okay";
		label = "QDEC_0";
	};

};

&tc3 {
	status = "okay";

	qdec3: qdec {
		status = "okay";
		label = "QDEC_3";
	};

};
```
and finally used it (as before)

```
devX = device_get_binding("QDEC_0");
devY = device_get_binding("QDEC_3");

sensor_sample_fetch(devX);
struct sensor_value valueX = {0};
sensor_channel_get(devX, SENSOR_CHAN_ROTATION, &valueX);

ret = sensor_sample_fetch(devY);
struct sensor_value valueY = {0};
sensor_channel_get(devY, SENSOR_CHAN_ROTATION, &valueY);
```
looks like it works :)

PR #73619 (2024-06-17): drivers: counter: sam: Add qdec as tc special mode
The current atmel,sam-tc-qdec sensor implementation shared the timer counter node. This create issues when users wants define both modes. The current proposal changes the qdec dedinition to be a child of tc and refactor all the chain of definitions.

Fixes #71312
commit 0561d74d (2024-06-17): drivers: counter: sam: Add qdec as tc special mode

The current atmel,sam-tc-qdec sensor implementation shared the timer
counter node. This create issues when users wants define both modes.
The current proposal changes the qdec dedinition to be a child of
tc and refactor all the chain of definitions.

Fixes #71312
closed 2024-06-17 by nashif