zephyrproject-rtos/zephyr #71333: STM32 IWDG watchdog is not started automatically

vgeneves, 2024-04-10:
**Describe the bug**
When option CONFIG_WDT_DISABLE_AT_BOOT is not set, STM32 IWDG is not started during device init.

Since : https://github.com/zephyrproject-rtos/zephyr/commit/842a6ba02dfd95d0148491796b5b4b02785dc073, IWDG wathdog is not started in `iwdg_stm32_install_timeout` but in `iwdg_stm32_setup`.

The problem is only `iwdg_stm32_install_timeout` is called during watchdog device initialization. Thus watchdog is not started anymore.

**To Reproduce**
Using `samples/drivers/watchdog` sample code is enough to reproduce.
When entering `main` watchdog is still disabled

**Expected behavior**
Watchdog timer should be started during device driver initialization

**Environment (please complete the following information):**
 - OS: Linux
 - Zephyr SDK 0.16.3
 - main

FRASTM, 2024-04-17:
thanks @vgeneves to report your observation
 I understand from the https://docs.zephyrproject.org/latest/hardware/peripherals/watchdog.html  that "_IWDG wathdog is not started in iwdg_stm32_install_timeout but in iwdg_stm32_setup._" --> this as expected 
The `wdt_install_timeout` function is for preparing a watchdog which is started _after_, by the wdt_setup()  (wdt_install_timeout is not supposed to star the watchdog)

vgeneves, 2024-04-17:
Yes, my issue was not about that.
It is about `WDT_DISABLE_AT_BOOT`. From my understanding if `WDT_DISABLE_AT_BOOT` is disabled, watchdog should be configured with default timeout (`CONFIG_IWDG_STM32_INITIAL_TIMEOUT`) and started.

Actually the driver only configure the watchdog with `wdt_install_timeout` but it is not started.
Or maybe my understanding of `WDT_DISABLE_AT_BOOT` is wrong. Anyway the behavior of the iwdg changed since https://github.com/zephyrproject-rtos/zephyr/commit/842a6ba02dfd95d0148491796b5b4b02785dc073.

FRASTM, 2024-04-23:
if WDT_DISABLE_AT_BOOT is set "Disable watchdog at Zephyr system startup."
so that if WDT_DISABLE_AT_BOOT is not set , as you say --> watchdog should be configured with default timeout (CONFIG_IWDG_STM32_INITIAL_TIMEOUT)  by  `iwdg_stm32_install_timeout` function and started by `wdt_setup` function
So this could be:
```patch
diff --git a/drivers/watchdog/wdt_iwdg_stm32.c b/drivers/watchdog/wdt_iwdg_stm32.c
index 2be891e9b59..012921f4837 100644
--- a/drivers/watchdog/wdt_iwdg_stm32.c
+++ b/drivers/watchdog/wdt_iwdg_stm32.c
@@ -192,8 +192,9 @@ static int iwdg_stm32_init(const struct device *dev)
 	struct wdt_timeout_cfg config = {
 		.window.max = CONFIG_IWDG_STM32_INITIAL_TIMEOUT
 	};
-
+	/* Watchdog should be configured and started by `wdt_setup`*/
 	iwdg_stm32_install_timeout(dev, &config);
+	iwdg_stm32_setup(dev, 0);
 #endif
 
 	/*
```

vgeneves, 2024-04-23:
I tested it and it works for me.

closed 2024-04-30 by nashif

Oleh-Kravchenko, 2025-04-24:
Hehe, this “fix” cost me a few hours of debugging 🚬 

Default timeout is so small that it barely provides time for other drivers to initiate,
application logic not even started.