zephyrproject-rtos/zephyr #71347: USB Device Stack Initialization from the App

gl-agnx, 2024-04-10:
**Is your enhancement proposal related to a problem? Please describe.**
The current USB Device stack initializes at the POST_KERNEL stage. This creates two issues:

1) Callbacks like the usb_update_sn_string_descriptor callback are very limited to using serials that are already in flash/ram. This means an application cannot, for example, provide a serial number after reading it from an external EEPROM. It essentially just shifts the Serial from the KConfig to C code. Several posts online have users showing this exact issue. The USB Device Stack documentation does not describe that this is a limitation, but one discovers it quickly with some testing.

2) In some situations, the USB stack should only be initialized after some external hardware is ready, such as a Hub or clocks that are platform specific. But since the USB Device stack is designed as "fire and forget" at the start, it's not possible, or it requires stopping the stack and restarting it which is not desirable

**Describe the solution you'd like**
A PR has been created with the change:

https://github.com/zephyrproject-rtos/zephyr/pull/70767

This change enables an optional configuration that prevents the USB stack from initializing at the POST_KERNEL stage, where the application will then call the initialization function itself at the right time. This gives developers the flexibility to start the stack when they need to, while leaving previous behavior the same for those who don't

**Describe alternatives you've considered**
The usb init function could be made external to the module instead of adding a new function. The function was added to keep everything as compatible as possible

**Additional context**
Examples of issue occurring for zephyr users:

https://devzone.nordicsemi.com/f/nordic-q-a/92440/settings-subsystem-init-fails-nvs-backend
https://devzone.nordicsemi.com/f/nordic-q-a/90475/nrf5340---runtime-change-of-iserialnumber

mkaranki, 2024-04-10:
(Not opposing the change, just considering alternatives.)

Would it be enough to only have possibility to tweak the initialization priorities of USB device, filesystems and storage device drivers? USB device is initialized on `POST_KERNEL` with priority `CONFIG_APPLICATION_INIT_PRIORITY` which is by default 90. `CONFIG_APPLICATION_INIT_PRIORITY` is used by few other things as well, so the configurability is not 100%, but could be still enough for most cases? Quite many drivers can be initialized by that time.
 
The linked tickets mentioned two different cases. One problem seems to be in getting a serial number from settings subsystem on `POST_KERNEL` time. To make that work, the underlying backend device must be ready. So if the initialization time of it can be modified to happen earlier than USB init, then I think getting the setting from it should be possible.

The second ticket was about getting a serial number from littlefs. Could this change help in such use cases: https://github.com/zephyrproject-rtos/zephyr/pull/67896? Getting serial number kind of data from file system was one driver of that change.

gl-agnx, 2024-04-12:
It may help but still limiting. 
For example, in our case we needed to read the serial using GPIOs, which meant setup and reading. That's done easily in the application.

The second case you mentioned is the more extreme case where the PR changes make sense. Starting a file system reading files, etc can't really be achieved anywhere else.

gl-agnx, 2024-04-18:
Any movement on this? I don't see why there's resistance to this feature. There's at least 2 different entities (I saw a 3rd one somewhere) having an issue with how this operates.

* The change is very limited and can be even more limited to the same item
* It resolves an issue that's happening in practice
* It does not impact anyone else

PR #71695 (2024-05-08): usb: device: get rid of the partial initialization called by SYS_INIT
Part of the initialization is done in the static usb_device_init() function called by SYS_INIT(). We can move part of it, which initialize the descriptor and calls usb_set_config(), to usb_enable(). This allow some usb_get_device_descriptor() dependencies called at application desired point.

Also make sure that usb_fix_descriptor() is called only once.

Resolves #71347
closed 2024-05-08 by carlescufi