zephyrproject-rtos/zephyr #71392: doc: Documentation issue in 'build/dts/zephyr-user-node'

TorstenRobitzki, 2024-04-11:
**Describe the bug**

To me, it looks like, that there is an undocumented requirement, that a gpio attribute name needs to end in `-gpios`. For more experienced users, this requirement might be clear. For me as a newbe, following the documentation, that came to a surprise, that toke me quite some time.

If the attribute does not end in `-gpios`, like in this example: 

    / {
        zephyr,user{
            led-enables = <&gpio0 1 GPIO_ACTIVE_HIGH>;
        };
    };

The following error is reported: `devicetree error: <Node /soc/peripheral@50000000/gpio@842500 in '/Users/yyyy/xxxxx/zephyr-workspace/zephyr/misc/empty_file.c'> lacks #led-enable-cells
`

**Environment**

* Page: `build/dts/zephyr-user-node`
* Version: main
* SHA-1: cd3008a9ac85ef65c301892f9304e3e3bdfb1622

henrikbrixandersen, 2024-04-11:
What you are looking for is called "specifier spaces". More information available here: https://docs.zephyrproject.org/latest/build/dts/phandles.html#specifier-spaces

TorstenRobitzki, 2024-04-12:
Hi @henrikbrixandersen ,
yes, that is probably exactly what I need to understand, to understand the error message above. I do not fully understand it, but following the link to the dt-bindings-cells documentation, show a warning box: 

    It is an abuse of this feature to use it to name properties in unconventional ways.

    For example, this feature is not meant for cases like naming a property my-pin, then assigning it to the “gpio” specifier space using this feature. Properties which refer to GPIOs should use conventional names, i.e. end in -gpios or -gpio.

So, I think all GPIOs should then and with a `-gpios` attribute name.

Question now, is: is this obvious, or should this requirement be mentioned in the documentation above?

henrikbrixandersen, 2024-04-21:
> Question now, is: is this obvious, or should this requirement be mentioned in the documentation above?

I think it is obvious from the documentation, but I may be biased.

closed 2024-04-22 by TorstenRobitzki