zephyrproject-rtos/zephyr #71410: SDL display driver doesn't support API calls from a thread other than main

kartben, 2024-04-11:
**Describe the bug**

Trying to call display functions (ex `display_write`) from a thread other than `main` does not seem to work.

**To Reproduce**
Steps to reproduce the behavior:
1. change `samples/drivers/display/src/main.c` so that the contents of main() is effectively executed in a different thread
2. build and run on native_sim --> screen is black

**Expected behavior**

SDL driver should work in a multithreaded environment, just like other display drivers.

**Impact**

can't run custom GUI code that requires a dedicated thread

**Environment (please complete the following information):**

Linux  x86_64, 3.6.99 f419ea7990998ee13476d5c206a6bdb83a4a0567

aescolar, 2024-04-12:
@kartben can you provide a minimal repro case?

kartben, 2024-04-12:
@aescolar Tweaking the display driver sample like the below is the minimal repro. After applying the change, sample still works fine on e.g. M5Stack Core 2 (ILI9342C display controller), but SDL/native_sim_64 gives a black screen.

```diff
diff --git a/samples/drivers/display/src/main.c b/samples/drivers/display/src/main.c
index a9267b51c7..af14abc3b0 100644
--- a/samples/drivers/display/src/main.c
+++ b/samples/drivers/display/src/main.c
@@ -170,8 +170,24 @@ static inline void fill_buffer_mono10(enum corner corner, uint8_t grey,
 	fill_buffer_mono(corner, grey, 0xFFu, 0x00u, buf, buf_size);
 }
 
-int main(void)
+int gui_entry_point(void *arg1, void *arg2, void *arg3);
+
+K_THREAD_STACK_DEFINE(thread_stack, 4096);
+static struct k_thread thread;
+
+int main(void) {
+	k_thread_create(&thread, thread_stack, K_THREAD_STACK_SIZEOF(thread_stack),
+			(k_thread_entry_t)gui_entry_point, NULL, NULL, NULL,
+			K_PRIO_PREEMPT(5), 0, K_NO_WAIT);
+	return 0;
+}
+
+int gui_entry_point(void *arg1, void *arg2, void *arg3)
 {
+	ARG_UNUSED(arg1);
+	ARG_UNUSED(arg2);
+	ARG_UNUSED(arg3);
+	
 	size_t x;
 	size_t y;
 	size_t rect_w;
```

SDL documentation is pretty clear on the fact that e.g. [`SDL_RenderPresent()`](https://wiki.libsdl.org/SDL2/SDL_RenderPresent) or [`SDL_PollEvent()`](https://wiki.libsdl.org/SDL2/SDL_PollEvent) should only be called from the thread that initialized the video system.

aescolar, 2024-04-12:
CC @jfischer-no @danieldegrasse @dcpleung

aescolar, 2024-04-12:
@kartben I can reproduce it with that example.

> SDL documentation is pretty clear on the fact that e.g. [SDL_RenderPresent()](https://wiki.libsdl.org/SDL2/SDL_RenderPresent) or [SDL_PollEvent()](https://wiki.libsdl.org/SDL2/SDL_PollEvent) should only be called from the thread that initialized the video system.

`sdl_display_init()` is run in the same "zephyr boot" thread that will eventually run the app `main()`
So it may be that you are right and that is the cause.
Also, as with a quick hack like this below, on top of your patch it renders properly (so sdl_display_init_bottom is called from the same thread that does the display updates):

```diff
diff --git a/drivers/display/display_sdl.c b/drivers/display/display_sdl.c
index f21288d1b0f..44013c5adfe 100644
--- a/drivers/display/display_sdl.c
+++ b/drivers/display/display_sdl.c
@@ -48,7 +48,10 @@ static inline uint32_t mono_pixel_order(uint32_t order)
        }
 }
 
-static int sdl_display_init(const struct device *dev)
+static int sdl_display_init(const struct device *dev) {
+       return 0;
+}
+int _sdl_display_init(const struct device *dev)
 {
        const struct sdl_display_config *config = dev->config;
        struct sdl_display_data *disp_data = dev->data;

diff --git a/samples/drivers/display/src/main.c b/samples/drivers/display/src/main.c
index a9267b51c74..1142d36ed30 100644
--- a/samples/drivers/display/src/main.c
+++ b/samples/drivers/display/src/main.c
@@ -189,6 +204,9 @@ int main(void)
        fill_buffer fill_buffer_fnc = NULL;
 
        display_dev = DEVICE_DT_GET(DT_CHOSEN(zephyr_display));
+       _sdl_display_init(display_dev);
        if (!device_is_ready(display_dev)) {
                LOG_ERR("Device %s not found. Aborting sample.",
                        display_dev->name);

```
Anyhow, SDL is a bit far from my expertise.
But I guess that if this is indeed the reason, a fix would be to move all the sdl display handling into the same thread.
I won't have time to do this now though.
So I better remove myself as assignee.

Maybe one of the display maintainers has the BW?

KimBP, 2024-07-16:
> @kartben I can reproduce it with that example.
> 
> > SDL documentation is pretty clear on the fact that e.g. [SDL_RenderPresent()](https://wiki.libsdl.org/SDL2/SDL_RenderPresent) or [SDL_PollEvent()](https://wiki.libsdl.org/SDL2/SDL_PollEvent) should only be called from the thread that initialized the video system.
> 
> `sdl_display_init()` is run in the same "zephyr boot" thread that will eventually run the app `main()` So it may be that you are right and that is the cause. Also, as with a quick hack like this below, on top of your patch it renders properly (so sdl_display_init_bottom is called from the same thread that does the display updates):
> 
> ```diff
> diff --git a/drivers/display/display_sdl.c b/drivers/display/display_sdl.c
> index f21288d1b0f..44013c5adfe 100644
> --- a/drivers/display/display_sdl.c
> +++ b/drivers/display/display_sdl.c
> @@ -48,7 +48,10 @@ static inline uint32_t mono_pixel_order(uint32_t order)
>         }
>  }
>  
> -static int sdl_display_init(const struct device *dev)
> +static int sdl_display_init(const struct device *dev) {
> +       return 0;
> +}
> +int _sdl_display_init(const struct device *dev)
>  {
>         const struct sdl_display_config *config = dev->config;
>         struct sdl_display_data *disp_data = dev->data;
> 
> diff --git a/samples/drivers/display/src/main.c b/samples/drivers/display/src/main.c
> index a9267b51c74..1142d36ed30 100644
> --- a/samples/drivers/display/src/main.c
> +++ b/samples/drivers/display/src/main.c
> @@ -189,6 +204,9 @@ int main(void)
>         fill_buffer fill_buffer_fnc = NULL;
>  
>         display_dev = DEVICE_DT_GET(DT_CHOSEN(zephyr_display));
> +       _sdl_display_init(display_dev);
>         if (!device_is_ready(display_dev)) {
>                 LOG_ERR("Device %s not found. Aborting sample.",
>                         display_dev->name);
> ```
> 
> Anyhow, SDL is a bit far from my expertise. But I guess that if this is indeed the reason, a fix would be to move all the sdl display handling into the same thread. I won't have time to do this now though. So I better remove myself as assignee.
> 
> Maybe you or one of the display maintainers has the bandwith?

This "workaround" wont work if you also add lvgl into your application. lvgl_init() is called too early and will miss a display to work at.
Forcing lv_task_handler() to be called in context of main-thread like suggested by @jakkra  I also find very limiting.
Wouldn't a solution be to (optionally) make the sdl-display-driver spawn a thread being responsible for all the SDL critical stuff like initializing 'video system' and calling RenderPresent etc?
I doubt the overhead of a thread in a native_sim application would cause any resource issues

aescolar, 2024-07-17:
@KimBP just to be clear: That (https://github.com/zephyrproject-rtos/zephyr/issues/71410#issuecomment-2051649697) was not a workaround, and I would very much discourage anybody from doing that. That comment was only meant to provide information about the possible cause of the issue.

kartben, 2024-11-10:
@Finomnis just in case you feel like spending more time on the SDL driver :-)

Finomnis, 2024-11-10:
@kartben Might take a look at it.

PetervdPerk-NXP, 2025-03-24:
Related to #87519, setting `SDL_DISPLAY_USE_HARDWARE_ACCELERATOR=n` fixes screen calling from other threads. But then sdl input breaks.

Finomnis, 2025-03-24:
> Related to [#87519](https://github.com/zephyrproject-rtos/zephyr/issues/87519), setting `SDL_DISPLAY_USE_HARDWARE_ACCELERATOR=n` fixes screen calling from other threads. But then sdl input breaks.

I would intercept this and weaken it a little; we did not confirm that it actually fully breaks SDL only. It could be just in your project, where the priority of the render thread starves the input thread.

Finomnis, 2025-03-24:
Also, he careful with the word "fixes", both are undefined behavior in the current implementation, it just happens to work better.

Finomnis, 2025-03-24:
> Good point could it be that this is a problem in general? Render thread being more important then input thread? Your typical mouse cursor on desktop also has a high priority.

Those priorities should be configurable. Maybe they should be reversed in your project?

faxe1008, 2025-03-27:
I took some time yesterday and did some testing. An easy way to break rendering is to have an additional flush thread:
```
west build -b native_sim/native/64 -d build samples/modules/lvgl/demos -- -DCONFIG_LV_Z_FLUSH_THREAD=y
```
What I found is that rendering breaks the second writes are done from a different thread:
```
diff --git a/drivers/display/display_sdl.c b/drivers/display/display_sdl.c
index 7396b105052..9f65c4e44c9 100644
--- a/drivers/display/display_sdl.c
+++ b/drivers/display/display_sdl.c
@@ -8,6 +8,7 @@
 #define DT_DRV_COMPAT zephyr_sdl_dc

 #include <zephyr/drivers/display.h>
+#include <zephyr/kernel.h>

 #include <string.h>
 #include <stdlib.h>
@@ -38,6 +39,7 @@ struct sdl_display_data {
        enum display_pixel_format current_pixel_format;
        uint8_t *buf;
        uint8_t *read_buf;
+       k_tid_t sdl_thread_id;
 };

 static inline uint32_t mono_pixel_order(uint32_t order)
@@ -55,6 +57,7 @@ static int sdl_display_init(const struct device *dev)
        struct sdl_display_data *disp_data = dev->data;
        bool use_accelerator = true;
        LOG_DBG("Initializing display driver");
+    disp_data->sdl_thread_id = k_current_get();

        IF_DISABLED(CONFIG_SDL_DISPLAY_USE_HARDWARE_ACCELERATOR, (use_accelerator = false));

@@ -254,6 +257,10 @@ static int sdl_display_write(const struct device *dev, const uint16_t x,
        const struct sdl_display_config *config = dev->config;
        struct sdl_display_data *disp_data = dev->data;

+       if (disp_data->sdl_thread_id != k_current_get()) {
+               LOG_WRN_ONCE("Display write called from wrong thread");
+       }
+
        LOG_DBG("Writing %dx%d (w,h) bitmap @ %dx%d (x,y)", desc->width,
                        desc->height, x, y);
```

What I find strange is that currently even without the `-DCONFIG_LV_Z_FLUSH_THREAD` handling of input events also happens from another thread https://github.com/zephyrproject-rtos/zephyr/blob/main/boards/native/common/sdl/sdl_events.c#L13
But this seems to work?

Anyways I feel like all operations need to be happening in one thread:

- Init of display (partially done in display_sdl_bottom.c and sdl_events.c, but the part within sdl_bottom seems to be the important one)
- Handling of events (currently in sdl_events.c)
- Display write operations

The input_sdl_touch binding should be ok since it only adds a callback within the `init` everything else pretty much happens when the callback is triggered.

Don't know what is the best solution for this - let the display driver have an own thread where `init` happens and a message queue for  pending writes? Any ideas @Finomnis @aescolar

aescolar, 2025-03-27:
> Don't know what is the best solution for this - let the display driver have an own thread where init happens and a message queue for pending writes? 

Something like that makes sense to me.

Finomnis, 2025-03-27:
I personally would probably implement it very similar to how the [LVGL flush thread](https://github.com/zephyrproject-rtos/zephyr/blob/43e8f11d9445afb1aa1328709be6466691374fe3/modules/lvgl/lvgl_display.c#L21) is implemented.

faxe1008, 2025-03-27:
Did a very dirty quick sketch for this:
https://github.com/faxe1008/zephyr/commit/f1f31bc6672852cb42318f3e8eb3c477459275d6

I find that this indeed fixes the rendering, would be intersting if this also fixes @PetervdPerk-NXP and @JarmouniA issues.

Keeping the message buffer in memory really will not be needed if we make sure that there is only write message at a time, so thats something I will cleanup before submitting.

> About priorities: does this thread need a high priority or will it automatically inherit the priority of the calling thread if they synchronize via semaphore?

I think this will be a matter of priorities.

Finomnis, 2025-03-27:
@faxe1008 gotta be careful about recursion, the case where the input handlers get called from this thread which in turn causes a new write message. Might need to check if the calling thread is the SDL thread and then bypass the synchronization primitives.

PR #87883 (2025-04-08): drivers: display: sdl: Ensure draw is performed from init thread
Add a thread to the SDL display driver to ensure that init and renderer calls are performed from the same thread.

Fixes #71410.

Ping @Finomnis @PetervdPerk-NXP
commit 11b9eba3 (2025-04-08): drivers: display: sdl: Ensure draw is performed from init thread

Add a thread to the SDL display driver to ensure that init and renderer
calls are performed from the same thread.

Fixes #71410.
closed 2025-04-08 by kartben