zephyrproject-rtos/zephyr #71416: Compilation error with ARM Compiler 6

sreeramIfx, 2024-04-12:
**Describe the bug**
Compiling blinky with GCC works fine, but the same app compiled with ARM Compiler 6 fails with the below described error when linking.

```
C:\WINDOWS\system32\cmd.exe /C "cd . && C:\workspace\ArmCompilers\bin\armlink.exe  --cpu=cortex-m4   --via=CMakeFiles\zephyr_pre0.rsp -o zephyr\zephyr_pre0.elf && C:\WINDOWS\system32\cmd.exe /C "cd /D C:\workspace\gitlab\zephyrproject-rtos\zephyr\build\zephyr && "C:\Program Files\CMake\bin\cmake.exe" -E true""
Error: L6384E: No Load Execution Region of name bss seen yet at line 11.
Error: L6384E: No Load Execution Region of name bss seen yet at line 15.
```

The errors seems to be from the below lines (not able to find `__bss_start` symbol) in the cmake\linker_script\arm\linker.cmake. I am able to work-around the errors by hardcoding instead of using `__bss_start`

```
zephyr_linker_symbol(SYMBOL __kernel_ram_start EXPR "(@__bss_start@)")
zephyr_linker_symbol(SYMBOL __kernel_ram_end  EXPR "(${RAM_ADDR} + ${RAM_SIZE})")
zephyr_linker_symbol(SYMBOL __kernel_ram_size EXPR "(@__kernel_ram_end@ - @__bss_start@)")
```

**To Reproduce**
Steps to reproduce the behavior:
1. `$Env:ZEPHYR_TOOLCHAIN_VARIANT="armclang"`
2. `west build -p always -b cy8cproto_063_ble .\samples\basic\blinky\`

**Expected behavior**
blinky should compile and link successfully with ARM compiler as well

**Impact**
Unable to use ARM Compiler for our platforms

**Logs and console output**
```
west build -p always -b cy8cproto_063_ble .\samples\basic\blinky\
-- west build: making build dir C:\workspace\gitlab\zephyrproject-rtos\zephyr\build pristine
-- west build: generating a build system
Loading Zephyr default modules (Zephyr base).
-- Application: C:/workspace/gitlab/zephyrproject-rtos/zephyr/samples/basic/blinky
-- CMake version: 3.29.0
-- Found Python3: C:/workspace/gitlab/zephyrproject-rtos/.venv/Scripts/python.exe (found suitable version "3.11.0", minimum required is "3.8") found components: Interpreter
-- Cache files will be written to: C:/workspace/gitlab/zephyrproject-rtos/zephyr/.cache
-- Zephyr version: 3.6.99 (C:/workspace/gitlab/zephyrproject-rtos/zephyr)
-- Found west (found suitable version "1.2.0", minimum required is "0.14.0")
-- Board: cy8cproto_063_ble, qualifiers: cyble_416045_02
-- Found armclang: C:/workspace/ArmCompilers/bin/armclang.exe (found suitable version "6.17", minimum required is "6.17")
-- Found Dtc: C:/ProgramData/chocolatey/bin/dtc.exe (found suitable version "1.5.0", minimum required is "1.4.6")
-- Found BOARD.dts: C:/workspace/gitlab/zephyrproject-rtos/zephyr/boards/cypress/cy8cproto_063_ble/cy8cproto_063_ble.dts
-- Generated zephyr.dts: C:/workspace/gitlab/zephyrproject-rtos/zephyr/build/zephyr/zephyr.dts
-- Generated devicetree_generated.h: C:/workspace/gitlab/zephyrproject-rtos/zephyr/build/zephyr/include/generated/devicetree_generated.h
-- Including generated dts.cmake file: C:/workspace/gitlab/zephyrproject-rtos/zephyr/build/zephyr/dts.cmake
C:/workspace/gitlab/zephyrproject-rtos/zephyr/build/zephyr/zephyr.dts:231.20-235.5: Warning (simple_bus_reg): /soc/bless: missing or empty reg/ranges property
C:/workspace/gitlab/zephyrproject-rtos/zephyr/build/zephyr/zephyr.dts:39.29-54.5: Warning (avoid_unnecessary_addr_size): /soc/pinctrl@40310000: unnecessary #address-cells/#size-cells without "ranges" or child "reg" property
Parsing C:/workspace/gitlab/zephyrproject-rtos/zephyr/Kconfig
Loaded configuration 'C:/workspace/gitlab/zephyrproject-rtos/zephyr/boards/cypress/cy8cproto_063_ble/cy8cproto_063_ble_defconfig'
Merged configuration 'C:/workspace/gitlab/zephyrproject-rtos/zephyr/samples/basic/blinky/prj.conf'
Configuration saved to 'C:/workspace/gitlab/zephyrproject-rtos/zephyr/build/zephyr/.config'
Kconfig header saved to 'C:/workspace/gitlab/zephyrproject-rtos/zephyr/build/zephyr/include/generated/autoconf.h'
-- The C compiler identification is ARMClang 6.17.0
-- The CXX compiler identification is ARMClang 6.17.0
CMake Warning (dev) at C:/Program Files/CMake/share/cmake-3.29/Modules/Compiler/ARMClang.cmake:91 (message):
  Policy CMP0123 is not set: ARMClang cpu/arch compile and link flags must be
  set explicitly.  Run "cmake --help-policy CMP0123" for policy details.  Use
  the cmake_policy command to set the policy and suppress this warning.

  For compatibility, CMake will automatically add cpu/arch flags based on the
  CMAKE_SYSTEM_PROCESSOR and/or CMAKE_SYSTEM_ARCH variables.
Call Stack (most recent call first):
  C:/Program Files/CMake/share/cmake-3.29/Modules/Compiler/ARMClang-C.cmake:3 (__compiler_armclang)
  C:/Program Files/CMake/share/cmake-3.29/Modules/CMakeCInformation.cmake:25 (include)
  C:/workspace/gitlab/zephyrproject-rtos/zephyr/cmake/modules/kernel.cmake:128 (project)
  C:/workspace/gitlab/zephyrproject-rtos/zephyr/cmake/modules/zephyr_default.cmake:141 (include)
  C:/workspace/gitlab/zephyrproject-rtos/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:66 (include)
  C:/workspace/gitlab/zephyrproject-rtos/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:92 (include_boilerplate)
[... 47 lines cut ...]
  C:/Program Files/CMake/share/cmake-3.29/Modules/Compiler/ARMClang-ASM.cmake:9 (__compiler_armclang)
  C:/Program Files/CMake/share/cmake-3.29/Modules/CMakeASMInformation.cmake:22 (include)
  C:/workspace/gitlab/zephyrproject-rtos/modules/lib/picolibc/CMakeLists.txt:38 (project)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Configuring done (27.9s)
-- Generating done (0.4s)
-- Build files have been written to: C:/workspace/gitlab/zephyrproject-rtos/zephyr/build
←[92m-- west build: building application
[2/150] Generating include/generated/version.h
-- Zephyr version: 3.6.99 (C:/workspace/gitlab/zephyrproject-rtos/zephyr), build: v3.6.0-2462-g17081222e2e7
[146/150] Linking C executable zephyr\zephyr_pre0.elf
FAILED: zephyr/zephyr_pre0.elf zephyr/zephyr_pre0.map C:/workspace/gitlab/zephyrproject-rtos/zephyr/build/zephyr/zephyr_pre0.map
C:\WINDOWS\system32\cmd.exe /C "cd . && C:\workspace\ArmCompilers\bin\armlink.exe  --cpu=cortex-m4   --via=CMakeFiles\zephyr_pre0.rsp -o zephyr\zephyr_pre0.elf && C:\WINDOWS\system32\cmd.exe /C "cd /D C:\workspace\gitlab\zephyrproject-rtos\zephyr\build\zephyr && "C:\Program Files\CMake\bin\cmake.exe" -E true""
Error: L6384E: No Load Execution Region of name bss seen yet at line 11.
Error: L6384E: No Load Execution Region of name bss seen yet at line 15.
Not enough information to list the image map.
Finished: 1 information, 0 warning and 2 error messages.
ninja: build stopped: subcommand failed.
```

**Environment (please complete the following information):**
 - OS: Windows
 - Toolchain: ARM Compiler 6
 - Commit SHA or Version used : 17081222e2e761378f9402fdebe1052983b8a5e5

sreeramIfx, 2024-04-12:
I am also seeing the issue described @ https://github.com/zephyrproject-rtos/zephyr/discussions/41435#discussioncomment-1922417

I am able to work around the issue by using ```-DCONFIG_OUTPUT_DISASSEMBLY=n``` (as described in the above link). But its indicated that it should not happen with versions ? #6.17 I am seeing it with ver #6.22 as well

closed 2024-09-09 by github-actions[bot]

krish2718, 2025-06-16:
Based on  [Arm Compiler 6 — Zephyr Project Documentation](https://docs.zephyrproject.org/latest/develop/toolchains/arm_compiler_6.html) , installed the toolchain, but hitting this issue, tried the suggested `-DCONFIG_OUTPUT_DISASSEMBLY=n` but it doesn't work, see below relevant info.

```
west build -p --no-sysbuild -b nrf5340dk/nrf5340/cpuapp ../zephyr/samples/net/zperf/ -d ../zephyr/samples/net/zperf/build_5340/  -- -DEXTRA_CONF_FILE="overlay-loopback.conf" -DCONFIG_NET_L2_ETHERNET=n -DCONFIG_OUTPUT_DISASSEMBLY=n -DCONFIG_PARTITION_MANAGER_ENABLED=n
```

```
 armclang -v
Product: Hardware Success Kit (Early Access)
Component: Arm Compiler for Embedded 6.24
Tool: armclang [5f371900]

Target: unspecified-arm-none-none
armclang: fatal error: no target architecture given; use --target=arm-arm-none-eabi or --target=aarch64-arm-none-eabi
```
```
samples/net/zperf/build_5340/modules/picolibc/libc.a  -o zephyr/zephyr_pre0.elf && cd zephyr/samples/net/zperf/build_5340/zephyr && /usr/bin/cmake -E true
Error: L6384E: No Load Execution Region of name  seen yet at line 618.
Error: L6226E: Missing base address for region soc_reset_hook.
```

wearyzen, 2025-08-29:
> [@wearyzen](https://github.com/wearyzen) do you have any input on this, or are able to take a look ?

I will take a look at this next week.

wearyzen, 2026-01-12:
@sreeramIfx @krish2718 I had a look at this issue and it seems the issue is because system symbols are added to the first non-empty region instead of FLASH (https://github.com/zephyrproject-rtos/zephyr/blob/76dd50833fabb56a8de65f01629f205de6c4f155/cmake/linker/armlink/scatter_script.cmake#L20) so adding the symbols in FLASH instead should fix this particular issue.
Here is a small patch for it:
```
From 1278eba5d94a9236f88e1a26af03dd2add230a43 Mon Sep 17 00:00:00 2001
From: Sudan Landge <sudan.landge@arm.com>
Date: Sat, 10 Jan 2026 21:10:43 +0000
Subject: [PATCH] fix: armclang undefinde symbols

Put system symbols in flash instead of the 1st non-empty section which
is null_ptr_detect in most cases.
---
 cmake/linker/armlink/scatter_script.cmake | 6 ++++--
 cmake/linker/linker_script_common.cmake   | 7 +++++++
 2 files changed, 11 insertions(+), 2 deletions(-)

diff --git a/cmake/linker/armlink/scatter_script.cmake b/cmake/linker/armlink/scatter_script.cmake
index b32f67b7b09..8ffc4a273e6 100644
--- a/cmake/linker/armlink/scatter_script.cmake
+++ b/cmake/linker/armlink/scatter_script.cmake
@@ -13,10 +13,12 @@ function(process_region)

   get_property(empty GLOBAL PROPERTY ${REGION_OBJECT}_EMPTY)
   if(NOT empty)
-    # For scatter files we move any system symbols into first non-empty load section.
+    # For scatter files we move any system symbols into FLASH.
     get_parent(OBJECT ${REGION_OBJECT} PARENT parent TYPE SYSTEM)
     get_property(symbols GLOBAL PROPERTY ${parent}_SYMBOLS)
-    set_property(GLOBAL APPEND PROPERTY ${REGION_OBJECT}_SYMBOLS ${symbols})
+
+    get_property(flash GLOBAL PROPERTY ${parent}_REGION_BY_NAME_FLASH)
+    set_property(GLOBAL APPEND PROPERTY ${flash}_SYMBOLS ${symbols})
     set_property(GLOBAL PROPERTY ${parent}_SYMBOLS)
   endif()

diff --git a/cmake/linker/linker_script_common.cmake b/cmake/linker/linker_script_common.cmake
index 93a179569cc..c34ae4d5570 100644
--- a/cmake/linker/linker_script_common.cmake
+++ b/cmake/linker/linker_script_common.cmake
@@ -266,6 +266,13 @@ function(add_region)

   set_property(GLOBAL PROPERTY ${ADD_REGION_REGION}_PARENT ${ADD_REGION_OBJECT})
   set_property(GLOBAL APPEND PROPERTY ${ADD_REGION_OBJECT}_REGIONS ${ADD_REGION_REGION})
+
+  # Build lookup: <system>_REGION_BY_NAME_<NAME> -> <region object>
+  # (Used by armlink to grab FLASH without scanning all regions.)
+  get_property(_r_name GLOBAL PROPERTY ${ADD_REGION_REGION}_NAME)
+  if(_r_name STREQUAL "FLASH")
+    set_property(GLOBAL PROPERTY ${ADD_REGION_OBJECT}_REGION_BY_NAME_${_r_name} ${ADD_REGION_REGION})
+  endif()
 endfunction()

 # add_group OBJECT o GROUP g adds group g to object o
--
2.43.0
```

Would you be able to try this patch and confirm if it works for you?

I wouldn't try on Zephyr main though because there are a lot of conflicts with picolibc.

wearyzen, 2026-01-20:
Hi @sreeramIfx @krish2718 PR #102509 fixes the build issues with armclang but for Infineon specifically you'll need https://github.com/zephyrproject-rtos/zephyr/compare/3abbd895b9ff1c794a518e3664d5bcbe1df8f4f0..5c3a58cee373382518e6f09f8f6ef3ccc78d1568 and some more fixes that you'll see in Infineon code.