zephyrproject-rtos/zephyr #71422: BLE MESH REMOTE PROVISIONING:Remote provisioning server do not work when receive response PB-ADV message before the ack message of sent PB-ADV request message from unprovisioned device

zhaishiqiang, 2024-04-12:
**Describe the bug:**
We use the BLE mesh stack to try the mesh remote provisioning function.But there is occur a problem between remote provisioning server and unprovisioned device
The message sequence in air:
1.The rpr server(remote provisioning server) send a PB-ADV message(we named this message as msgA) to unprovisioned device;
2.After unprovisioned device have received msgA, it will send the msgA's ACK and response message of msgA(we name this response message as msgB)
3.In some case,the open space is so noisy(with a lot of adv) that rpr server may receive msgB earlier than msgA's ACK.
In the file pb_adv.c, when rpr server receive msgB first, it will set link.tx.adv[0] to NULL through gen_prov_start()-->prov_clear_tx()-->free_segments()-->set link.tx.adv[0] to NULL.But when rpr server receive msgA's ACK later,as in gen_prov_ack(), this function will return early because link.tx.adv[0] is NULL.So can not call link.tx.cb() function(the pdu_send_complete() in rpr_srv.c) to set link state from BT_MESH_RPR_LINK_SENDING to BT_MESH_RPR_LINK_ACTIVE.And the rpr server can never send any PB-ADV message because the link state is always BT_MESH_RPR_LINK_SENDING.

**Expected behavior:**
We expect the zephyr mesh stack can deal with this situation:the rpr server receive msgB earlier than msgA's ACK,then it can normal work(can send PB-ADV message normally) for this situation.

**To Reproduce**
The method for reproduce this situation is:
For unprovisioned device,when it receive a PB-ADV message(we named it as msgA),it do not send msgA's ACK temporarily.It should send the response PB-ADV message for msgA first(we name this message as msgB).At the same time,you should make sure that all retransmission packets of msgB are send complete,then can send msgA's ACK. For this case,I guess you can get the result describe above,and the remote provisioning will stop fail.

**Personal analysis:**
I guess the sending of msgA's ACK and msgB is concurrently,maybe the first packet of msgA's ACK is earlier than the first packet of msgB. But it still maybe occur the situation above and will cause a problem.
In addition the mesh spec may not require that msgA's ACK is earlier received than msgB for rpr server,so I think the mesh stack should cover this situation.

Thank you for help

PavelVPV, 2024-04-21:
@alxelax, could you please have a look?

alxelax, 2024-04-22:
Hello @zhaishiqiang,
thanks for report. At the first glance it seems like a bug. I'll investigate the issue. However, it requires clarification just to refuse misinterpretation of provisioning behavior.

> Personal analysis:
> I guess the sending of msgA's ACK and msgB is concurrently,maybe the first packet of msgA's ACK is earlier than the first packet of msgB. But it still maybe occur the situation [...]

Generic provisioning layer works in form of transactions. Transactions are handled sequentially. There is no concurrent sending\transactions.
According to spec:
```
 5.3.3 Generic Provisioning behavior
...
If the sender receives a Transaction Acknowledgment message, then the transaction has completed.
If the sender receives a message with other PDU types, then the message shall be ignored.
...
```
I guess, in your use case, remote server shall ignore msgB until it receives acknowledgement for msgA.

alxelax, 2024-04-22:
@zhaishiqiang, please try my fix. I wonder if it solves the issue.

zhaishiqiang, 2024-04-23:
HI @alxelax 
Thanks for your help. I have tried your fix and my issue really be solved.