zephyrproject-rtos/zephyr #71442: qspi: nrf driver does not properly wait until chip erase is completed

deveritec-rosc, 2024-04-12:
**Describe the bug**
When issuing a flash erase using `flash_erase` with the flash size, the driver decides to use the chip-erase command (expected). However, the following call to `qspi_wait_for_completion` immediately returns while the chip erase is running on the flash, so successive wrtite operations fail.
If I wait long enough (~10-15s) after issuing the command, a write is possible without issues.

This happens both with the Zephyr version from the nrconnect-sdk v2.6.0 as well as vanilla Zephyr.

Target platform:
- custom board with an NRF52840 (checked this behavior also on the NRF52840 development kit)
- flash: Macronix MX25R8035F

**To Reproduce**
1. set up a simple application using the QSPI
2. run the `flash_erase` command with the target flash size
3. immediately afterward try to write values to the flash

**Impact**
Can't use QSPI for this reason.

**Environment (please complete the following information):**
 - OS: MacOS
 - Toolchain: Zephyr SDK 0.16.5; nrfconnect-sdk v2.6.0

aescolar, 2024-04-23:
CC @de-nordic

de-nordic, 2024-04-23:
Have you tried to alter the `CONFIG_NORDIC_QSPI_NOR_TIMEOUT_MS` ?

This is actually known issue, generally in Zephyr SPI support, as there is no mechanism to properly pass timeout characteristic of a device to SPI/QSPI peripherals that support them.

Basically such characteristic depends on device, and how certain functions of it are executed by that device.

lxbrz, 2024-04-24:
Hello, jumping in here to help @deveritec-rosc .

> Have you tried to alter the CONFIG_NORDIC_QSPI_NOR_TIMEOUT_MS ?

From what I can see this has no influence on the call to `flash_erase` returning immediately. Looking at  the implementation in `modules/hal/nordic/nrfx/drivers/src/nrfx_qspi.c` :

```
nrfx_err_t nrfx_qspi_erase(nrf_qspi_erase_len_t length,
                           uint32_t             start_address)
{
    NRFX_ASSERT(m_cb.state != NRFX_QSPI_STATE_UNINITIALIZED);

    if (!nrfx_is_word_aligned((void const *)start_address))
    {
        return NRFX_ERROR_INVALID_ADDR;
    }

    if (m_cb.handler && m_cb.state != NRFX_QSPI_STATE_IDLE)
    {
        return NRFX_ERROR_BUSY;
    }

    nrf_qspi_erase_ptr_set(NRF_QSPI, start_address, length);
    m_cb.timeout_signal = false;

    if (!m_cb.handler)
    {
        if (!m_cb.activated && qspi_activate(true) == NRFX_ERROR_TIMEOUT)
        {
            return NRFX_ERROR_TIMEOUT;
        }

        nrf_qspi_event_clear(NRF_QSPI, NRF_QSPI_EVENT_READY);
        nrf_qspi_task_trigger(NRF_QSPI, NRF_QSPI_TASK_ERASESTART);

        return qspi_ready_wait();
    }

    m_cb.state = NRFX_QSPI_STATE_ERASE;
    nrf_qspi_event_clear(NRF_QSPI, NRF_QSPI_EVENT_READY);
    nrf_qspi_int_enable(NRF_QSPI, NRF_QSPI_INT_READY_MASK);

    if (!m_cb.activated)
    {
        (void)qspi_activate(false);
    }
    else
    {
        nrf_qspi_task_trigger(NRF_QSPI, NRF_QSPI_TASK_ERASESTART);
    }

    return NRFX_SUCCESS;
}
```

The timeout only has an influence if `qspi_ready_wait()`  is called, which is not the case, because `if (!m_cb.handler)` condition is false. `m_cb.handler` is set to `qspi_handler`.

When stepping through the code I noticed there is `qspi_wait_for_completion` which waits for some interrupt to occur if I'm not mistaken, but this interrupt seems to fire almost immediately. Shouldn't it only fire after the erase operation is completed? This is just a gut feeling, looking at this code from a high level, I don't know details of the implementation or QSPI.

deveritec-rosc, 2024-04-24:
@de-nordic

I checked the code with `CONFIG_NORDIC_QSPI_NOR_TIMEOUT_MS=30000` (since this is the stated maximum time a chip erase should run according to the MX25R8035F datasheet), and it appears to work. I observed that, when selecting a time >5000ms, the function returns as soon as the operation is finished (which is in my case roughly after 5000ms, but I also found boards where it takes some time longer).

So `CONFIG_NORDIC_QSPI_NOR_TIMEOUT_MS` specifies a maximum timeout?

Also, if a chip-erase is performed and I do an MCU reset, the Jedec id can't be read. I would have expected that `CONFIG_NORDIC_QSPI_NOR_TIMEOUT_MS` also takes care of this, but this is not the case?

de-nordic, 2024-04-24:
There are several issues here:
1) yes this is max timeout for any operation, which is a little bit ptoblematic because operations have different timings;
2) if you look at mx spec, or any other mem, you will see that there are different times specified for operations and they also will have ranges (min/max);
3) erase can take minutes on devices if big area is erased and that is done by the mx logic, mcu is not involved; 
4) if you reset mcu the external device may still be completing some op, when you ask it for something it will respond busy.

deveritec-rosc, 2024-04-24:
Okay, fine with what I have here (small 8Mbit flash). Need to check if the error during busy will make any problem, but this is another thing not for this issue.

Thanks for pointing me to the correct setting, haven't found anything regarding this in the documentation.

closed 2024-04-24 by deveritec-rosc

de-nordic, 2024-04-24:
Still, I wonder whether initialization of QSPI external device, involving read of JEDEC ID, should not be immune to busy response.
I also wonder whether reading the JEDEC ID is needed at all, as you probably know what you have on your board anyway, and if it differs from what you expect the firmaware is useless.
And another issue is that JEDEC ID is actually not read correctly either, because it can be up to 7 bytes, as far as I remember the JEDEC spec.

deveritec-rosc, 2024-04-24:
Without looking into the driver code, afaik a failed jedec id read will stop further initialization. I think the driver is looking for a match since I didn't add the data blob with the flash information to the dts file, and the driver wants to ensure that its expected id and data match.

de-nordic, 2024-04-24:
> Without looking into the driver code, afaik a failed jedec id read will stop further initialization. I think the driver is looking for a match since I didn't add the data blob with the flash information to the dts file, and the driver wants to ensure that its expected id and data match.

Sure, is it possible for you to release into the field device that fails jedec id?

deveritec-rosc, 2024-04-25:
Yeah, that will work out. I was curious about why this fails and if this is somehow related to the timing issue, and can be fixed somehow.

For the target application, this shouldn't be a big deal.

de-nordic, 2024-04-25:
> Yeah, that will work out. I was curious about why this fails and if this is somehow related to the timing issue, and can be fixed somehow.
> 
> For the target application, this shouldn't be a big deal.

The problem is not in application, because that can happen any time.
We assume idle state of the QSPI device when MCU starts, because why would you suspect anything happening there?
The problem is that when you do erase and then reset (or any op that takes time), the Zephyr will try to init driver and communicate with the MX, and will get the busy response.
The question is how to fix it? If we really want to do the JEDEC ID check, then it is possible that we would have to wait for minutes till mx completes operation, if that happens on init, the MCU will lag for that time.

deveritec-rosc, 2024-04-25:
> The problem is not in application, because that can happen any time. We assume idle state of the QSPI device when MCU starts, because why would you suspect anything happening there? The problem is [...]

This is completely fine behavior - if necessary I need to take care of it in my application, that's what I meant with "this shouldn't be a big deal".